LinuxCNC/linuxcnc #1407: Gscreen: broken parsing of jog increments without commas

kfoltman, 2021-11-26:
In gscreen gaxis interface, if jog increments are specified using spaces as a delimiter, and 'continuous' is not among the options, the parsing code will add ",continuous" and then notice the newly added comma and proceed to split by comma instead of by whitespaces. This means that the list of jog distances will consist of only 2 entries - "continuous" and something like "10mm 1mm 0.1mm 0.01mm" as a single, unusable entry.

The existing code (src/emc/usr_intf/gscreen/gscreen.py, ~line 651 in linuxcnc-2.8):

            if not "continuous" in increments:
                increments +=",continuous"
            if "," in increments:
                self.data.jog_increments = [i.strip() for i in increments.split(",")]
            else:
                self.data.jog_increments = increments.split()

Note the comma in the string that is added right before the string is checked for the presence of commas. This probably wasn't intended.

This is present both in 2.8 branch and in the master branch. I'm using the 2.8.2 distro.

I know gscreen is a bit old, but this seems to be easy to fix.

c-morley, 2021-11-27:
Why do you use space as a delimited? Comma is the delimited.

kfoltman, 2021-11-27:
Space works fine with axis and gmoccapy. It is also the default format that pncconf is generating, both for metric and for inch increments:

https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/usr_intf/pncconf/data.py#L288

And stepconf, too:

https://github.com/LinuxCNC/linuxcnc/blob/1365386aebea9c9b3f4fe5dc101d1ce0c592b05c/src/emc/usr_intf/stepconf/build_INI.py#L117

Besides, the gscreen code clearly attempts to support both formats, yet fails to do it properly because of what appears to be a simple mistake. I suppose what could work is adding the 'continuous' entry _after_ parsing whichever format is present.

        if "continuous" not in self.data.jog_increments:
            self.data.jog_increments.append("continuous")

c-morley, 2021-11-27:
Interesting - details I've forgotten or didn't notice I suppose.
Yes your solution looks good.

c-morley, 2021-12-07:
Fixed in 2.8/master.
Interestingly I had already fixed it in Qtvcp in much the same way.
Thanks for your discussion!

closed 2021-12-07 by c-morley