LinuxCNC/linuxcnc #1418: patch for lib/linuxcnc when building for non-run-in-place

mnarigon, 2021-11-28:
Building master branch on linux configured with prefix (non-run-in-place)
'''
cd src
./autogen.sh
./configure --with-prefix=/opt/linuxcnc --with-realtime=/usr/realtime
make
sudo make setuid
sudo make install
'''

the halrun.in script template expects the realtime script to be installed in $prefix/lib/linuxcnc/realtime. However, the install-dirs target in the Makefile doesn't create the /lib/linuxcnc directory with respect to the prefix so the install fails.

The patch is to create the /lib/linuxcnc directory with respect to the prefix. The patch is as follows:

'''
diff --git a/src/Makefile b/src/Makefile
index f37812228..482097fa4 100644
--- a/src/Makefile
+++ b/src/Makefile
@@ -613,7 +613,7 @@ install: install-kernel-dep install-kernel-indep
 install-dirs:
 	$(DIR) $(DESTDIR)$(EMC2_RTLIB_DIR) \
 		$(DESTDIR)$(sysconfdir)/linuxcnc $(DESTDIR)$(bindir) $(DESTDIR)$(libdir) \
-		$(DESTDIR)/lib/linuxcnc $(DESTDIR)$(includedir)/linuxcnc \
+		$(DESTDIR)$(prefix)/lib/linuxcnc $(DESTDIR)$(includedir)/linuxcnc \
 		$(DESTDIR)$(docsdir) $(DESTDIR)$(ncfilesdir) \
 		$(DESTDIR)/etc/X11/app-defaults $(DESTDIR)$(tcldir)/bin \
 		$(DESTDIR)$(tcldir)/scripts \
'''

mnarigon, 2021-11-28:
[linuxcnc-003.txt](https://github.com/LinuxCNC/linuxcnc/files/7614161/linuxcnc-003.txt)

BsAtHome, 2026-03-15:
fixed in #2299.

closed 2026-03-15 by BsAtHome