LinuxCNC/linuxcnc #1438: motion.feed-upm not as documented

phillc54, 2021-12-10:
The motion man page states that motion.feed_upm is in machine units per minute.

The actual reported value is G-Code file units per minute. e.g. a metric machine running a G20 file, feed-upm reports inches per minute rather than mm per minute.

I can change the documents if necessary but I am reporting this in the hope that the documentation is correct and the reported value can be changed.

andypugh, 2021-12-10:
I wonder if anyone is depending on the current behaviour?

phillc54, 2021-12-13:
Knowing my luck there would be. :)

andypugh, 2021-12-18:
Maybe the answer is to create yet another pin, motion.feed_mupm ? (or something else, of your choice)

SebKuzminsky, 2021-12-18:
Pins are very cheap in LinuxCNC.  I suggest creating two new pins, `motion.feed_inches_per_minute` and `motion.feed_mm_per_minute`, and updating the documentation to correctly describe the current behavior of `motion.feed_upm`.

This provides full backwards compatibiliy and maximum flexibility for people who want to know the feed in either of the standard units.

rodw-au, 2022-02-09:
Because pins are cheap, I think the units should be per second in line with all of the other velocities reported.
With motion.feed_upm, its not possible to easily compare feed rate with current velocity in halscope which I have found very frustrating. Either add these pins or use them instead of Seb's  suggestions.
```
motion.feed_inches_per_second
motion.feed_inches_per_second
```

SebKuzminsky, 2022-02-10:
How about four new pins:
* `motion.feed-inches-per-minute`
* `motion.feed-inches-per-second`
* `motion.feed-mm-per-minute`
* `motion.feed-mm-per-second`

We should probably keep `motion.feed-upm` unchanged for now, but note in the docs the behavior that @phillc54 reported.  The current manpage is self-contradictory, claiming both that `motion.feed-upm` is in machine units, and that it's based on the program's F-word (which changes units with G20/G21).

rmu75, 2025-12-18:
Closing as this seems to be resolved with #1592