LinuxCNC/linuxcnc #1472: Gmoccapy: Resize buttons on window resize

hansu, 2021-12-26:
@gmoccapy What do you think of resizing the side buttons if the window size is increased?
Image if you have a relatively small screen with a high resolution - the resulting buttons might be too small.

[image]

---> 
 
[image]

gmoccapy, 2023-10-07:
I like the idea, but IMHO it is cosmetic and there are other thinks with more priority.
Solution would be to make our own resize function and link all parts with "self.widgets.window1.resize(width, height) to that function to be able to calculate the available size and resize the buttons according to the available space. 

If you use the corners of the screen to change size, this may lead to a noticeable delay.

hansu, 2023-10-07:
>I like the idea, but IMHO it is cosmetic and there are other thinks with more priority.

I agree. I only wanted to know your general opinion about that.

>Solution would be to make our own resize function and link all parts with "self.widgets.window1.resize(width, height) to that function to be able to calculate the available size and resize the buttons according to the available space.

Maybe we can do this after a mouse-release event after a window-resize event. I have done that for Halshow in Tcl/Tk, not entirely sure if this works the same way with Gtk.

gmoccapy, 2023-10-07:
Sounds like a good idea. I am in stuttgart trying to solve other bugs, but this will stay open. 
Should I assign that one to you?

hansu, 2023-10-07:
Yes -- done.