LinuxCNC/linuxcnc #1473: Gmoccapy: Keyboard jogging ignores halui.max-velocity setting

hansu, 2021-12-26:
When jogging with keyboard it always jogs with the velocity set it gmoccapy even when the max. velocity is reduced by halui.max-velocity.
If using shift+button then the halui.max-velocity setting is applied. 

Procedure: 
1. Set Jogging speed in Gmoccapy to maximum
2.
   setp halui.max-velocity.direct-value 1
   setp halui.max-velocity.scale 1
   setp halui.max-velocity.counts 5 (for example)
3. Jog. Jog velocity is not reduced to halui.max-velocity.value

@gmoccapy We already talked about that issue, I only want to mention it here that it doesn't get lost.

c-morley, 2021-12-26:
Need values to understanf.
In your example max-velocity is set to (1x5x60) 300 ipm or mmpm
You are saying of jog velocity is set to higher then that it will not be capped at 300?

hansu, 2021-12-27:
>You are saying of jog velocity is set to higher then that it will not be capped at 300?

Exactly! I have a max. machine velocity of 2000 and I can still jog with 2000 when this is set in Gmoccapy regardless of the halui.max-velocity setting.

c-morley, 2021-12-28:
Ok Thanks I can confirm Jogging doesn't honour max-velocity cap.
I don't believe this is a GMoccapy bug - it's a linuxcnc bug.

hansu, 2022-01-06:
Yes, in Axis this value is also ignored. But strange that in Axis also Shift+Jog ignores the halui setting.

c-morley, 2022-01-07:
Actually that is what I'd expect. I'd also expect Homing to ignore these settings (and in homing's case, probably should).
Jogging is handled separately from interpreted commands. I'd bet there is no (current system) max_velocity check in the jogging code.
It checks max acceleration and max velocity of the jogging joint. I think you can also set the jogging max accl and max velocity differently then 'interpreted' joint movement (but that's a side note.)

hansu, 2022-01-07:
>Actually that is what I'd expect. 

Yeah I thought about that too. So then throwing the ball back to Gmoccapy and rename it to "Gmoccapy: Shift+Jog shouldn't be affected by halui.max-velocity setting" ?

c-morley, 2022-01-08:
I would guess the docs are wrong. Stat probably shows you the current max velocity setting capped at [TRAJ]MAX_VELOCITY.

I would guess Norbert accounted for max_velocity in Gmoccapy when it sets the jog rate.

The proper way, if we wanted jogging to honour max velocity would be to have the low level jogging code check the requested speed is at or below linuxcnc's current  max velocity setting. Then all the UIs would work the same.

Weather it's worth the effort/disearable may be a question to ask to I guess.

hansu, 2022-01-08:
> Stat probably shows you the current max velocity setting capped at [TRAJ]MAX_VELOCITY.

Yes it does.
So there is no way to get just this value from the config file?

Yes, using `max_velocity` would be the easiest fix that would probably work in many cases.
But I agree that it is safer to use the low level jog command in this case.

hansu, 2022-01-08:
A closer look to the `emc_interface.continuous_jog(axis,cmd)` function offers a very simple fix for this.
Because this function just joggs with a velocity setting of 100, which would be higher than the maximum velocity of most machines (I suppose). So the low level part will then reduce this t the maximum velocity setting.

See PR above for details.

PR #1498 (2022-02-26): Gmoccapy: fix jogging with max. velocity
For details see https://github.com/LinuxCNC/linuxcnc/issues/1473

closes #1473

@gmoccapy
closed 2022-02-26 by c-morley