LinuxCNC/linuxcnc #1486: qtplasma does not work on Wayland

dwrobel, 2021-12-31:
`qtplasma` does not seem to work on the latest Gnome 41.2 / Wayland 1.19 environment available on Fedora 35 _(tested on https://github.com/LinuxCNC/linuxcnc/commit/14a02d19b05caf06e6fccd0865df81440760bbac)_. Note, that `axis` works fine on the same environment.

Here is the screenshot on Wayland after an attempt to load `metric_circles.ngs` file:

[image]

Please note that the `Select` button has been pressed, but the screen hadn't been refreshed _(there is still a file selector dialog visible)_. Shortly after that `Gnome` starts to complain that `qtplasma` is not responding:

[image]

For the reference the same `metric_circles.ngs`  loaded correctly on the `X11` session:
[image]

dwrobel, 2021-12-31:
One more screenshot from Wayland, taken directly after launching `qtplasma` without pressing any buttons:
[image]

Please note that the white rectangle part of the screen does not seem to be refreshed at all and it looks that in `qtplasma` this part is completely transparent as the text _dw@t630:~/Pictures_ comes from the Gnome terminal title bar window which was beneath `qtplasma` when I launched it.

c-morley, 2022-01-01:
Could you try running gcode_graphics directly in a terminal:  python3 gcode_graphics.py
Also interesting if any gladevcp screen grahics display works with wayland.
They use alot of common code.

dwrobel, 2022-01-01:
>Could you try running gcode_graphics directly in a terminal: python3 gcode_graphics.py

Please find attached a screen recording from running it on a `Wayland` session:
```
[dw@t630 linuxcnc]$ loginctl 
SESSION  UID USER SEAT  TTY 
     16 1000 dw   seat0 tty2

1 sessions listed.
[dw@t630 linuxcnc]$ loginctl show-session 16 -p Type
Type=wayland
```
https://user-images.githubusercontent.com/228873/147849733-4bc61b6e-e100-4903-9464-582d5a8069fd.mp4

The output of both console and graphics screens are pretty much fully consistent between `Wayland` and `X11` sessions. 

The code of gcode_graphics.py seems to be session agnostic _(OpenGL based)_ however, grepping the `qtvcp` code showed other places which contains `X11` specific code:

```
[dw@t630 linuxcnc]$ git grep -i -- x11 lib/python/qtvcp
lib/python/qtvcp/lib/xembed.py:# embed qt instance 'window' into X11 window 'parent'
lib/python/qtvcp/lib/xembed.py:def reparent_qt_to_x11(window, parent):
lib/python/qtvcp/lib/xembed.py:# forward events to an X11 window id
lib/python/qtvcp/lib/xembed.py:class X11ClientMessage():
lib/python/qtvcp/lib/xembed.py:        print('X11 message sent')
lib/python/qtvcp/widgets/xembed.py:            self.x11mess = xembed.X11ClientMessage(self.external_id)
lib/python/qtvcp/widgets/xembed.py:    def send_x11_message(self,):
lib/python/qtvcp/widgets/xembed.py:        self.x11mess.send_client_message()
```

```
[dw@t630 linuxcnc]$ git grep -i -- xembed lib/python/qtvcp
lib/python/qtvcp/lib/xembed.py:    # Honor XEmbed spec
lib/python/qtvcp/lib/xembed.py:    atom = d.get_atom('_XEMBED_INFO')
lib/python/qtvcp/lib/xembed.py:class XEmbedFowarding():
lib/python/qtvcp/plugins/simplewidgets_plugin.py:from qtvcp.widgets.xembed import XEmbed
lib/python/qtvcp/plugins/simplewidgets_plugin.py:# XEmbed
lib/python/qtvcp/plugins/simplewidgets_plugin.py:class XEmbedPlugin(QPyDesignerCustomWidgetPlugin):
lib/python/qtvcp/plugins/simplewidgets_plugin.py:        super(XEmbedPlugin, self).__init__(parent)
lib/python/qtvcp/plugins/simplewidgets_plugin.py:        return XEmbed(parent)
lib/python/qtvcp/plugins/simplewidgets_plugin.py:        return "XEmbed"
lib/python/qtvcp/plugins/simplewidgets_plugin.py:        return QtGui.QIcon(QtGui.QPixmap(ICON.get_path('xembed')))
lib/python/qtvcp/plugins/simplewidgets_plugin.py:        return '<widget class="XEmbed" name="xembed" />\n'
lib/python/qtvcp/plugins/simplewidgets_plugin.py:        return "qtvcp.widgets.xembed"
lib/python/qtvcp/qt_istat.py:        # XEmbed tabs
lib/python/qtvcp/qt_makegui.py:    # These can get class patched by xembed library to catch events
lib/python/qtvcp/widgets/screen_options.py:from qtvcp.widgets.xembed import XEmbeddable
lib/python/qtvcp/widgets/screen_options.py:        # add any XEmbed tabs
lib/python/qtvcp/widgets/screen_options.py:            self.add_xembed_tabs()
lib/python/qtvcp/widgets/screen_options.py:    # XEmbed program into tabs
lib/python/qtvcp/widgets/screen_options.py:    def add_xembed_tabs(self):
lib/python/qtvcp/widgets/screen_options.py:                temp = XEmbeddable()
lib/python/qtvcp/widgets/xembed.py:from qtvcp.lib import xembed
lib/python/qtvcp/widgets/xembed.py:class XEmbeddable(QWidget, _HalWidgetBase):
lib/python/qtvcp/widgets/xembed.py:        super(XEmbeddable, self).__init__(parent)
lib/python/qtvcp/widgets/xembed.py:class XEmbed(XEmbeddable, _HalWidgetBase):
lib/python/qtvcp/widgets/xembed.py:        super(XEmbed, self).__init__(parent)
lib/python/qtvcp/widgets/xembed.py:            self.x11mess = xembed.X11ClientMessage(self.external_id)
lib/python/qtvcp/widgets/xembed.py:    ex = XEmbed()
```

```
$ git grep -i -- xset lib/python/qtvcp
lib/python/qtvcp/qt_action.py:            subprocess.Popen('xset -r {}'.format(k), stdout = subprocess.PIPE, shell = True)
lib/python/qtvcp/qt_action.py:            subprocess.Popen('xset r {}'.format(k), stdout = subprocess.PIPE, shell = True)
```

All of those places _(the list might not be exhaustive)_ will very likely not work on the `Wayland` session.

c-morley, 2022-01-01:
Xset is to work around a keystroke problem on some systems.
xembed is for embedding third party programs into qtvcp windows
Both should be optional - I don't think QtPlasmac uses them unless requested.

I don't know pretty much anything about Wayland and if there are equivalent functionality.

rodw-au, 2023-01-14:
Setting the environment variable proposed by Seb should fix this issue on Wayland.
https://github.com/LinuxCNC/linuxcnc/issues/ #2264
It lets QTplasmac run on my Chromebook under Debian Bullseye which has been broken for over 12 months.

There is no doubt a recent change in Bookworm caused this and previously my workaround was to install XFCE and not the default desktop.

dwrobel, 2023-01-16:
>should fix this issue on Wayland.

I'm afraid it won't. Please don't mix two separate issues.

Switching to X11 session type workaround it and I confirmed that also on the initial report _(search for: "For the reference the same metric_circles.ngs loaded correctly on the X11 session:" above)_. 

As long as LinuxCNC will use X11 specific code _(please have a look at the 'grep' output - in one of my above posts)_ this won't work natively on Wayland session type.

rodw-au, 2023-01-16:
I'm not so sure..
Sebs additional post from today isolates the issue and explains what is causing it.
https://github.com/LinuxCNC/linuxcnc/issues/2264#issuecomment-1383523049

I have been using qtplasmac on Bookworm for a long time and it stopped working during November as discussed later in the above issue..

Wayland also includes a Xorg server for legacy applications
Ref: https://docs.fedoraproject.org/en-US/fedora/latest/system-administrators-guide/Wayland/

On  my chromebook, X11 does not exist. Running the commands on that page, confirms that we are running Wayland but that when QTplasmac is running, additional X11 sessions are opened.

`
rod@penguin:~$ echo $WAYLAND_DISPLAY
wayland-0
rod@penguin:~$ xlsclients (no output, no X11 clients running)
rod@penguin:~$ xlsclients (with qtplasmac running shows two X11 clients running)
penguin  python3.9
penguin  python3.9
rod@penguin:~$ 
`
I suspect the two sessions would be the main sim and the torch sim window.

So it would appear that running 
export PYOPENGL_PLATFORM=x11
allows Wayland to serve X11 due to the buggy python3-opengl 3.1.6

phillc54, 2023-01-16:
FWIW I tried Sebs suggestion of _export PYOPENGL_PLATFORM=x11_ on my Bookworm VM.
It does prevent the error messages and allows QtPlasmaC to load but it then has the same transparent preview that @dwrobel has in the second post.

rodw-au, 2023-01-16:
> FWIW I tried Sebs suggestion of _export PYOPENGL_PLATFORM=x11_ on my Bookworm VM. It does prevent the error messages and allows QtPlasmaC to load but it then has the same transparent preview that @dwrobel has in the second post.

Odd, I'm not seeing any issues at all.  I'll try on some real hardware

rodw-au, 2023-01-16:
Hmm, ran tasksel on my xfce box to get wayland installed and the QTplasmac screen became badly corrupt. Its a bit funny I think as it is still using the xfce login screen.  Retreated to xfce and my mesa 7i76e config and it all works,,,

phillc54, 2023-02-11:
[This PR](https://github.com/LinuxCNC/linuxcnc/pull/2340) seems to workaround this issue.