LinuxCNC/linuxcnc #1487: configure: error: boost::python is required to build LinuxCNC (BOOST_PYTHON_LIBS='')

GitHubrist, 2022-01-01:
I am trying to compile a run-in-place (rip) LinuxCNC v2.8 on Ubuntu 20.04.03 compiled with real-time
kernel 5.4.161-rt67.

(This is a re-presentation of the nearly identical ticket related to Arch Linux reportedly resolved in #1007 by the migration to python3.)

## What I'm trying to do:
 * I am trying to build ...
   * a binary version of LinuxCNC v2.8 from source cloned from github.
## What I'm hoping for:
 * A way to postively indicate the location of boost::python/boost_python/BOOST_PYTHON_LIBS 
     libraries/files to ./configure.

## Here are the steps I follow to reproduce the issue:
 1. ~/linuxcnc_2.8/src/$ ./configure --with-realtime=uspace verbose=yes

## This is what I expected to happen:
Successful ./configure.

## This is what happened instead:
"configure: error: boost::python is required to build LinuxCNC"

## Information about my hardware and software:
 * I am using Ubuntu 20.04.3
 * I am using this kernel: 5.4.161-rt67
 * I have installed all libboost*-dev packages
 * I have installed python3* and python3*-dev
 * I have planted "echo $BOOST_PYTHON_LIBS" inside ~/linuxcnc_2.8/src/configure and convinced myself 
     that it is neither pre-defined nor being defined in configure.
 * I have tried, ignorantly and vainly, to 'export BOOST_PYTHON_LIBS=/usr/lib/x86_64-linux-gnu/' to see if 
     'configure' might pick it up.
   * the result was that 'configure' did in fact correctly fetch the value of $BOOST_PYTHON_LIBS, but
       later reset the value to the null string.

 * I am attaching:
   * ~/linuxcnc_2.8/src/config.log.
   * 'pkgs_list': the results of 'sudo apt list --installed' | grep [libboost | python]
   * 'echoes_list.txt': the results of the 'echo' statements planted inside './configure' script. (I may have shot 
        myself in the foot here, nothing that $LINENO is passed to some in-shell functions, and I have altered line 
        numbers by inserting echoes. I don't however have enough Bourne-shell sense to understand what is 
        happening.
[config.log](https://github.com/LinuxCNC/linuxcnc/files/7797857/config.log)
[pkgs_list.txt](https://github.com/LinuxCNC/linuxcnc/files/7797859/pkgs_list.txt)
[echoes_list.txt](https://github.com/LinuxCNC/linuxcnc/files/7797956/echoes_list.txt)

andypugh, 2022-01-07:
Does the master branch compile? If you are building from source I don't know of any current reasons not to use it.

vetsin, 2025-04-29:
This should work with 2.9.4 -- if not try exporting `BOOST_PYTHON_LIB="-lboost_python3.12"; or whatever the python is

andypugh, 2025-06-03:
I am tempted to think that this issue can be closed as having been overtaken by events. 
Also, the original reporter never engaged in the discussion. 

It can be re-opened if anyone disagrees.

closed 2025-06-03 by andypugh