LinuxCNC/linuxcnc #1491: qtplasma: TypeError: setMaximum(self, int): argument 1 has unexpected type 'float'

dwrobel, 2022-01-04:
@c-morley I tested your latest changes 011ebe4846d2100759111c3787220abd9041279d with fixes for QSlider() and `float'  and it broke qtplasma 'Fast/Slow' jogging.

Here is the backtrace when I pressed Jogging Fast button:

```
Traceback (most recent call last):
  File "/home/dw/projects/linuxcnc/share/qtvcp/screens/qtplasmac_4x3/qtplasmac_4x3_handler.py", line 1843, in jog_slow_pressed
    self.w.jog_slider.setMaximum(self.w.jog_slider.maximum() / self.slowJogFactor)
  File "/home/dw/projects/linuxcnc/lib/python/qtvcp/widgets/status_slider.py", line 64, in setMaximum
    return super(DoubleSlider, self).setMaximum(value * self._multi)
TypeError: setMaximum(self, int): argument 1 has unexpected type 'float'
```

dwrobel, 2022-01-04:
Casting arguments to `int()` fixed the issue for me:

```
diff --git a/lib/python/qtvcp/widgets/status_slider.py b/lib/python/qtvcp/widgets/status_slider.py
index 666ea00c86..798aacaf54 100644
--- a/lib/python/qtvcp/widgets/status_slider.py
+++ b/lib/python/qtvcp/widgets/status_slider.py
@@ -58,10 +58,10 @@ class DoubleSlider(QtWidgets.QSlider):
         return float(super(DoubleSlider, self).value()) / self._multi
 
     def setMinimum(self, value):
-        return super(DoubleSlider, self).setMinimum(value * self._multi)
+        return super(DoubleSlider, self).setMinimum(int(value * self._multi))
 
     def setMaximum(self, value):
-        return super(DoubleSlider, self).setMaximum(value * self._multi)
+        return super(DoubleSlider, self).setMaximum(int(value * self._multi))
 
     def setSingleStep(self, value):
         return super(DoubleSlider, self).setSingleStep(value * self._multi)

```

phillc54, 2022-01-10:
I don't see this issue with the latest master branch.
If I add this print statement to the jog_slow_pressed function:
```
def jog_slow_pressed(self, external=False):
    print("self.w.jog_slider.maximum()",self.w.jog_slider.maximum(),type(self.w.jog_slider.maximum()))
```
The result in a terminal is:
`self.w.jog_slider.maximum() 15000 <class 'int'>`

dwrobel, 2022-01-10:
>I don't see this issue with the latest master branch.

I'm observing this issue on `python 3.10` and `python3-qt5 5.15.6`, which is currently available on `Fedora 35`. So, if you don't have enough fresh versions of those packages you might not be able to observe those type of errors yet.

For the record here is the list of currently fixed _(already merged)_ similar issues:
- https://github.com/LinuxCNC/linuxcnc/pull/1478
- https://github.com/LinuxCNC/linuxcnc/pull/1479
- https://github.com/LinuxCNC/linuxcnc/pull/1480
- https://github.com/LinuxCNC/linuxcnc/pull/1482
- https://github.com/LinuxCNC/linuxcnc/pull/1484
- https://github.com/LinuxCNC/linuxcnc/pull/1489
- https://github.com/LinuxCNC/linuxcnc/pull/1485
- https://github.com/LinuxCNC/linuxcnc/pull/1489
- https://github.com/LinuxCNC/linuxcnc/pull/1490

c-morley, 2022-01-12:
Thank you I pushed your suggested changes.

closed 2022-01-12 by c-morley