LinuxCNC/linuxcnc #1499: pncconf adds stepgen DPLL setup boilerplate on non-stepgen configs

pcw-mesa, 2022-01-08:
Pncconf adds:

setp hm2_7i92.0.stepgen.timer-number 1

To 7I92 7I77 only configurations causing failure to load
(since there is no stepgen hardware in the firmware)

c-morley, 2022-01-10:
Ok I pushed a fix to master to look for steppers in the firmware.... but do all stepper firmware have the DPLL option?

pcw-mesa, 2022-01-10:
No, its present and needed mostly on Ethernet hardware

c-morley, 2022-01-11:
is DPLL something that can be read from the mesaflah --readhmid?

pcw-mesa, 2022-01-11:
Yes:

Modules in configuration:

  Module: DPLL
  There are 1 of DPLL in configuration
  Version: 0
  Registers: 7
  BaseAddress: 7000
  ClockFrequency: 100.000 MHz
  Register Stride: 256 bytes
  Instance Stride: 4 bytes