LinuxCNC/linuxcnc #1501: mb2hal Broken pipe with TCP

remy-mr, 2022-01-09:
Hi.
I get "ERROR Broken Pipe" , when trying to communicate with my VfD in LinuxCNC and mb2hal component. 

Is this from mb2hal, LinuxCNC, or Linux Mint 19?

When I send the commands via the program QModBus everything is fine.
[image]
[image]

andrewheeler82, 2022-01-21:
Hello, I looked at the pictures. Can it be that there is no connection? Is it possible to get a detailed log?

hansu, 2022-01-21:
Which LinuxCNC version are you using?

remy-mr, 2022-02-06:
Hi.
Sorry for late reply.
LCNC 2.8.1

For now I switched back to wired connection via USB-RS485 adapter to my Vfd and only one TCP block in the INI for the relays.
I have to mention, that I use an ESP32 module as modbus-server.

First I run TCP-RTU-bridge together with server on this ESP.
Testing its function with QModBus run without problems. But mb2hal shows "Broken pipe".
So all I want to know is, who is writing this error message. Is it linux kernel, Lcnc, or the mb2hal comp. From there on I hope to go on examine what is the problem.

hansu, 2022-02-10:
>I get "ERROR Broken Pipe"  ...
>Is this from mb2hal, LinuxCNC, or Linux Mint 19?

i think the "ERROR" comes from the libmodbus library: 
https://github.com/stephane/libmodbus/blob/7c8dbd29e5a924f3bb157f0d37ff522bd821e0af/src/modbus.c#L84
and the "Broken pipe" from the system.

andrewheeler82, 2022-02-10:
> > I get "ERROR Broken Pipe"  ...
> > Is this from mb2hal, LinuxCNC, or Linux Mint 19?
> 
> i think the "ERROR" comes from the libmodbus library: https://github.com/stephane/libmodbus/blob/7c8dbd29e5a924f3bb157f0d37ff522bd821e0af/src/modbus.c#L84 and the "Broken pipe" from the system.

I agree with that. Also think it comes from libmodbus.