LinuxCNC/linuxcnc #1505: problem with mitsub_vfd.py

hitchhiker85, 2022-01-10:
hi... i want tu use mitsub_vfd.py

i install via halcmd

when i start the mitsub_vfd.py i get error.
its a fresh installed linuxcnc from a livecd.

vfd is ready and connected, same setup works on mach3...

 i am very new to linuxcnc and linux... sorry for the request.thanks

cnc@cnc:~$ halcmd loadusr mitsub_vfd -p /dev/ttyUSB1 spindle=01
cnc@cnc:~$ ('/dev/ttyUSB1', 9600)
Mitsubishi VFD serial computer link has loaded
Port: /dev/ttyUSB1,
baudrate: 9600
8 data bits, no parity, 2 stop bits

Mitsubishi spindle VFD: slave# 01 added

Traceback (most recent call last):
  File "/usr/bin/mitsub_vfd", line 436, in <module>
    app = mitsubishi_serial(vfd_names=device_ids,baudrate=int(baud),port=port)
  File "/usr/bin/mitsub_vfd", line 117, in __init__
    self.set_special_monitor()
  File "/usr/bin/mitsub_vfd", line 296, in set_special_monitor
    word = self.prepare_data(cmd,option)
  File "/usr/bin/mitsub_vfd", line 318, in prepare_data
    return bytes(converted_data, 'utf-8')
TypeError: str() takes at most 1 argument (2 given)

c-morley, 2022-01-11:
answered on the forum - might be a python2/3 problem depending on linuxcnc version.
https://forum.linuxcnc.org/10-advanced-configuration/44782-mitsubishi-vfd-modbus-cant-get-it-work

hitchhiker85, 2022-01-11:
hi again..

i send on terminal :  
sudo halcmd loadusr mitsub_vfd --port /dev/ttyUSB0

and get:

cnc@cnc:~$ /dev/ttyUSB0 9600
ERROR : mitsub_vfd - No serial interface found at /dev/ttyUSB0

dialout is set and ttyUSB0 is the correct Device,Port, its ch341 rs232 to rs485

thanks

c-morley, 2022-01-31:
Closing this - the problem was using modbus protocol rather then Mitsubishi protocol.

closed 2022-01-31 by c-morley