LinuxCNC/linuxcnc #1511: Gmoccapy & Gscreen: Keyboarg jogging continues when window looses focus

hansu, 2022-01-16:
If you switch to another window (or another window comes into the foreground) and a jogging key is held down - the jogging will continue until you go into the LinuxCNC window and press the jog key again.

One possibility to solve this would be to catch the Gtk `focus-out-event`  and stop all jogging on this.
Or we could check the status of the jogging keys periodically.

c-morley, 2022-01-31:
probably focus-out would be the safest. I really dislike keyboard jogging :)

andypugh, 2022-01-31:
I think that perhaps we should disable keyboard jogging by default, and only turn it on after the users have ticked a box saying that they understand that it's a stupid idea.

hansu, 2022-01-31:
> I think that perhaps we should disable keyboard jogging by default, and only turn it on after the users have ticked a box saying that they understand that it's a stupid idea.

Yeah, maybe we could add a warning somewhere.

closed 2022-03-23 by hansu