LinuxCNC/linuxcnc #1517: documented apt-key public key not found

ebo, 2022-01-19:
While following the instructions in http://linuxcnc.org/docs/2.8/html/getting-started/getting-linuxcnc.html
7.1. Installing on Debian Buster (with Preempt-RT kernel): 

we have the following code
``sudo apt-key adv --keyserver hkp://keys.openpgp.org --recv-key 3cb9fd148f374fef``

This generates an error that it cannot be found.  Looking around I found I could use:

`sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com --recv-key 3CB9FD148F374FEF`

Someone may need to review the issue and resolve it by registering the key with the specific server or changing the doc.

andypugh, 2022-02-09:
The alternate keysever is now listed:
http://linuxcnc.org/docs/2.8/html/getting-started/getting-linuxcnc.html#_alternate_install_methods
And here: http://buildbot.linuxcnc.org/

Is that enough to close this?

ebo, 2022-02-10:
I do not have time at the moment to go through this and check.  Go ahead and mark it closed if you think it resolves the issue for now and I will reopen if necessary.

ebo, 2022-02-10:
ok, I 've only taken a minute to check .  Ehrn I run the command:

    sudo apt-key adv --keyserver hkp://keys.openpgp.org --recv-key 3cb9fd148f374fef
   sudo apt-key adv --keyserver hkp://keys.openpgp.org --recv-key EF1B07FEE0EE663E

which are both found in  documents listed, I now get the error:

    gpg: key 3CB9FD148F374FEF: new key but contains no user ID - skipped
and
    gpg: key EF1B07FEE0EE663E: new key but contains no user ID - skipped

Also to note.  apt-key has been deprecated.  You might want to add this to the documentation:

    Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
    Executing: /tmp/apt-key-gpghome.i8RmOmvnAW/gpg.1.sh --keyserver hkp://keys.openpgp.org --recv-key 3cb9fd148f374fef

Let me know when the new keys have been sorted out and I will try again.

smoe, 2022-02-18:
There is some good chance to soon land in Debian-proper "real soon" :)  Daily builds aside I hope that we can substitute this repository with regular backports within Debian. The md5 sum is presented, so no immediate action is required, I think, but I admittedly could not resist to have a look myself:
```
$ gpg --search-key EF1B07FEE0EE663E
gpg: data source: https://keys.openpgp.org:443
(1)       2048 bit RSA key EF1B07FEE0EE663E, erzeugt: 2014-06-27
```
The 2048 bits are below the standard of 4096 that Debian has now, but ... does this work for you?