LinuxCNC/linuxcnc #1519: uncontrollered move to max limit when F3 (manual) to MDI mode switch

eginap, 2022-01-20:
# The issue tracker is not a support forum

The LinuxCNC issue tracker is to report bugs in the software.
If you have a question about how to use the software, use one of the other methods detailed on our community support page: http://linuxcnc.org/community/

(delete this section before submitting your bug report)

## Here are the steps I follow to reproduce the issue:

 1. press button --> move joint
 2. press MDI mode tab
 3. unpress joint button
 4. then actual position thats joint moves to max limit that all 

## This is what I expected to happen:
stop move untill I unpress button when MDI mode active
## This is what happened instead:

move to limit joint

## Information about my hardware and software:

sim and real hardware repeat these behavior

andypugh, 2022-01-20:
Which button are you pressing? Which User Interface are you using?

eginap, 2022-01-20:
AXIS
keyboard buttons for move spindel/ tool /

pcw-mesa, 2022-01-23:
That's a pretty bad bug. Maybe disallow mode changes when a jog is in progress?

andypugh, 2022-01-23:
I just checked with 2.7, and in that case the jog stops when you change to the MDI window.

c-morley, 2022-01-23:
So you are pressing a holding a jog with keyboard) and at the same time changing to MDI mode with the meyboard?
or same thing but using the mouse to change to MDI?

pcw-mesa, 2022-01-23:
I duplicated the problem (with 2.9+Axis) 

By keyboard jogging (say pressing the right arrow key) and the changing to
 MDI mode with the mouse. 

Admittedly a strange thing to do but one that causes a hazard. Not sure if 
blocking mode changes while jogging,  aborting jogs in progress on a mode change , 
or suspending jogs in progress when not in manual mode or something entirely 
different is the best solution...

andypugh, 2022-01-23:
Either clicking with the mouse on the MDI tab, or pressing F5 will leave the system jogging when you release the arrow key.

c-morley, 2022-01-23:
related to this fix?
https://github.com/LinuxCNC/linuxcnc/issues/691

andypugh, 2022-01-24:
According to a git bisect the issue was introduced here:
3de8778374bcee0b184c0c4ad04a373585efb7e0

andypugh, 2022-01-28:
@eginap This issue should be fixed in the next 2.8 release. If you need the fix sooner, then you should be able to get a build from the buildbot an hour or so from now.

closed 2022-12-25 by andypugh