LinuxCNC/linuxcnc #1524: pncconf kernel error

seebaer1976, 2022-01-23:
## Here are the steps I follow to reproduce the issue:

 1. install a fresh 2.8.2, pncconf work
 2. update to last 2.9 master, pncconf don´t start
 3. no matter which kernel, after the update to 2.9. I even installed 5.10 
 
 [image]

## It worked properly before this:
after upgrading from LinuxCNC 2.8.2 to 2.9

## Information about my hardware and software:

 * lsb_release -a:
    No LSB modules are available.
    Distributor ID: Debian
    Description:    Debian GNU/Linux 10 (buster)
    Release:        10
    Codename:       buster

 * uname -a:
   Linux linuxcnc 5.10.0-0.bpo.9-rt-amd64 SMP PREEMPT_RT Debian 5.10.70-1~bpo10+1 (2021-10-10) x86_64 GNU/Linux
   no matter which kernel, after the update to 2.9. I even installed 5.10

 * I am running ...
   A binary version from linuxcnc.org (including buildbot.linuxcnc.org)

 * 2.9

 * AXIS

 * ethernet port, mesa 7i96

c-morley, 2022-01-23:
So I have just pushed a fix for the error.
but this message is supposed to be used when using a RTAI realtime system with the wrong kernel.
Are you using a RTAI kernel?
Otherwise I'm confused why you got the message at all - and may need to do further work to get the tests to work.

seebaer1976, 2022-01-29:
pcconf runs.

I only installed the PREEMPT_RT kernel.
as I said, the message came after the update from 2.8.2 to master 2.9, after a fresh 2.8.2 installation.

c-morley, 2022-01-31:
strange - ok I'll close this. feek free to open it again if there is more trouble.

closed 2022-01-31 by c-morley