LinuxCNC/linuxcnc #1527: Gtk3 front end for hal_manualtoolchange

hansu, 2022-01-23:
I added the possibility to hal_manualtoolchange to have a Gtk window instead of Tk to better fit to the user interface.
It detects from the ini file which [DISPLAY] is used and decides based on that if is uses gtk or Tk.
It can also be selected by argument.

Looks so:
[image]

But as I realized now, that Gmoccapy and Qtvcp have there own toolchanger dialogs, this seems a bit superfluous.
But may someone has a use for it:
https://github.com/hansu/linuxcnc/blob/manualtoolchange/src/emc/usr_intf/axis/scripts/hal_manualtoolchange.py

c-morley, 2022-01-23:
The advantage of separate dialogs is the ability to modify them within the main screen.
Usually this is just the style or theme of the dialog, but can be screen position and different button options. 
The original was built for AXIS which was notorious for it's lack of customization-ability.

hansu, 2022-01-24:
Yes and they can provide more information like a tool description. So that's why this change is useless in most cases.

c-morley, 2022-01-27:
Ok I'm going to close this then.

closed 2022-01-27 by c-morley

hansu, 2022-01-27:
This was more intended to be a template if someone want's to do something like that than a feature request. But maybe the wrong place for that.