LinuxCNC/linuxcnc #1529: Documentation - Developer - Missing Coding Style for Documentation

smoe, 2022-01-24:
Hello,

The developer manual describes the C++ coding styles and I would very much like tot see the analog for the asciidoc files. The idea here is that once we have an agreement on how the descriptions of functions should look or how tables and figures should be described/referenced, what is translated and what not, then this may help the looks of the documentation a lot and the acceptance of LinuxCNC for whoever is deciding to adopt the technology.

What I also see is that the task to further improve the documentation can be split a bit more between those who provide the content, or a skeleton of it, those who can transform that skeleton into some text with figures and tables, and then those who have such a description to bring it all into shape.

I volunteer to be the one to come up with a skeleton and some details on how I would like to see tables be presented.

Steffen

silopolis, 2022-01-24:
Hi,

Man you're doing such a job! 😲💪

Le lun. 24 janv. 2022 à 13:32, Steffen Möller ***@***.***> a
écrit :

> The developer manual describes the C++ coding styles and I would very much
> like tot see the analog for the asciidoc files. The idea here is that once
> we have an agreement on how the descriptions of [...]

Can't agree more with this proposal as it would solve a bunch of the
concerns expressed a few days ago on #linuxcnc-devel!

From top of my head, I'd add:
* header tags block like toc etc
* anchors for titles and other blocks
* index entries for titles and other blocks
* lists formatting
* usage of lists vs titles
* use of bold and italic
* structure/page splitting guidance (2k+ lines pages doesn't look good to
me)
* footnotes
* comments

I volunteer to be the one to come up with a skeleton and some details on
> how I would like to see tables be presented.
>

That skeleton would surely be valuable! 👍

As I'm already struggling to complete french docs restructuring, I won't
offer help to write these docs, but would love to discuss and comment along
dev... Actually, I've already thought looking at other significant projects
using Asciidoc to see how they use it and if they have such documents to
derive....

Pushing it further, I think we should discuss adoption of
https://antora.org/ to manage our various versions, and eventually be ready
for project code restructuring like MK has already done.

TY
Jé

smoe, 2022-01-24:
> Le lun. 24 janv. 2022 à 13:32, Steffen Möller ***@***.*** a écrit :

> Can't agree more with this proposal as it would solve a bunch of the concerns expressed a few days ago on #linuxcnc-devel!
> From top of my head, I'd add:
> * header tags block like toc etc
> * anchors [...]

>> I volunteer to be the one to come up with a skeleton and some details on how I would like to see tables be presented.

> That skeleton would surely be valuable! +1
> As I'm already struggling to complete french docs restructuring, I won't offer help to write these docs,

I somehow sense you already came up with quite a nifty skeleton.

> but would love to discuss and comment along dev... Actually, I've already thought looking at other significant projects using Asciidoc to see how they use it and if they have such documents to [...]

Looks neat, did not know about it. Don't think that this tech solves our immediate "problems", though. These are not really problems in the first place, it is more like a nagging discrepancy between all the good bits in LinuxCNC and the clunkiness (was looking for a word that combines "steep" with "not smooth") with which it presents itself to its users. Some parts are great. Others are not.

> and eventually be ready for project code restructuring like MK has already done.

Not even the MK seem to want to split our now monolithic git repository. That is another thing. Maybe we are so eager to restructure everything since we fail to understand the LinuxCNC building blocks and are hoping any such restructuring to help conquer LinuxCNC in parts A better documentation may come with a bit of extra piece of mind.

silopolis, 2022-01-24:
Le lun. 24 janv. 2022 à 20:48, Steffen Möller ***@***.***> a
écrit :

> Le lun. 24 janv. 2022 à 13:32, Steffen Möller *@*.*** a écrit :
>
> As I'm already struggling to complete french docs restructuring, I won't
> offer help to write these docs,
>
> I somehow sense you already came up with quite a nifty skeleton.
>

Really just a rough idea trying to mix the best of the source and french
docs coding styles.

For starter, I'd be in favor of having a ::toc:: at the beginning of every
file. We should also investigate if there are other intersting tags to
include in file headers.

Then, a policy should be designed for [[title anchors]] so crosslinks
become easy to use and understand. Style guide should explicit where to use
cha/sec/sub prefixes and how to name anchors according to titles.

Same applies for (((index entries))). A rich and clear index would be a
very helpful part of the docs

Along with the index, we should have listings of illustrations, tables and
significant code blocks. I still have to understand how this can be done...
But I presume this also revolves around anchoring and tagging each and
every one of these in a suitable way.

A rich and detailed glossary is of prime importance and we need to first
understand if and how weblate's glossary features can help with this. Then
devise how references to entries should be included in docs pages.

 Pushing it further, I think we should discuss adoption of
> https://antora.org/ to manage our various versions,
>
> Looks neat, did not know about it. Don't think that this tech solves our
> immediate "problems", though.
>

No it doesn't, but considering the amount of work we're about to invest,
I'd rather not miss the opportunity to future proof our investment making
it ready for higher level tools.

These are not really problems in the first place, it is more like a nagging
> discrepancy between all the good bits in LinuxCNC and the clunkiness (was
> looking for a word that combines "steep" with "not smooth") with which it
> presents itself to its users. Some parts are great. Others are not.
>

And that's why all this is well worth the sweat! 😉

and eventually be ready for project code restructuring like MK has already
> done.
>
> Not even the MK seem to want to split our now monolithic git repository.
> That is another thing.
>

Mainly, I was aiming at the different versions branches.

Maybe we are so eager to restructure everything since we fail to understand
> the LinuxCNC building blocks and are hoping any such restructuring to help
> conquer LinuxCNC in parts A better documentation may come with a bit of
> extra piece of mind.
>

Couldn't have put it better!

TY
Jérémie

silopolis, 2022-01-30:
https://github.com/owncloud/docs/blob/master/docs/best-practices.md may be worth a read also.

havardAasen, 2022-07-29:
I know this is still a work in progress, and this is only partly relevant, but I've noticed that we have several files related to code style and contribution. Some of these files is buried quite deep, and difficult to find. This is all nice when we build and generate the documentation, but when you only want to create a simple PR. I don't think these files is accessible enough.

Could we have a file called `CONTRIBUTING.adoc` in the root directory and stitch in the relevant files?
This would make it a lot easier to discover from both GitHub and from your file system if you fork and download the repo.

The potential file candidates I've seen so far:
```
src/CodingStyle
docs/src/code/style-guide.adoc
docs/src/code/contributing-to-linuxcnc.adoc
```
`docs/src/code/building-linuxcnc.adoc` is also a file that could be more accessible.

When it comes to the files `src/CodingStyle`and `docs/src/code/style-guide.adoc`, they contain the same information.

silopolis, 2022-07-30:
All good valid points 👍

My option would be first and of course to merge/restructure/clean up all
these files so we have a nice complete and consistent documentation for all
developers and contributors of any kind.

BTW, my opinion is also that this should be split in a dedicated manual
(along with a user one, and an integrator one).

Then the index of this guide should be pointed at in a section of the main
root readme. The documentation would then be where it should and only
there, as well as located on/by the very first page of the project.

hansu, 2022-09-02:
What about add a link in  `docs/src/code/contributing-to-linuxcnc.adoc` to `docs/src/code/style-guide.adoc`? Because contributing gives a more a general overview.

`src/CodingStyle` could be removed, yes.

And I wonder if we may should create a separate file for the documentation style guide if it becomes too long for the code style guide.

silopolis, 2022-09-02:
Le ven. 2 sept. 2022 à 10:22, Hans Unzner ***@***.***> a
écrit :

> What about add a link in docs/src/code/contributing-to-linuxcnc.adoc to
> docs/src/code/style-guide.adoc? Because contributing gives a more a
> general overview.
>
> src/CodingStyle could be removed, yes. [...]

I believe so.
Like I think the contributing page should be introductory and holding the
ToC for this part.
This would put structure in place for this part to expend, maybe someday in
a proper dev guide...

hansu, 2022-09-02:
> Like I think the contributing page should be introductory and holding the
ToC for this part.

Yeah good idea :-)

## Contributing to LinuxCNC
## Introduction
...
### Code Style
### Documentation Style

I never thought that so much is needed for a good documentation 🙈

smoe, 2022-09-02:
I suggest we just start with it. Any such document will never be completed, so we should not wait. Would also rather see it as a structured documentation of decisions we made while discussion PRs to the docs.

smoe, 2022-09-02:
I just saw that I had come up with this issue and was a bit shocked. @silopolis , please consider yourself the owner of this issue.

andypugh, 2022-09-03:
Trying to catch up, and just skimming. 
Should this mention "no more than three layers of headers" as otherwise the build logs end up full of errors. 
(Might be 4 levels. The issue is that a top level is added during docs assembly and that breaks adoc as it only supports a limited number of header levels.

silopolis, 2022-09-03:
Le sam. 3 sept. 2022 à 10:33, andypugh ***@***.***> a écrit :

> Trying to catch up, and just skimming.
> Should this mention "no more than three layers of headers" as otherwise
> the build logs end up full of errors.
> (Might be 4 levels. The issue is that a top level [...]

This is 5 total actually.

This limitation is also a technical point to add to my restructuring
proposal.

silopolis, 2022-09-04:
@smoe could you please link the PR #1530 to that issue ?