LinuxCNC/linuxcnc #1550: Request HAL comp - or3, or4, and3, and4

zz912, 2022-01-29:
I would like to ask you to add a HAL component:
or3
or4
and3
and4
ideally up to version 2.8

It's annoying when I have to replace and3 with two and2.

Thank you

hansu, 2022-01-29:
You can use the hal component `logic` for that: http://linuxcnc.org/docs/html/man/man9/logic.9.html

zz912, 2022-01-30:
Logic looks great. I would like to ask why the and2 and or2 components were created. If and2 and or2 only exist for historical reasons, would it be possible to point out the logic function in the manual? http://linuxcnc.org/docs/2.8/html/man/man9/and2.9.html
http://linuxcnc.org/docs/2.8/html/man/man9/or2.9.html

hansu, 2022-01-30:
Maybe because `and2` and `or2` are very common components, so they are provided separately for easier use.
But good point, I'll add a link to `logic` in the man pages of `and2`, `or2` and so on.

andypugh, 2022-01-30:
On Sat, 29 Jan 2022 at 21:27, Hans ***@***.***> wrote:

> You can use the hal component logic for that:
> http://linuxcnc.org/docs/html/man/man9/logic.9.html
>

Also consider LUT5, it is very flexible once you figure it out. It gives a
defined output for every combination of 5 inputs.
Where it can be made very powerful is if one of the inputs is driven by the
output, then it becomes state-aware and can be useful for things like
interlocks.

https://linuxcnc.org/docs/2.8/html/man/man9/lut5.9.html

I made an online spreadsheet for figuring out the mask:
https://docs.google.com/spreadsheets/d/1_VlL39jUOPpS8Tt9mpRK-eaSTS6C6PHBDoE3JitvLzs/edit?usp=sharing

-- 
atp

hansu, 2022-01-31:
The link is now in the docs. So can this issue be closed?

zz912, 2022-01-31:
Yes.

PR #1553 (2022-01-31): Some work on docs (2.8)
closes #1550
closed 2022-02-01 by c-morley