LinuxCNC/linuxcnc #1551: Python module in 2.8 reports requested instead of actual spindle speed

kentavv, 2022-01-30:
Maybe it's intended behavior that the LinuxCNC module report requested spindle speed instead of actual. Given the description in the documentation I was expecting actual. Below is all I used to test while the spindle was running. Could be nice to have actual and requested speeds available in the spindle dictionary. If others are interested, I could try to hack it in. Thank you

```
import linuxcnc
while True:
   lcnc = linuxcnc.stat()
   lcnc.poll()
   print(lcnc.spindle[0]['speed'])
```

c-morley, 2022-01-30:
I'm not sure it's worth the effort. the actual speed can be easily had with:
 hal.get_value('spindle.0.speed-out')

If you use the hal_glib library, this is how the actual speed is derived.

kentavv, 2022-01-30:
I had no idea one could pull any value so easily, and without restarting LinuxCNC. Thank you!

spindle.0.speed-out contained the commanded speed, but lowpass.spindle.out (and others) held actual speed in various forms.

```
>>> import hal
>>> h = hal.component("testing")
>>> hal.get_value('spindle.0.speed-out')
301.0
>>> hal.get_value('lowpass.spindle.out')
313.8461538461539
```

closed 2022-01-30 by kentavv

c-morley, 2022-01-30:
yes sorry I meant  
hal.get_value('spindle.0.speed-in')
but you can use whatever you want.