LinuxCNC/linuxcnc #1564: Linuxcnc v2.8.2 and 2.9.0 STATUS Bug

Lcvette, 2022-02-02:
Linuxcnc Master 2.9 and Stable 2.8 STATUS has broken recently and are not updating correctly. Iit appears that 2.8 is not updating gcodes and mcodes and 2.9 is not updating mcodes.  This appears to be an issue with programs, but seems to work when using MDI on both versions.  actions also are not updating in status.

This has broken some core functionality in the QtPyVCP user interfaces.  the test was performed in the axis sim on both versions and is repeatable.

## Here are the steps I follow to reproduce the issue:

 1. open axis on either version and set the machine up to run the default LinuxCNC program.
 2. open the status window and setup the workoffsets to run the program.
 3. run the program and note that the status is not registering some/all of the g and m codes as they are called in the program.

## This is what I expected to happen:
I expected the status to update with the currently active g and m codes

## This is what happened instead:
the status does not update for some item in version 2.9
the status does not update for most items both g and m codes in 2.8

## It worked properly before this:
this functionality was working perfectly before and has been called out by numerous users recently as it effects QtPyVCP interfaces by causing issues loading the backplotter drawn lines at program load.  it also effect other critical functionality relying on proper status updating.

## Information about my hardware and software:
tests were run on debian 10 and debian 11 for both versions.

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

and

lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye

 * I am using this kernel version (shown by `uname -a`):

uname -a
Linux PBDEV 4.19.0-18-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.208-1 (2021-09-29) x86_64 GNU/Linux

and

uname -a
Linux py3dev 5.10.0-10-rt-amd64 #1 SMP PREEMPT_RT Debian 5.10.84-1 (2021-12-08) x86_64 GNU/Linux

 * I am running ...
iso install from linuxcnc.org for the deb10 version 2.8.2

and running this install for deb 11 version 2.9
https://github.com/joco-nz/lcnc-bullseye-installer

 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
The issue effects all gui's i hav tested on in cluding QtPyVCP gui's and Stock Linuxcnc supplied Gui's

 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 

I am running the following hardware for deb 11 version 2.8:
cnc@PBDEV:~$ lspci -v
00:00.0 Host bridge: Intel Corporation Core Processor DRAM Controller (rev 02)
	Subsystem: Lenovo Core Processor DRAM Controller
	Flags: bus master, fast devsel, latency 0
	Capabilities: <access denied>

00:01.0 PCI bridge: Intel Corporation Core Processor PCI Express x16 Root Port (rev 02) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 24
[... 219 lines cut ...]
	I/O ports at f0d0 [size=8]
	I/O ports at f0c0 [size=4]
	I/O ports at f0b0 [size=8]
	I/O ports at f0a0 [size=4]
	I/O ports at f060 [size=32]
	Memory at f7d36000 (32-bit, non-prefetchable) [size=2K]
	Capabilities: <access denied>
	Kernel driver in use: ahci
	Kernel modules: ahci

00:1f.3 SMBus: Intel Corporation 7 Series/C216 Chipset Family SMBus Controller (rev 04)
	Subsystem: Hewlett-Packard Company 7 Series/C216 Chipset Family SMBus Controller
	Flags: medium devsel, IRQ 18
	Memory at f7d35000 (64-bit, non-prefetchable) [size=256]
	I/O ports at f040 [size=32]
	Kernel driver in use: i801_smbus
	Kernel modules: i2c_i801

01:00.0 Network controller: Intel Corporation Centrino Advanced-N 6205 [Taylor Peak] (rev 34)
	Subsystem: Intel Corporation Centrino Advanced-N 6205 AGN
	Flags: bus master, fast devsel, latency 0, IRQ 28
	Memory at f7c00000 (64-bit, non-prefetchable) [size=8K]
	Capabilities: <access denied>
	Kernel driver in use: iwlwifi
	Kernel modules: iwlwifi

c-morley, 2022-02-04:
I confirm something as you describe in master. If I run chips program and single step while watching the status window, I can see about every third step that the m3/m5 changing when it should not - might coincide with m2 being called after subroutine. I tried AXIS and Qtdragon. I haven't had time to git-bisect yet.
When you say linuxcnc 2.8.2, (If you know) does it include the bug fixes after 2.8.2 that are not officially released yet?

Lcvette, 2022-02-04:
> 

Chris,  this test was done with installed iso download and then updated via sudo apt update/upgrade.  I am guessing it would not have any of the mentioned unreleased fixes.  non of the status seem to be updating from a program in the 2.8 installed version i was testing.  they do update from mdi.  they also do not update from action button calls.

andypugh, 2022-02-06:
In master I see "M2" a lot of the time in the M-codes. Which doesn't seem right. 
I have gone back to 2.7.15 via 2.8.2 and bsasically current master seems to be the best of the lot. It does at least show G0 / G1 / G2 / G3 correctly. 
I was going to do a Git Bisect to find where it stopped working, but I am not sure that it ever worked, and is more nearly working now. 
Perhaps a test program is needed that exercises all the codes that you feel should be updating, so that it is easier to work out whether a particular build is "good" or "bad" as I am not sure that I have managed to find a "good" status.

satiowadahc, 2022-02-18:
Started a testing app for status. Not sure if this has a place in this repository.
https://github.com/satiowadahc/LinuxcncTests
- Requires pytest, pytest-qt
- Run linuxcnc with another UI ( I might look into finding a way to add it to the tests)
- pytest test_status.py

Lcvette, 2022-02-18:
Awesome!!

Lcvette, 2022-02-18:
no i was using the standard release on 2.8.2, just what updates with sudo
apt update/sudo apt upgrade.

silopolis, 2022-02-19:
Le ven. 18 févr. 2022 à 22:11, Chad A. Woitas ***@***.***> a
écrit :

> Started a testing app for status. Not sure if this has a place in this
> repository.
> https://github.com/satiowadahc/LinuxcncTests
>
>    - Requires pytest, pytest-qt
>    - Run linuxcnc with another UI
>
>
Just reading pytest and LinuxCNC in the same message made my day!

Makes me dream of a tests/functional subdir with a large coverage suite 🙏

Great work, hope we'll get more of this

TY
Jérémie

PS: another entry for the 3.0/GSoC project list

Message ID: ***@***.***>
>

satiowadahc, 2022-02-19:
3 tests complete... 85 more to go. This may be a slow process.

Lcvette, 2022-02-19:
I have PCs available for testing or running tests if needed, bot 2.8.2 and
2.9

andypugh, 2022-02-22:
How is this going?

satiowadahc, 2022-02-22:
The test stubs have been written, documenting current issues, state and plans as I go. Some of status seems vague to me, so someone may need to validate the codes are right. 

Once it's more complete, is there a place in this repository for it?

Lcvette, 2022-02-22:
awesome!  let me know if you need testing of any sort! and thank you for
looking into this!

Lcvette, 2022-02-22:
looks like Chad has some things cooking \o/ did you find anything else
Andy?

cakeslob, 2022-03-07:
Hey, sorry to chime in,  I had issues with this when playing with the feedrate pin , motion.feed-upm. Here is early observations I had in this thread
https://forum.linuxcnc.org/38-general-linuxcnc-questions/41031-problem-with-feed-rates

Ill keep it short because I dont know what Im talking about compared to you guys, and because I wasnt working with G/M codes but with the F/S codes in the linuxcnc status. The main thing I observed was the difference between using G64, and G61/ G64 P0.001 Q0.001
You can watch in linuxcnc status under the "settings" line as the current line changes with the settings, and not the motion line

With G64, on my test program, it reads ahead 3000+ lines and displays F and S from the last point in the read ahead F/S were changed, which was 2000+ lines from the current line. You can observe jumping around of these settings as the it goes from g0 to g1/2/3 moves, 

With G61/ G64 P0.001 Q0.001 it displays F/S correctly because it is not reading to far ahead of the actual motion line. This does not seem to help with M codes, for my program it just reads m5 and starts to toggle m30 towards the end of the program.

silopolis, 2022-08-25:
How hard is this one to fix ?
Any chances to have it for 2.9 ?

satiowadahc, 2022-08-27:
Summer got busy for me, hoping to get back to this come October/November.

silopolis, 2022-08-27:
Le sam. 27 août 2022 à 14:01, Chad A. Woitas ***@***.***> a
écrit :

> Summer got busy for me, hoping to get back to this come October/November.
>

Having this fixed by the end of the year would be awesome!

My understanding is that UIs would greatly benefit from this (at least
QtPyVCP crew is praying it) so it'd be great to have it in 2.9 to unleash
their development...

Thanks for the (personal) STATUS update 😉