LinuxCNC/linuxcnc #1566: Translation of hal_mdihistory breaks Gmoccapy translation

hansu, 2022-02-03:
In https://github.com/LinuxCNC/linuxcnc/commit/446e7addd3be542c1da5d6ac87eeaf3111f6181e I made the `hal_mdihistory.py` translatable.
But this breaks the Gmoccapy translation.

If i remove
https://github.com/LinuxCNC/linuxcnc/blob/13eb99bec7c95904bf25be9ab5a2f28354e7f0a5/lib/python/gladevcp/hal_mdihistory.py#L77
the gmoccapy translation works but the mdi history is not translated. Is there a way to have both or is that a problem with the two different translation domains?

c-morley, 2022-02-05:
Why is the Gmoccapy translation different the the linuxcnc one? Or are they the same but just break each other?
Probably the linuxcnc one would be more right to support.

hansu, 2022-02-05:
The translation for gmoccapy is in a separate file. That's no problem. In the offsetpage widget it works with the differing translation domain 'linuxcnc'. But in that widget are only the strings of the glade file taken into account. Here is it about the strings of the python code. 
I think I must not use ` gettext.install("linuxcnc", localedir=LOCALEDIR) ` as it overwrites the line
https://github.com/LinuxCNC/linuxcnc/blob/dfe0260cd122719e8c5366b07c549e77cd724c0f/src/emc/usr_intf/gmoccapy/gmoccapy.py#L147
Maybe it's not possible to set another translation domain for an embedded widget code.

PR #1606 (2022-02-25): fix: translation of hal_mdihistory broke other translations
Sometimes less is more... :slightly_smiling_face: 

Gmoccapy translation in now working again and hal_mdihistory too.

fixes #1566
closed 2022-02-25 by c-morley