LinuxCNC/linuxcnc #1570: return of pmErrno that is a && b is somewhat too close to assembly - or wrong

smoe, 2022-02-05:
Hello,

I stumbled across

https://github.com/LinuxCNC/linuxcnc/blob/89a41a275ba34d6a6797898280964abd6336b812/src/libnml/posemath/_posemath.c#L1479

which I presume to mean that in case of an error in the left function invocation the second invocation should not be performed. But then it is unclear what error was exactly found. And it is difficult to read. In analogy to the many other analogous situations in the code, I suggest
```
int r1,r2
r1 = a(...);
if (!r1) r2= b(...);
return pmErrno:
```
Best,
Steffen

andypugh, 2022-02-06:
I don't think that is the case, as it seems that pmQuatQuatCompare and pmCartCartCompare each only return 0 or 1. 

pmErrno just seems like a temporary result variable in this case. In fact is seems a little odd to use it here as it might potentially clear an error value from it. In other parts of the file it seems to be passed-through or reset with a ternary, 

In fact, the more I look the more puzzled I get about it.

smoe, 2022-02-06:
Without the pmErrno it may have a chance to make some sense. You want to know if the two poses are the same for both the tran and the rot. Otherwise, there is always an error when the function returns 1? That line is from 2013, one is tempted to think  it does not do too much bad :)

```
$ grep -r PosePoseCompare src/
src/libnml/posemath/_posemath.c:int pmPosePoseCompare(PmPose const * const p1, PmPose const * const p2)
src/libnml/posemath/_posemath.c:        pmPrintError("Bad quaternion in pmPosePoseCompare\n");
src/libnml/posemath/posemath.cc:    return pmPosePoseCompare(&_p1, &_p2);
src/libnml/posemath/posemath.cc:    return !pmPosePoseCompare(&_p1, &_p2);
src/libnml/posemath/posemath.h:    extern int pmPosePoseCompare(PmPose const * const, PmPose const * const);
```
but it is the actual comparator for poses:
```
int operator ==(const PM_POSE &p1, const PM_POSE &p2)
{
    PmPose _p1, _p2;

    toPose(p1, &_p1);
    toPose(p2, &_p2);

    return pmPosePoseCompare(&_p1, &_p2);
}
```
which does just not check on the pmErrno.

closed 2022-02-06 by smoe