LinuxCNC/linuxcnc #1572: Add Vietnamese language on Weblate

vnwildman, 2022-02-06:
Hi,
Please add Vietnamese language on [Weblate](https://hosted.weblate.org/projects/linuxcnc/linuxcnc/). I will try translate it when I have free time.
Thanks,

silopolis, 2022-02-06:
Hi

Le dim. 6 févr. 2022 à 08:37, Trần Ngọc Quân ***@***.***> a
écrit :

> Please add Vietnamese language on Weblate
> <https://hosted.weblate.org/projects/linuxcnc/linuxcnc/>. I will try
> translate it when I have free time.
>

Done ! Find these here:
https://hosted.weblate.org/projects/linuxcnc/linuxcnc/vi/
https://hosted.weblate.org/projects/linuxcnc/gmocappy/vi/

Thanks in advance for your contributions 👍🙏

TY

SebKuzminsky, 2022-02-06:
Thanks for taking care of this @silopolis !

closed 2022-02-06 by SebKuzminsky