LinuxCNC/linuxcnc #1596: gladevcp hal_bar.py: AttributeError: 'NoneType' object has no attribute 'red_float'

arvidbrodin, 2022-02-12:
## Here are the steps I follow to reproduce the issue:

 1. Install Debian 11.2
 2. Clone linuxcnc from github (fae919506), build for uspace
 3. . scripts/rip-environment
 4. sudo apt install glade
 5. Run 'glade', Create new project
 6. Add GtkWindow (optionally: set ID to 'window1')
 7. Add GtkBox
 8. Add HAL_VBar
 9. (optional: set Packing->Expand to see the widget. Not necessary to produce error)
 
## This is what I expected to happen:

glade should be able to display the HAL_VBar.

## This is what happened instead:

Traceback (most recent call last):
  File "/home/arvidb/src/linuxcnc-dev/lib/python/gladevcp/hal_bar.py", line 304, in expose
    cr, (w, h), set_color, alpha = self._expose_prepare(widget)
  File "/home/arvidb/src/linuxcnc-dev/lib/python/gladevcp/hal_bar.py", line 151, in _expose_prepare
    set_color(self.bg_color)
  File "/home/arvidb/src/linuxcnc-dev/lib/python/gladevcp/hal_bar.py", line 136, in set_color
    return cr.set_source_rgba(c.red_float, c.green_float, c.blue_float, alpha)
AttributeError: '_ResultTuple' object has no attribute 'red_float'

Same error appears if the project is saved as my-panel.ui and ran with 'gladevcp my-panel.ui'. (Requires GtkWindow ID to be set to 'window1' above.)

## It worked properly before this:

Don't know, haven't tried older versions.

hansu, 2022-02-27:
Can you please check if  https://github.com/hansu/linuxcnc/commit/923710f796cb9a15381680d85bb621106d683df3 is working as intended?

arvidbrodin, 2022-02-27:
Hi, thanks for looking into this!

I tested your patch on both Debian 10.11 and 11.2 by cherry-picking your commit onto current master. The results are the same on both systems:

* Creating the widget in glade works as expected.
* Saving the .ui file and running it with gladevcp works as expected.
* However, reopening the .ui file in glade results in the following error, and undrawn widget:

```
Traceback (most recent call last):
  File "/home/arvidb/src/linuxcnc-dev/lib/python/gladevcp/hal_bar.py", line 311, in expose
    cr, (w, h), set_color, alpha = self._expose_prepare(widget)
  File "/home/arvidb/src/linuxcnc-dev/lib/python/gladevcp/hal_bar.py", line 158, in _expose_prepare
    set_color(self.bg_color)
  File "/home/arvidb/src/linuxcnc-dev/lib/python/gladevcp/hal_bar.py", line 143, in set_color
    return cr.set_source_rgba(c.red, c.green, c.blue, alpha)
AttributeError: 'NoneType' object has no attribute 'red'
```

hansu, 2022-02-28:
Okay, try this: https://github.com/hansu/linuxcnc/commit/8f93176795cfe48ad674595ee6b17fcaab8fe916

arvidbrodin, 2022-03-01:
A quick test today on 11.2 only: The error in my last post is fixed, but colors aren't displayed properly in glade: any Zone 0/Zone1 color but black is displayed as white. (Black is displayed as black.)

hansu, 2022-03-01:
Hmmm. Can you post your example to verify the colors?

arvidbrodin, 2022-03-01:
I'm not at my test computer(s) now, but in short I created a new project in glade, added a main window and a vertical hal_bar. I then used the controls in glade (the component's color pickers) to set the zone colors. I increased the value field in glade to see the bar.

The background of the bar is light gray I believe (I'm red-green color blind so might be mistaken on hues). The bar itself was black by default, both visibly and according to the color pickers. Any change of color in the color picker resulted in a pure white bar (and changing it back to black made it black again). If I set e.g. zone 0 to black and zone 1 to some other color, I got a black/white bar with a gradient between the fields. But always just pure black or pure white.

arvidbrodin, 2022-03-02:
That seems to have done the trick. Tested on both debian 10.11 and 11.2. Great work! Thanks!

PR #1639 (2022-03-06): gladevcp: fix color in VCP elements
In several VCP elements it should be
`color = Gdk.Color.parse('colorstring')[1]` 
instead of 
`color = Gdk.Color.parse('colorstring')` 
because `Gdk.Color.parse()` returns `(bool, Gdk.Color)`

fixes  #1596
closed 2022-03-06 by c-morley