LinuxCNC/linuxcnc #1599: Debian bullseye boost python issue

smoe, 2022-02-12:
## Here are the steps I follow to reproduce the issue:

 1. install Debian bullseye
 2. run linuxcnc
 3. press "OK" (cancel is fine)

## This is what I expected to happen:

no idea, it is the first time I started this

## This is what happened instead:
$ linuxcnc
LINUXCNC - 2.9.0~pre0
Machine configuration directory is '/home/moeller/linuxcnc/configs/sim.axis-1'
Machine configuration file is 'axis_mm.ini'
Starting LinuxCNC...
Found file(lib): /usr/share/linuxcnc/hallib/core_sim.hal
Note: Using POSIX non-realtime
Found file(lib): /usr/share/linuxcnc/hallib/sim_spindle_encoder.hal
Found file(lib): /usr/share/linuxcnc/hallib/axis_manualtoolchange.hal
Found file(lib): /usr/share/linuxcnc/hallib/simulated_home.hal
PYTHON: exception during 'this' export:
TypeError: 'Boost.Python.class' object is not iterable

The above exception was the direct cause of the following exception:

Traceback (most recent call last):

  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load

  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked

  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked

  File "<frozen importlib._bootstrap>", line 565, in module_from_spec

  File "<frozen importlib._bootstrap>", line 763, in create_module

SystemError: _PyEval_EvalFrameDefault returned a result with an error set

link (updating variable file): Datei oder Verzeichnis nicht gefunden
note: MAXV     max: 53.340 units/sec 3200.400 units/min
note: LJOG     max: 53.340 units/sec 3200.400 units/min
note: LJOG default: 30.480 units/sec 1828.800 units/min
note: jog_order='XYZ'
note: jog_invert=set()
PYTHON: exception during 'this' export:
TypeError: No Python class registered for C++ class Interp

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/OpenGL/latebind.py", line 43, in __call__
    return self._finalCall( *args, **named )
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/axis", line 3979, in <module>
    commands.set_view_p()
  File "/usr/bin/axis", line 2181, in set_view_p
    o.set_view_p()
  File "/usr/lib/python3/dist-packages/glnav.py", line 448, in set_view_p
    glRotateScene(self, 1.0, mid[0], mid[1], mid[2], 0, 0, 0, 0)
  File "/usr/lib/python3/dist-packages/glnav.py", line 116, in glRotateScene
    mat = glGetDoublev(GL_MODELVIEW_MATRIX)
  File "/usr/lib/python3/dist-packages/OpenGL/latebind.py", line 47, in __call__
    return self._finalCall( *args, **named )
  File "/usr/lib/python3/dist-packages/OpenGL/wrapper.py", line 668, in wrapperCall
    cArgs = tuple(calculate_cArgs( pyArgs ))
  File "/usr/lib/python3/dist-packages/OpenGL/wrapper.py", line 471, in calculate_cArgs
    yield converter( pyArgs, index, self )
  File "/usr/lib/python3/dist-packages/OpenGL/converters.py", line 251, in __call__
    return super( SizedOutputOrInput,self ).__call__( pyArgs, index, baseOperation )
  File "/usr/lib/python3/dist-packages/OpenGL/converters.py", line 196, in __call__
    return self.arrayType.zeros( self.getSize(pyArgs) )
  File "/usr/lib/python3/dist-packages/OpenGL/arrays/arraydatatype.py", line 177, in zeros
    return cls.returnHandler().zeros( dims, typeCode or cls.typeConstant )
  File "/usr/lib/python3/dist-packages/OpenGL/arrays/arraydatatype.py", line 80, in get_output_handler
    self.output_handler = self.handler_by_plugin_name( preferred )
  File "/usr/lib/python3/dist-packages/OpenGL/arrays/arraydatatype.py", line 67, in handler_by_plugin_name
    return plugin.load()
  File "/usr/lib/python3/dist-packages/OpenGL/plugins.py", line 16, in load
    return importByName( self.import_path )
  File "/usr/lib/python3/dist-packages/OpenGL/plugins.py", line 38, in importByName
    module = __import__( ".".join(moduleName), {}, {}, moduleName)
  File "/usr/lib/python3/dist-packages/OpenGL/arrays/numpymodule.py", line 12, in <module>
    import numpy
  File "/usr/lib/python3/dist-packages/numpy/__init__.py", line 150, in <module>
    from . import core
  File "/usr/lib/python3/dist-packages/numpy/core/__init__.py", line 22, in <module>
    from . import multiarray
  File "/usr/lib/python3/dist-packages/numpy/core/multiarray.py", line 12, in <module>
    from . import overrides
  File "/usr/lib/python3/dist-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
TypeError: ("'Boost.Python.class' object is not iterable", 'Failure in cConverter <OpenGL.converters.SizedOutputOrInput object at 0x7fda4a4b9cc0>', (GL_MODELVIEW_MATRIX, <object object at 0x7fda4afecd40>), 1, <OpenGL.platform.baseplatform.glGetDoublev object at 0x7fda4a48c280>)
Shutting down and cleaning up LinuxCNC...
task: 1763 cycles, min=0.000007, max=0.005435, avg=0.001065, 0 latency excursions (> 10x expected cycle time of 0.001000s)
Note: Using POSIX non-realtime

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):

No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux bookworm/sid
Release:        unstable
Codename:       sid

 * I am using this kernel version (shown by `uname -a`):

Linux x230 5.15.0-3-amd64 #1 SMP Debian 5.15.15-2 (2022-01-30) x86_64 GNU/Linux

 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [X] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org

 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):

A recent master, the bug has been reported also for the RPi.

 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):

Just the linuxcnc application started from the home directory - KDE.

 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 

Nothing attached. Just a blank laptop, not even running a realtime module.

 * Other info that possibly is useful:
 ```
 $ cat /home/moeller/linuxcnc_debug.txt
508317
508359
Stopping realtime threads
Unloading hal components
```
```
$ cat /home/moeller/linuxcnc_print.txt
RUN_IN_PLACE=no
LINUXCNC_DIR=
LINUXCNC_BIN_DIR=/usr/bin
LINUXCNC_TCL_DIR=/usr/lib/tcltk/linuxcnc
LINUXCNC_SCRIPT_DIR=
LINUXCNC_RTLIB_DIR=/usr/lib/linuxcnc/modules
LINUXCNC_CONFIG_DIR=
LINUXCNC_LANG_DIR=/usr/lib/tcltk/linuxcnc/msgs
INIVAR=inivar
HALCMD=halcmd
LINUXCNC_EMCSH=/usr/bin/wish8.6
INIFILE=/home/moeller/linuxcnc/configs/sim.axis-1/axis_mm.ini
VERSION=1.1
PARAMETER_FILE=sim_mm.var
TASK=milltask
HALUI=halui
DISPLAY=axis
COORDINATES=X Y Z
KINEMATICS=trivkins
Starting LinuxCNC server program: linuxcncsvr
Loading Real Time OS, RTAPI, and HAL_LIB modules
Starting LinuxCNC IO program: io
Starting HAL User Interface program: halui
Starting TASK program: milltask
Starting DISPLAY program: axis
Removing HAL_LIB, RTAPI, and Real Time OS modules
Removing NML shared memory segments
```

SebKuzminsky, 2022-02-13:
Hmm, are you running bullseye (like it says at the top) or bookworm/sid (like it says at the bottom?  From your kernel i'm guessing bookworm/sid.

I can not reproduce this on my bullseye machine.

smoe, 2022-02-13:
You are right. My laptop runs unstable. That is interesting observation. It is libboost-python1.74-dev and both python 3.9 and python 3.10 for me. Maybe the error is elsewhere. But this also means that we should continue with the upload to Debian and then figure this out with a larger user base. The problem is that any such failure would be the first impression for the folks on unstable.

anubhav-cs, 2022-02-13:
I just happened to run across this issue while looking to debug a Boost C++ library. I was able to reproduce this for my application - it arises when I import a boost python extension in `python3.9` which was built using `libboostpython310`. As such, it works fine with `python3.10`. I rebuilt the module using `libboostpython39` and now it works fine with both 3.9 and 3.10.

smoe, 2022-02-13:
@anubhav-cs Thank you for your detective work and for spreading the news. I do not yet know what the consequence for LinuxCNC should be.

smoe, 2022-02-20:
I fail to get it to run, still.

smoe, 2022-02-25:
Above PR got a nice comment by @dwrobel pointing to Python 3.10 only systems for which there is no problem.

smoe, 2022-12-25:
The error surfaces again, this time in reaction to an OpenGL context.
```
note: LJOG     max: 53.340 units/sec 3200.400 units/min
note: LJOG default: 30.480 units/sec 1828.800 units/min
note: jog_order='XYZ'
note: jog_invert=set()
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/OpenGL/latebind.py", line 43, in __call__
    return self._finalCall( *args, **named )
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/axis", line 3964, in <module>
    get_coordinate_font(vars.dro_large_font.get())
  File "/usr/bin/axis", line 3842, in get_coordinate_font
    glnav.use_pango_font(coordinate_font, 0, 128)
  File "/usr/lib/python3/dist-packages/glnav.py", line 65, in use_pango_font
    glBitmap(0, 0, 0, 0, 0, h-d, ''.encode())
  File "/usr/lib/python3/dist-packages/OpenGL/latebind.py", line 47, in __call__
    return self._finalCall( *args, **named )
  File "/usr/lib/python3/dist-packages/OpenGL/wrapper.py", line 700, in wrapperCall
    raise err
  File "/usr/lib/python3/dist-packages/OpenGL/wrapper.py", line 693, in wrapperCall
    result = wrappedOperation( *cArguments )
  File "/usr/lib/python3/dist-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
        err = 1285,
        description = b'out of memory',
        baseOperation = glBitmap,
        pyArgs = (0, 0, 0, 0, 0, 14.0, b''),
        cArgs = (0, 0, 0, 0, 0, 14.0, b''),
        cArguments = (0, 0, 0, 0, 0, 14.0, b'')
)
Shutting down and cleaning up LinuxCNC...
task: 1849 cycles, min=0.000011, max=0.005185, avg=0.001068, 0 latency excursions (> 10x expected cycle time of 0.001000s)
Note: Using POSIX realtime
LinuxCNC terminated with an error.  You can find more information in the log:
    /home/moeller/linuxcnc_debug.txt
and
    /home/moeller/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
```
So, the display of the error is what triggers the bigger error message. Ideas anyone?

This is with Debian unstable, both the official Debian package and a freshly rebuilt one.

smoe, 2022-12-25:
```
moeller@x230:~/Github/linuxcnc$ cat ~/linuxcnc_print.txt 
RUN_IN_PLACE=no
LINUXCNC_DIR=
LINUXCNC_BIN_DIR=/usr/bin
LINUXCNC_TCL_DIR=/usr/lib/tcltk/linuxcnc
LINUXCNC_SCRIPT_DIR=
LINUXCNC_RTLIB_DIR=/usr/lib/linuxcnc/modules
LINUXCNC_CONFIG_DIR=
LINUXCNC_LANG_DIR=/usr/lib/tcltk/linuxcnc/msgs
INIVAR=inivar
HALCMD=halcmd
LINUXCNC_EMCSH=/usr/bin/wish8.6
INIFILE=/home/moeller/linuxcnc/configs/sim.axis/axis_mm.ini
VERSION=1.1
PARAMETER_FILE=sim_mm.var
TPMOD=
HOMEMOD=
TASK=milltask
HALUI=halui
DISPLAY=axis
COORDINATES=X Y Z
KINEMATICS=trivkins
Starting LinuxCNC server program: linuxcncsvr
Loading Real Time OS, RTAPI, and HAL_LIB modules
Starting LinuxCNC IO program: io
Starting HAL User Interface program: halui
Starting TASK program: milltask
Starting DISPLAY program: axis
Removing HAL_LIB, RTAPI, and Real Time OS modules
Removing NML shared memory segments
moeller@x230:~/Github/linuxcnc$ cat ~/linuxcnc_debug.txt 
91930
91976
Stopping realtime threads
Unloading hal components
```

smoe, 2022-12-26:
I could apparently fix this by
```
diff --git a/lib/python/glnav.py b/lib/python/glnav.py
index 4c3ffe4279..392d7a061c 100644
--- a/lib/python/glnav.py
+++ b/lib/python/glnav.py
@@ -62,7 +62,7 @@ def use_pango_font(font, start, count, will_call_prepost=False):
         context.restore()
         w, h = int(w / Pango.SCALE), int(h / Pango.SCALE)
         glNewList(base+i, GL_COMPILE)
-        glBitmap(0, 0, 0, 0, 0, h-d, ''.encode())
+        #TEST#glBitmap(0, 0, 0, 0, 0, h-d, ''.encode())
         #glDrawPixels(0, 0, 0, 0, 0, h-d, '');
         if not will_call_prepost:
             pango_font_pre()
@@ -73,7 +73,7 @@ def use_pango_font(font, start, count, will_call_prepost=False):
             except Exception as e:
                 print("glnav Exception ",e)
 
-        glBitmap(0, 0, 0, 0, w, -h+d, ''.encode())
+        #TEST#glBitmap(0, 0, 0, 0, w, -h+d, ''.encode())
         if not will_call_prepost:
             pango_font_post()
         glEndList()
```
but do not  really know what I am doing (not doing) with that patch.

smoe, 2022-12-26:
@hansu  could you please have a look? This fixes both AXIS and GMOCCAPY for me on Debian unstable and likely also the bookworm-based LiveCD.

hansu, 2022-12-28:
I am a bit confused. Does this occur when you running Bookworm or Bullseye?

smoe, 2022-12-28:
I have not tried Bullseye. But I get that error with the self-created bookworm-based LiveCD (or an error very much like it) and what I reported here was with my Debian unstable (sid, which is just 5 days ahead of bookworm, typically) laptop.

hansu, 2022-12-28:
I encounter no problems running AXIS on a fresh Bookworm install.
But there are problems 
- building LinuxCNC
- running gmoccapy

 see https://forum.linuxcnc.org/9-installing-linuxcnc/47657-linux-cnc-errors-when-starting-gmoccapy

smoe, 2022-12-28:
Judging from the logs, is that forum-reported error prior or post the problem I had reported? I had the hunch that it looks like prior but that cannot be since AXIS is working for you. Hm.

hansu, 2022-12-28:
Not sure if this is related to you OpenGL error.

But the error I get when running `configure` seems related to https://github.com/LinuxCNC/linuxcnc/pull/1619

smoe, 2022-12-28:
Hm. Python 11 is among the versions checked. We should at some point skype and share screens. That live usb distro is another means that I thought we should use more for such debugging.

petterreinholdtsen, 2023-01-06:
I ran into this error to.  I can trigger it every time on my updated
Bookworm installation.

Here is the relevant part of the output from linuxcnc:

xhc-hb04.tcl: Using existing outpin signame: spindle-vel-cmd-rps-abs
note: MAXV     max: 500.000 units/sec 30000.000 units/min
note: LJOG     max: 500.000 units/sec 30000.000 units/min
note: LJOG default: 500.000 units/sec 30000.000 units/min
note: jog_order='XYZ'
note: jog_invert=set()
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/OpenGL/latebind.py", line 43, in __call__
    return self._finalCall( *args, **named )
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/axis", line 3960, in <module>
    get_coordinate_font(vars.dro_large_font.get())
  File "/usr/bin/axis", line 3838, in get_coordinate_font
    glnav.use_pango_font(coordinate_font, 0, 128)
  File "/usr/lib/python3/dist-packages/glnav.py", line 65, in use_pango_font
    glBitmap(0, 0, 0, 0, 0, h-d, ''.encode())
  File "/usr/lib/python3/dist-packages/OpenGL/latebind.py", line 47, in __call__
    return self._finalCall( *args, **named )
  File "/usr/lib/python3/dist-packages/OpenGL/wrapper.py", line 700, in wrapperCall
    raise err
  File "/usr/lib/python3/dist-packages/OpenGL/wrapper.py", line 693, in wrapperCall
    result = wrappedOperation( *cArguments )
  File "/usr/lib/python3/dist-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
        err = 1285,
        description = b'out of memory',
        baseOperation = glBitmap,
        pyArgs = (0, 0, 0, 0, 0, 14.0, b''),
        cArgs = (0, 0, 0, 0, 0, 14.0, b''),
        cArguments = (0, 0, 0, 0, 0, 14.0, b'')
)
Shutting down and cleaning up LinuxCNC...

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2023-01-06:
[Petter Reinholdtsen]
> I ran into this error to.  I can trigger it every time on my updated
> Bookworm installation.

To rule out memory fragmentation, I rebooted the machine in question to
see if the error went away, and it stayed the same.

Sebastian mentioned on IRC that this might be related to the existance
of two different versions of python installed.  I got both python3.10
and python3.11 installed.  Trying to remove either of them will also
remove linuxcnc-uspace (and others like freecad), so just removing the
"redundant" python version do not seem like the correct solution.  The
output of 'apt-mark showmanual' in the machine in question list these
packages: "acpid apt-file emacs etckeeper firmware-amd-graphics
firmware-linux-free firmware-misc-nonfree firmware-realtek freecad
ifupdown iputils-ping isenkram-cli libnss-myhostname
linux-image-rt-amd64 linuxcnc-uspace locales lsof lvm2 memtest86+ nano
needrestart nullmailer openssh-client openssh-server popularity-contest
procps sitesummary-client sudo task-english task-xfce-desktop tmux
unattended-upgrades vim-tiny".  I hope others can reproduce the problem
in Bookworm by installing the same set of packages.

-- 
Happy hacking
Petter Reinholdtsen

ve7it, 2023-01-07:
I had a similar opengl out of memory error on the linuxcnc-sim package 
for arch linux. It turns out if you have intel builtin graphics, they 
have broken the latest libraries. I fixed my system by installing 
mesa-amber and lib32-mesa-amber packages to support my 2013 vintage i7 box.

  This may help with the opengl out of memory errors on intel graphics 
systems.

see below for discussion:
https://aur.archlinux.org/packages/linuxcnc-sim

cheers
Lawrence VE7IT

petterreinholdtsen, 2023-01-07:
[Lawrence Glaister]
> I had a similar opengl out of memory error on the linuxcnc-sim package
> for arch linux. It turns out if you have intel builtin graphics, they
> have broken the latest libraries. I fixed my system by installing
> mesa-amber and lib32-mesa-amber packages to support my 2013 vintage i7
> box.

I am on Debian and indeed lspci show this VGA entry on my HP ProDesk 600
G2 SFF from 2015:

 00:02.0 VGA compatible controller: Intel Corporation HD Graphics 530 (rev 06)

I find no mesa-amber related package from Debian, but find it in
<URL: https://archlinux.org/packages/extra/x86_64/mesa-amber/ >.  I
guess it might have a different name in Debian.p

I guess this should be reported as a bug against some driver package.
Just need to track down which. :)

> see below for discussion:
> https://aur.archlinux.org/packages/linuxcnc-sim

Thank you a lot of this pointer.

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2023-01-07:
[Petter Reinholdtsen]
> I am on Debian and indeed lspci show this VGA entry on my HP ProDesk 600
> G2 SFF from 2015:

In case this was related to the amount of video memory, I tested
visiting the BIOS setup and changing the amount reserved from 32 MB to
512 MB, without any change.

-- 
Happy hacking
Petter Reinholdtsen

smoe, 2023-01-08:
@petterreinholdtsen , can you confirm that commenting out those two lines as described in https://github.com/LinuxCNC/linuxcnc/issues/1599#issuecomment-1365303941 does fix it?

petterreinholdtsen, 2023-01-09:
[Steffen Möller]
> @petterreinholdtsen , can you confirm that commenting out those two
> lines as described in
> https://github.com/LinuxCNC/linuxcnc/issues/1599#issuecomment-1365303941
> does fix it?

Not really.  I have no idea what those lines do nor what the effect is
from commenting them out.  Do you know this?

Are you just talking about avoiding the crash when you say 'fix it'?  I
did not test so far, as I want to understand where the error is.  I
guess some intel related X driver, but am unsure.

-- 
Happy hacking
Petter Reinholdtsen

smoe, 2023-01-09:
The glBitmap expects a pointer to a bitmap https://registry.khronos.org/OpenGL-Refpages/gl2.1/xhtml/glBitmap.xml and that last argument apparently fails to get converted. The max effect of commenting out that invocation that I see is that the bitmap is not shown. Over here everthing looked just fine without those bitmaps. If that is indeed the problem then we can likely find a way to derive a minimal program to discuss with others outside of LinuxCNC, so I would very much like to get a confirmation that this is indeed the culprit and nothing else.

 

Gesendet: Montag, 09. Januar 2023 um 10:47 Uhr
Von: "petterreinholdtsen" ***@***.***>
An: "LinuxCNC/linuxcnc" ***@***.***>
Cc: "Steffen Möller" ***@***.***>, "State change" ***@***.***>
Betreff: Re: [LinuxCNC/linuxcnc] OpenGL error messages at startup cause error messages (Issue #1599)

 

[Steffen Möller]
> @petterreinholdtsen , can you confirm that commenting out those two
> lines as described in
> https://github.com/LinuxCNC/linuxcnc/issues/1599#issuecomment-1365303941
> does fix it?

Not really. I have no idea what those lines do nor what the effect is
from commenting them out. Do you know this?

Are you just talking about avoiding the crash when you say 'fix it'? I
did not test so far, as I want to understand where the error is. I
guess some intel related X driver, but am unsure.

 

 

Message ID: ***@***.***>

antbarney, 2023-01-10:
I have also started hitting this recently, on my system a cheap Intel Celeron based Mini PC running bookworm + LCNC 2.9

I did comment out those 2 qlBitmap references and this allows Axis to load successfully. This system was working up until earlier last week, so my guess is there was some update that triggered things, but I did not pay attention when doing updates to know what was recently updated/installed.

> The glBitmap expects a pointer to a bitmap https://registry.khronos.org/OpenGL-Refpages/gl2.1/xhtml/glBitmap.xml and that last argument apparently fails to get converted. The max effect of commenting out that invocation that I see is that the bitmap is not shown. Over here everthing looked just fine without those bitmaps. If that is indeed the problem then we can likely find a way to derive a minimal program to discuss with others outside of LinuxCNC, so I would very much like to get a confirmation that this is indeed the culprit and nothing else.   Gesendet: Montag, 09. Januar 2023 um 10:47 Uhr Von: "petterreinholdtsen" ***@***.***> An: "LinuxCNC/linuxcnc" ***@***.***> Cc: "Steffen Möller" ***@***.***>, "State change" ***@***.***> Betreff: Re: [LinuxCNC/linuxcnc] OpenGL error messages at startup cause error messages (Issue #1599)   [Steffen Möller]
> @petterreinholdtsen , can you confirm that commenting out those two lines as described in [#1599 (comment)](https://github.com/LinuxCNC/linuxcnc/issues/1599#issuecomment-1365303941) does fix it?
> Not really. I have no idea what those lines do nor what the effect is from commenting them out. Do you know this? Are you just talking about avoiding the crash when you say 'fix it'? I did not test so far, as I want to understand where the error is. I guess some intel related X driver, but am unsure.     Message ID: ***@***.***>

smoe, 2023-01-10:
The error message is also reported on https://stackoverflow.com/questions/56767373/how-to-blit-a-png-image-as-an-image-overlay-in-pyopengl but I fail to understand the (the relevance of the) answer.

petterreinholdtsen, 2023-01-11:
[Anthony Barney]
> I have also started hitting this recently, on my system a cheap
> Intel Celeron based Mini PC running bookworm + LCNC 2.9

Perhaps you can find likely candidates causing this by looking in /var/log/dpkg.log
or /var/log/apt/history.log?  My test machine had been left alone for
too long for me to find any useful information in those files.
-- 
Happy hacking
Petter Reinholdtsen

smoe, 2023-01-11:
It is some unknown difference between bookworm and bullseye. I have just brought https://github.com/LinuxCNC/linuxcnc/pull/2259 on its way, suggesting to introduce FIXMEs and see how it goes.

jethornton, 2023-01-11:
I also started having this problem on bookworm after an update the other day.

jethornton, 2023-01-11:
I updated my repo, checked out 2.9, commented out the two calls to glBitmap, built a deb then installed it and now on Bookworm Axis loads.

antbarney, 2023-01-12:
what I do notice is that, yes this gets Axis to start and run the DRO on the preview tab looks to be broken. 
[image]

antbarney, 2023-01-12:
> [Anthony Barney]
> I have also started hitting this recently, on my system a cheap Intel Celeron based Mini PC running bookworm + LCNC 2.9
> Perhaps you can find likely candidates causing this by [...]

this machine was roughly in the same boat, there was a large number of updates that got installed at once and can't find anything useful in those files either.

NTULINUX, 2023-01-13:
I posted this to the mailing list, didn't see this issue until now. Same problem here on Gentoo:

LINUXCNC - 2.10.0~pre0
Machine configuration directory is '/home/ntu/linuxcnc/configs/sim.axis-2'
Machine configuration file is 'axis.ini'
Starting LinuxCNC...
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
Note: Using POSIX non-realtime
Found file(lib): /usr/share/linuxcnc/hallib/core_sim.hal
Found file(lib): /usr/share/linuxcnc/hallib/sim_spindle_encoder.hal
Found file(lib): /usr/share/linuxcnc/hallib/axis_manualtoolchange.hal
Found file(lib): /usr/share/linuxcnc/hallib/simulated_home.hal
Found file(lib): /usr/share/linuxcnc/hallib/check_xyz_constraints.hal
note: MAXV     max: 5.000 units/sec 300.000 units/min
note: LJOG     max: 5.000 units/sec 300.000 units/min
note: LJOG default: 0.250 units/sec 15.000 units/min
note: jog_order='XYZ'
note: jog_invert=set()
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/OpenGL/latebind.py", line 43, in __call__
    return self._finalCall( *args, **named )
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/axis", line 3964, in <module>
    get_coordinate_font(vars.dro_large_font.get())
  File "/usr/bin/axis", line 3842, in get_coordinate_font
    glnav.use_pango_font(coordinate_font, 0, 128)
  File "/usr/lib/python3.10/site-packages/glnav.py", line 65, in use_pango_font
    glBitmap(0, 0, 0, 0, 0, h-d, ''.encode())
  File "/usr/lib/python3.10/site-packages/OpenGL/latebind.py", line 47, in __call__
    return self._finalCall( *args, **named )
  File "/usr/lib/python3.10/site-packages/OpenGL/wrapper.py", line 700, in wrapperCall
    raise err
  File "/usr/lib/python3.10/site-packages/OpenGL/wrapper.py", line 693, in wrapperCall
    result = wrappedOperation( *cArguments )
  File "/usr/lib/python3.10/site-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
  File "/usr/lib/python3.10/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
        err = 1285,
        description = b'out of memory',
        baseOperation = glBitmap,
        pyArgs = (0, 0, 0, 0, 0, 13.0, b''),
        cArgs = (0, 0, 0, 0, 0, 13.0, b''),
        cArguments = (0, 0, 0, 0, 0, 13.0, b'')
)
Shutting down and cleaning up LinuxCNC...
task: 477 cycles, min=0.000004, max=0.001061, avg=0.001017, 0 latency excursions (> 10x expected cycle time of 0.001000s)
Note: Using POSIX non-realtime
LinuxCNC terminated with an error.  You can find more information in the log:

rodw-au, 2023-01-14:
This should be resolved by #2264  if you are running on the Default Debian Desktop which runs on Wayland.
Alternatively use XFCE or select Xorg on the login screen
Works on my a couple of systems and my Chromebook with Debian Bullseye which has been broken for over 12 months. It only has Wayland.

antbarney, 2023-01-16:
I have tried the fixes in #2264 by downgrading python3-opengl to 3.1.5 well I am sure that is fixing the problem noted in that issue which is related to the missing getGLUTFontPointer  functions. it does not appear to fix/help with the "out of memory" error I am others have been seeing.

smoe, 2023-01-16:
I'll prepare a bug report on Debian for python3-opengl that fails to find the freeglut library because of what I understand to be the missing symbolic link. I have minimal script for that. Is there another minimal script for the "out of memory" error?

antbarney, 2023-01-16:
> I'll prepare a bug report on Debian for python3-opengl that fails to find the freeglut library because of what I understand to be the missing symbolic link. I have minimal script for that. Is there another minimal script for the "out of memory" error?

Not that I have found yet. Only "work around" is to comment out the glBitmap calls in glnav.py but that breaks the DRO on the preview panel in Axis.

petterreinholdtsen, 2023-01-28:
[Anthony Barney]
> Not that I have found yet. Only "work around" is to comment out the
> glBitmap calls in glnav.py but that breaks the DRO on the preview
> panel in Axis.

I tried downgrading python3-opengl to 3.1.5+dfsg-3 and installing the
needed freeglut3 library (used version 2.8.1-6 from snapshot), as well
as setting the PYOPENGL_PLATFORM=x11 environment variable, on my test
Bookworm machine.  Sadly the axis GUI still fail in the call to

  glnav.use_pango_font(coordinate_font, 0, 128)

with this error:

  OpenGL.error.GLError: GLError(
   err = 1285,
   description = b'out of memory',
   baseOperation = glBitmap,
   pyArgs = (0, 0, 0, 0, 0, 14.0, b''),
   cArgs = (0, 0, 0, 0, 0, 14.0, b''),
   cArguments = (0, 0, 0, 0, 0, 14.0, b'')
  )

This make me believe there are several OpenGL issues at the moment, and
this one (#1599) must be different from #2264.

-- 
Happy hacking
Petter Reinholdtsen

SebKuzminsky, 2023-02-08:
I think this issue talks about a number of problems, but the OpenGL "glBitmap out of memory" problem should be fixed by @jepler's #2328.  That fix is in 2.9 and master/2.10 now.

I'm closing this issue as fixed, please open new issues if any problems remain.

closed 2023-02-08 by SebKuzminsky

rodw-au, 2023-02-09:
Has anyone checked  a Gmocappy sim  works?
I am getting this error on a RIP build this morning on Bullseye with wayland. I'm sure it is broken on Bookworm to.
`Traceback (most recent call last):
  File "/home/rod/linuxcnc-dev/lib/python/gladevcp/hal_gremlin.py", line 277, in realize
    gremlin.Gremlin.realize(self, widget)
  File "/home/rod/linuxcnc-dev/lib/python/rs274/glcanon.py", line 395, in inner
    self.activate()
  File "/home/rod/linuxcnc-dev/lib/python/gremlin.py", line 224, in activate
    if(not GLX.glXMakeCurrent(self.xdisplay, self.xwindow_id, self.context)):
  File "/usr/lib/python3/dist-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
ctypes.ArgumentError: argument 2: <class 'TypeError'>: wrong type`

Following this error, Gmocappy displays a number of error windows before finally displaying a broken screen.
gremlin.py appears to be the culprit.
A number of other messages display showing for example
`73: DeprecationWarning: Gtk.ColorButton.set_color is deprecated`

SebKuzminsky, 2023-02-09:
The following sim configs currently fail to start in Bookworm (with X11, not Wayland), according to <https://github.com/SebKuzminsky/linuxcnc-config-test>:

```
222 configs, 203 passed, 18 failed
Failed configs:
FAIL: axis/profile_axis.ini
FAIL: axis/remap/rack-toolchange/racktoolchange.ini
FAIL: axis/remap/manual-toolchange-with-tool-length-switch/manualtoolchange.ini
FAIL: craftsman/craftsmancnc.ini
FAIL: gscreen/gscreen_custom/9axis.ini
FAIL: gscreen/gscreen_custom/gaxis_no_plot.ini
FAIL: gscreen/gscreen_custom/gscreen_gaxis.ini
FAIL: gscreen/gscreen_custom/industrial.ini
FAIL: gscreen/gscreen_custom/tester_metric.ini
FAIL: gscreen/gscreen_custom/tester.ini
FAIL: gscreen/industrial_lathe_wear/industrial_lathe_wear.ini
FAIL: gscreen/silverdragon/silverdragon.ini
FAIL: gscreen/spartan/spartan.ini
FAIL: qtdragon/qtdragon_multi_joint/qtdragon_xyyz.ini
FAIL: qtvcp_screens/qtdragon/qtdragon_xyyz.ini
FAIL: qtvcp_screens/qtvcp_custom/qtdefault_ngc.ini
FAIL: qtvcp_screens/qtvcp_experimental/qtdefault_ngc.ini
FAIL: qtvcp_screens/woodpecker/woodpecker_xyzya.ini
```

Note: this test does not detect failure of configs that start but throw up error windows instead of functional GUIs.

hansu, 2023-02-09:
>Has anyone checked a Gmocappy sim works?

I can run Gmoccapy from a RIP build on Bookworm with Wayland (with freeglut3 installed manually)

jepler, 2023-02-10:
I tried one of the gscreens, it's not opengl
```
gtk-builder-error-quark: /home/jepler/src/linuxcnc/share/gscreen/skins/9_axis/9_axis.glade:2778:51 Invalid property: GtkStatusbar.has_resize_grip (11)
**** Gscreen GLADE ERROR:    With main screen xml file: /home/jepler/src/linuxcnc/share/gscreen/skins/9_axis/9_axis.glade
```
seems to be the fatal error.

jepler, 2023-02-10:
.. the statusbar grip functionality seems to be removed in gtk3, it was present in gtk2.

rodw-au, 2023-02-10:
Doesn't the comment from Hans mean freeglut3 should be added to the dependencies in bookworm?

phillc54, 2023-02-10:
I think the use of glut was removed [here](https://github.com/LinuxCNC/linuxcnc/commit/48c6018)

SebKuzminsky, 2023-02-10:
That's right, LinuxCNC doesn't use GLUT any more.

Also, freeglut is not in Bookworm, it's transitioned to libglut (which we also don't need).
<https://packages.debian.org/freeglut>

rodw-au, 2023-02-10:
So if gmocappy is broken in bookworm because it's not been updated, should it even be included in bookworm? It will be a while before I can test it in bookworm. We can't send broken code out to the masses

SebKuzminsky, 2023-02-10:
There's a bunch of broken stuff, and I agree with you, they should ideally be fixed, but failing that they should not be in the package.  PRs gladly accepted.

smoe, 2023-02-10:
I am very happy to hear that we no longer have that dependency on GLUT. I just grepped for it and nothing popped up. Happy.

To make sure we are all on the same page, and have me recreate a new .iso for it, would you mind to upload a new version of the Debian package? That soft freeze the day after tomorrow would be another reason :o/

petterreinholdtsen, 2023-02-10:
[Steffen Möller]
> would you mind to upload a new version of the Debian package? That
> soft freeze the day after tomorrow would be another reason :o/

From #debian-devel:

  <pere> uploaded.

See <URL: https://tracker.debian.org/pkg/linuxcnc >
-- 
Happy hacking
Petter Reinholdtsen

hansu, 2023-02-12:
I tried again on a fresh Bookworm without GLUT and I can confirm that Gmoccapy runs under X11 (RIP).
But still problems on Wayland:
```
Xlib.error.DisplayConnectionError: Can't connect to display ":0": b'Authorization required, but no authorization protocol specified\n'
```

After `sudo xhost +`

I get

 ```
 File "/home/cnc/linuxcnc/linuxcnc-2.9/lib/python/gremlin.py", line 244, in reshape
    self.xwindow_id = GdkX11.X11Window.get_xid(widget.get_window())
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: argument self: Expected GdkX11.X11Window, but got __gi__.GdkWaylandWindow

```

I only wonder why gremlin is working under Axis. I thought it's the same code used there.

SebKuzminsky, 2023-02-12:
@hansu, does @JetForMe's suggestion of `GDK_BACKEND=x11` help?  <https://github.com/LinuxCNC/linuxcnc/issues/2264#issuecomment-1411221972>

hansu, 2023-02-12:
Yes `export GDK_BACKEND=x11` does the trick, great! Sorry, I lost a bit track on all these OpenGL stuff.
Yes let's add it there!

jepler, 2023-02-13:
Since gtk cannot create a compatibility context we need the x11 backend for now too create it via glx calls. Major updates are needed to gremlin before it works with "modern" opengl. I support adding the environment variable workaround unconditionally though I'm not fully aware of the downsides (I don't use Wayland anywhere yet)