LinuxCNC/linuxcnc #1608: ModuleNotFoundError: No module named 'gobject'

zhouguanxing, 2022-02-14:
Dears, after my build the linuxcnc successfuly by setting python3.6 (./configure --with-realtime=uspace --with-python=/usr/bin/python3).
then,  run "cd script",
then, run "./linuxcnc":
then, choose "qtvcp_screens->qtdefault",
now, cause errors:
"Traceback (most recent call last):
  File "/home/zgx/00_src/00_linuxcnc/linuxcnc-ms-v2.9.0-1.0.4-pre0/bin/qtvcp", line 12, in <module>
    from qtvcp.core import Status, Info, QComponent, Path
  File "/home/zgx/00_src/00_linuxcnc/linuxcnc-ms-v2.9.0-1.0.4-pre0/lib/python/qtvcp/core.py", line 5, in <module>
    import gobject
ModuleNotFoundError: No module named 'gobject'
Shutting down and cleaning up LinuxCNC...
task: 176 cycles, min=0.000032, max=0.001987, avg=0.001119, 0 latency excursions (> 10x expected cycle time of 0.001000s)
Note: Using POSIX non-realtime
LinuxCNC terminated with an error.  You can find more information in the log:
    /home/zgx/linuxcnc_debug.txt
and
    /home/zgx/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal"
Now, if i use "pip install gobject", then another issue was caused:
"Traceback (most recent call last):
  File "/home/zgx/00_src/00_linuxcnc/linuxcnc-ms-v2.9.0-1.0.4-pre0/bin/qtvcp", line 12, in <module>
    from qtvcp.core import Status, Info, QComponent, Path
  File "/home/zgx/00_src/00_linuxcnc/linuxcnc-ms-v2.9.0-1.0.4-pre0/lib/python/qtvcp/core.py", line 9, in <module>
    from hal_glib import GStat
  File "/home/zgx/00_src/00_linuxcnc/linuxcnc-ms-v2.9.0-1.0.4-pre0/lib/python/hal_glib.py", line 19, in <module>
    class GPin(gobject.GObject, hal.Pin):
AttributeError: module 'gobject' has no attribute 'GObject'
Shutting down and cleaning up LinuxCNC...
task: 188 cycles, min=0.000031, max=0.001744, avg=0.001097, 0 latency excursions (> 10x expected cycle time of 0.001000s)
Note: Using POSIX non-realtime
LinuxCNC terminated with an error.  You can find more information in the log:
    /home/zgx/linuxcnc_debug.txt
and
    /home/zgx/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
",
so, i think the module of gobject is not which install by pip.
How can i to solve this issue, any sugestions?
Thank you very much!

hansu, 2022-02-14:
It seems that you are not building the current master branch.
Because line 5 does not contain `import gobject` any longer, only on 2.8 branch.
And you don't need `--with-python=/usr/bin/python3` as master is always building with Python 3.

zhouguanxing, 2022-02-14:
> 

> It seems that you are not building the current master branch. Because line 5 does not contain `import gobject` any longer, only on 2.8 branch. And you don't need `--with-python=/usr/bin/python3` as master is always building with Python 3.
> 
> And after make you should run ` . scripts/rip-environment` from the main directory (or `. ../scripts/rip-environment` from source dir) and then just type `linuxcnc `in this console
Dear hansu,
Thank you very much for you suggesion.
Now, I can work well after pull the master branch.
Thanks again!

closed 2022-02-15 by zhouguanxing