LinuxCNC/linuxcnc #1618: config.guess - is there a chance to remove this file from source tree?

smoe, 2022-02-20:
Dear @jepler,

src/config.guess is changed during the build. Ideally, the regular source tree remains invariant, so everything remains reproducible and we know where a change is coming from.

I just compared src/config.guess it with /usr/share/autoconf/build-aux/config.guess and from a first clance preferred the version in /usr/share/autoconf. I failed to grasp if there is any particular problem you were adressing with a change to that config.guess file that I have missed. If not then I would prefer to copy over the config.guess in debian prior to the execution of configure - I have seen this in a few other packages, too.

Best,
Steffen

jepler, 2022-02-20:
I don't recall why I had to add this file. If you can make our builds happy without it, please do.

jepler, 2022-02-20:
aeb1c09ea7d81fac04fde2b751a396fbb56b241c adds a number of files simply copied in from the autoconf ecosystem. I'd rather that src/autogen.sh copy them in so that it correctly serves not only people building debian packages but people building run-in-place or for non-debian packaging systems.

jepler, 2022-02-20:
Some [autoconf documentation](https://www.gnu.org/savannah-checkouts/gnu/autoconf/manual/autoconf-2.71/autoconf.html) seems to suggest "just take a random copy of these files from the internet":

> The scripts needed by Autoconf core macros can be found in $(datadir)/autoconf/build-aux of the Autoconf installation (see [Installation Directory Variables](https://www.gnu.org/savannah-checkouts/gnu/autoconf/manual/autoconf-2.71/autoconf.html#Installation-Directory-Variables)). install-sh can be downloaded from https://git.savannah.gnu.org/cgit/automake.git/plain/lib/install-sh. config.sub and config.guess can be downloaded from https://git.savannah.gnu.org/cgit/config.git/tree/.

I also found indications that maybe I need to use some macros & `autoreconf --install` would copy them but it didn't for me. https://www.gnu.org/savannah-checkouts/gnu/autoconf/manual/autoconf-2.71/autoconf.html#index-install_002dsh

Please make it better if you know how. We don't know of love auto---- we just use it.

dwrobel, 2022-02-21:
Please find an alternative approach to address this issue in the https://github.com/LinuxCNC/linuxcnc/pull/1625.

smoe, 2022-02-25:
I must admit that having an autogenerated "Makefile" and a preserved "makefile" is a bit weird. This likely is why it is not yet accepted. Can we get comments from @jepler et al. (@SebKuzminsky ?) on how to proceed?