LinuxCNC/linuxcnc #1622: Request hal and other signal for machine is homed

alkabal, 2022-02-21:
Hi

Is possible to add 2 new hal pin for machine.is.homed and halui.machine.is.homed ?

The actual system allow only to see the status for each axis and joint separatly and this is not easy like that for check the machine homing status using external component.

I think having a full machine homing status can be a big enhancement. 

Best regards

SebKuzminsky, 2022-02-21:
I would welcome a PR with this change, though i think if motion exports a pin like this then there's no need for halui to also do it.  Motion's homing code has a function that tells you if all active joints are homed: https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/motion/homing.c#L393

alkabal, 2022-03-03:
Thanks for reply

Yes i have see the recent change and the allhomed flag, at this moment i see more or less what is needed to do but i do see how to write it without many difficulty.

hansu, 2022-03-30:
@alkabal can this be closed now as we have a motion.is-all-homed pin??

alkabal, 2022-03-30:
Oh i didn't show the patch , many thanks to all ! this is really usefull.

Yes off course i close this one.

Best regards

closed 2022-03-30 by alkabal