LinuxCNC/linuxcnc #1634: by_interface\parport\gantry configs uses 4 coordinates when it could use 3

JTrantow, 2022-02-28:
The gantry configurations under configs/by_interface/parport are not reducing to three coordinates nor are they using the negative homing sequence feature. 

These example configs could be improved by changing COORDINATES = X Y Z A to COORDINATES = X Y Z Y and dropping the [AXIS_A] section.
The push will also use negative homing sequences for the Y joints.

JTrantow, 2022-02-28:
The following gantry configurations under configs/by_interface/parport seemed outdated and have a lot more features than a simple XYZ gantry. 
gantry/gantry.ini
gantry/gantry-hs.ini
stepper-gantry/stepper_mm.ini
plasma-thc-sim/plasma_mm.ini

My goal is to clean up a good gantry example config so I think I will avoid stepper_mm.ini and plasma_mm.ini which are out of date but are more complicated than necessary for a gantry example.

I think it makes sense to start over using the 2.8.2 Stepconf Wizard to create a XYZA configuration and then convert it over to a three axis four joint config or two configs.

I see the value in providing a config that can independently home each Y joint.
I'm open to suggestions on whether there should be a separate config for this or whether a commented 
independent jointed XYZY kinstype=BOTH configuration that requires joint homing and another configuration with Y negative HOME_SEQUENCE values.

andypugh, 2022-03-01:
I think this is a decent idea. I don't know how "simulated_home" would work with this config, it might be useful to deliberately offset one of the tandem joints so that the sequence is obvious.

JTrantow, 2022-03-02:
Not sure about the "simulated_home" but at slow home speeds it doesn't take
much offset to hear/see the y home sequence occur. I saw someone else's
config that homed x, y, y2 all the same time HOME_SEQUENCE=1/-1/-1. I
didn't use that as it isn't as obvious. (I will have to try that myself now
that my confidence is up.)
I has a working gantry config in machinekit several years ago. Last few
years with linuxCNC I have been running with one y home switch and manually
de-racking if something goes bad. Using the gantry_joints.ini homing in
joint space was really nice for getting the home switches positioned
correctly.  Once the y home switches were positioned correctly to correct
any racking, the gantry.ini y home sequence does appear to happen
simultaneously.

I don't run any simulated machines, but I have tested these configs on my
router.

grandixximo, 2026-10-05:
Done in #1641 (merged 2022-03-02, in 2.8.3): gantry.ini uses COORDINATES = X Y Z Y with negative HOME_SEQUENCE for the tandem Y joints, and gantry-hs.ini was removed. stepper_mm.ini and plasma_mm.ini were deliberately left out per the discussion above. Closing.

closed 2026-10-05 by grandixximo