LinuxCNC/linuxcnc #1645: dlopen: libcobalt.so.2: shared object cannot be dlopen()ed

zhouguanxing, 2022-03-04:
Dears,
when i want to build the linuxcnc with the realtime xenomai, some issues caused, the detail informations as follows:

1. version:
LinuxCNC  version：master branch
Kernel version：4.19.163
xenomai  version：3.1.0

2. configuration steps：
$ cd src/
$ ./autogen.sh
$ ./configure --with-realtime=uspace
$ make
$ sudo make setuid
$ source ../script/rip-enviroment

3.issue:
By the way, I used the latency compiled by xenomai to test and achieved very good results,
but when I run linuxcnc-master\scripts\latency-test,  the results as follos： 
1."dlopen: libcobalt.so.2: shared object cannot be dlopen()ed"
2."Note: Using POSIX realtime" 
3."Unexpected realtime delay on task 0 with period 25000"

How to solve this issue: "dlopen: libcobalt.so.2: shared object cannot be dlopen()ed"
In source file "uspace_rtapi_app.cc", function "makeApp()":        
dll = dlopen(EMC2_HOME "/lib/libuspace-xenomai.so.0", RTLD_NOW);
if(!dll) fprintf(stderr, "dlopen: %s\n", dlerror());

How can I get the successful note that "Note: Using XENOMAI (posix-skin) realtime" .

zhouguanxing, 2022-03-05:
@hansu dear hansu, could you give me some suggesion? many thanks!

hansu, 2022-03-05:
Which Linux are you running?

zhouguanxing, 2022-03-06:
> 
I use Ubuntu 18.04, and the kernel is 4.19.165.  Thank you!

hansu, 2022-03-06:
I can't help you with this. It seems more related to the kernel than to LinuxCNC.
So please ask in the xenomai communitiy or https://www.forum.linuxcnc.org/.
And please take notice that the issue tracker is not a support forum.

zhouguanxing, 2022-03-07:
> 

got it! thanks!

logic233, 2023-06-19:
Hi, i met this problem and has solved it.
You need to recomplie and reinstall xenamai and use config **--enable-dlopen-libs**
You could see it clearly in this page,[https://source.denx.de/Xenomai/xenomai/-/wikis/Installing_Xenomai_3#_footnotedef_3](url)