LinuxCNC/linuxcnc #1648: Add a "permalink" to stable documenation

silopolis, 2022-03-05:
It would be nice to have the current stable version documentation available at http://linuxcnc.org/docs/stable/html/

TY
Jérémie

hansu, 2022-03-05:
But we have the current stable documentation at http://linuxcnc.org/docs/html/ . What speaks against this location?

silopolis, 2022-03-05:
Hi

Le sam. 5 mars 2022 à 12:52, Hans ***@***.***> a écrit :

> But we have the current stable documentation at
> http://linuxcnc.org/docs/html/ . What speaks against this location?
>

Well, nothing actually but the fact that it is not used on
https://linuxcnc.org/docs/ to link to stable documentation. Instead a
"versioned" link is used.

This made me ignore the existence of the link above until now.

hansu, 2022-03-05:
Yes the main page links to 2.8. 
Do you think "LinuxCNC 2.8 ([HTML](http://linuxcnc.org/docs/2.8/html/))" should link to http://linuxcnc.org/docs/html/ instead of http://linuxcnc.org/docs/2.8/html/ ?

silopolis, 2022-03-05:
Actually, yes. Because title says "Current stable version", and because it would reference main documentation with a stable (in time) link more suitable for bookmarking and reference in our own support contents.

I suggested http://linuxcnc.org/docs/stable/html/ to reproduce the http://linuxcnc.org/docs/devel/html/ pattern.

This was actually triggered by #997 where adding a versioned link didn't seemed like a good idea.

andypugh, 2022-03-22:
The URL on the documentation starter page is now:
http://linuxcnc.org/docs/html/ (and /pdf/)