LinuxCNC/linuxcnc #1653: loadrt failed

zhouguanxing, 2022-03-09:
Dears, 
while i use the loadrt command to load the realtime component, i found that in "halcmd_commands.c" file, the function of "do_loadusr_cmd" laoded the realtime component into the user space, the detail information as follows:
1. the version information:
    LinuxCNC  version：master
    Kernel version：4.19.163
    xenomai  version：3.1.0
    Ubuntu18.04
2.  configuration steps：
    $ cd src/
    $ ./autogen.sh
    $ ./configure --with-realtime=uspace
    $ make
    $ sudo make setuid
    $ source ../script/rip-enviroment
3. issue description:
    Config by the command "./configure --with-realtime=uspace", and it defined the macro of "RTAPI_USPACE" in the "config.h" file,  the detail as follows:
/* Realtime system is userspace */
#define RTAPI_USPACE /**/
the detail of the function "do_loadusr_cmd" as follows:
int do_loadrt_cmd(char *mod_name, char *args[])
{
    char arg_string[MAX_CMD_LEN+1];
    int m=0, n=0, retval;
    hal_comp_t *comp;
    char *argv[MAX_TOK+3];
    char *cp1;
#if defined(RTAPI_USPACE)
    argv[m++] = "-Wn";
    argv[m++] = mod_name;
    argv[m++] = EMC2_BIN_DIR "/rtapi_app";
    argv[m++] = "load";
    argv[m++] = mod_name;
    /* loop thru remaining arguments */
    while ( args[n] && args[n][0] != '\0' ) {
        argv[m++] = args[n++];
    }
    argv[m++] = NULL;
    retval = do_loadusr_cmd(argv);
#else
....
}
is the command of the "./configure" is not correct? 
So, how to configure the parameters of the "./configure" command to correctly execute the loadrt instruction?
@vmiklos  could you give me some suggesions? many thanks!

andypugh, 2022-03-09:
Which realtime component are you trying to load? 

Does the configure script seem to be finding Xenomai?

zhouguanxing, 2022-03-10:
dear andypugh,
After my config the realtime enviroment by the command './configure ...',  in 'do_loadrt_cmd' function, it go into the '#if defined(RTAPI_USPACE)' branch not the '#else' branch, is this meanings i config successful? or not?
int do_loadrt_cmd(char *mod_name, char *args[])
{
char arg_string[MAX_CMD_LEN+1];
int m=0, n=0, retval;
hal_comp_t *comp;
char *argv[MAX_TOK+3];
char cp1;
_**#if defined(RTAPI_USPACE)
argv[m++] = "-Wn";
argv[m++] = mod_name;
argv[m++] = EMC2_BIN_DIR "/rtapi_app";
argv[m++] = "load";
argv[m++] = mod_name;
/ loop thru remaining arguments */
while ( args[n] && args[n][0] != '\0' ) {
argv[m++] = args[n++];
}
argv[m++] = NULL;
retval = do_loadusr_cmd(argv);**_
#else
....
}

vmiklos, 2022-03-10:
I assume you wanted to tag someone else :-)

zhouguanxing, 2022-03-10:
> I assume you wanted to tag someone else :-)

ok, thanks!

andypugh, 2022-03-10:
You are configuring with --with-realtime=uspace so I would expect RTAPI_USPACE to be defined. 

If you look in the boot config, which of the CONFIG_IPIPE and CONFIG_XENO things do you see? 
https://github.com/LinuxCNC/linuxcnc/blob/a2f1fddfd35121bebef6b31560a45deaf66f677c/scripts/platform-is-supported#L41

zhouguanxing, 2022-03-11:
> You are configuring with --with-realtime=uspace so I would expect RTAPI_USPACE to be defined.
>
> If you look in the boot config, which of the CONFIG_IPIPE and CONFIG_XENO things do you see? [...]

dear andypugh, thank you!
after my check, i found that the function "def detect_kernel_flavor(uname):"  return value is "vanilla", is it say my realtime enviroment setting is incorrect?

andypugh, 2022-03-11:
What does uname -a say? Does that indicate a Xenomai kernel? 

Then look in the kernel config (/boot/config-{name-of-the-kernel-you-are-actually-running} and see if the file contains CONFIG_XENO and CONFIG_IPIPE

andypugh, 2022-04-05:
Is this issue still live? Or have you solved it?

zhouguanxing, 2022-04-07:
i am sorry to reply late, i had solved it. thank you!

closed 2022-04-07 by hansu