LinuxCNC/linuxcnc #1655: /usr/share/icons/hicolor/scalable/apps/linuxcnc-en.svg not found but referenced in .desktop file

smoe, 2022-03-09:
from https://appstream.debian.org/sid/main/issues/linuxcnc-uspace.html:

The icon '/usr/share/icons/hicolor/scalable/apps/linuxcnc-en.svg' was not found in the archive. This issue can have multiple reasons:
The icon is not present in the archive.
The icon is in a wrong directory.
The icon is not available in a suitable size (at least 64x64px)
On Debian and Ubuntu, the icon is a symlink. The generator cannot read symlinks on these distributions - make the icon a real file.
To make the icon easier to find, place it in /usr/share/icons/hicolor/<size>/apps and ensure the Icon= value of the .desktop file is set correctly.

andypugh, 2022-03-09:
On Wed, 9 Mar 2022 at 10:32, Steffen Möller ***@***.***>
wrote:

> from https://appstream.debian.org/sid/main/issues/linuxcnc-uspace.html:
>
> The icon '/usr/share/icons/hicolor/scalable/apps/linuxcnc-en.svg' was not
> found in the archive.
>
...

> The icon is not available in a suitable size (at least 64x64px)
>
It's an SVG, it doesn't do pixels.

It's also here:
https://github.com/LinuxCNC/linuxcnc/tree/master/debian/extras/usr/share/icons/hicolor/scalable/apps

-- 
atp

smoe, 2022-03-10:
I read:
 * maybe debian/extra is the wrong location for something that (so I expect) affects also Fedora
 * a quick grep suggests that the .desktop files are just copied when these should be inscaped into .pngs?!?
```
$ grep -r linuxcnc-en.svg debian/
debian/linuxcnc-uspace/usr/share/applications/linuxcnc-gcoderef.desktop:Icon=/usr/share/icons/hicolor/scalable/apps/linuxcnc-en.svg
debian/linuxcnc-doc-en/usr/share/applications/linuxcnc-integratorinfo.desktop:Icon=/usr/share/icons/hicolor/scalable/apps/linuxcnc-en.svg
debian/linuxcnc-doc-en/usr/share/applications/linuxcnc-documentation.desktop:Icon=/usr/share/icons/hicolor/scalable/apps/linuxcnc-en.svg
debian/linuxcnc-doc-en/usr/share/applications/linuxcnc-gettingstarted.desktop:Icon=/usr/share/icons/hicolor/scalable/apps/linuxcnc-en.svg
debian/extras/usr/share/applications/linuxcnc-gettingstarted.desktop:Icon=/usr/share/icons/hicolor/scalable/apps/linuxcnc-en.svg
debian/extras/usr/share/applications/linuxcnc-documentation.desktop:Icon=/usr/share/icons/hicolor/scalable/apps/linuxcnc-en.svg
debian/extras/usr/share/applications/linuxcnc-integratorinfo.desktop:Icon=/usr/share/icons/hicolor/scalable/apps/linuxcnc-en.svg
debian/extras/usr/share/applications/linuxcnc-gcoderef.desktop:Icon=/usr/share/icons/hicolor/scalable/apps/linuxcnc-en.svg
```
This is how to perform the conversion:
```
inkscape --export-width=64 --export-height=64 --export-type=png debian/extras/usr/share/icons/hicolor/scalable/apps/linuxcnc_en.svg
```
Inkscape is already a build-dependency.

havardAasen, 2022-03-10:
One thing that's wrong is the filename. The icon file was renamed `linuxcnc-en.svg`-> `linuxcnc_en.svg` and it looks like the .desktop files wasn't updated. Seems that the commit 32b2aec2e43b7e41ba61cf06de34823fb3e192e2 was incomplete. The only desktop entries that uses this icon, is also the one which is missing the icon.

I haven't been able to verify this but there exists other packages in the Debian repository which only ships .svg files (yelp is one of them), and they don't have these warnings. The only difference I can see is that they only have the filename in their icon entry, no path or extension. This differs from ours which includes the full path and extension.
Now, looking at the specification for the desktop files at https://specifications.freedesktop.org/desktop-entry-spec/desktop-entry-spec-1.1.html the icon entry can indeed be a full path, which might suggest that the problem comes from the conversion to appstream, if that's happening. I'm not sure about this.

A place to start would be to update the necessary .desktop files with the new filename `linuxcnc_en.svg`. A possible second step, which I haven't verified, would be to remove the path and extension of the icon entry in the .dekstop files, so we're ending up with only `linuxcnc_en`.

To clarify, I haven't verified that the appstream warning goes away if we remove the path and extension, I can however verify that the icon is displaying properly if we remove it.

smoe, 2022-03-10:
Thank you, @havardAasen . I did not expect this to have a phenotype, I must admit, which is of course much more important to address. Would you go for the PR (it is your contribution)? What we as a community also need to discuss at some point is eye candy. This "EN"-icon is a bit of a duckling, and I can see nothing LinuxCNC-ish in it.

smoe, 2022-03-10:
That is all difficult, indeed. With a bit of luck, the weblate community eventually grows strong enough to help out. For the larger icons for instance I would like to see a reading penguin (with eye protection still on?) - and I am not convinced, yet, that we need icons for the different languages in the first place.  There is your documentation.  Yup. Thanks.

LinuxCNC with its frequently used USB stick to boot from is special in that we as developers have a bit of an opportunity to define what the user's desktop should look like. Raspbian is similar. So, when starting this up, this needs to make a good impression - not only on the operator of the mill but also on friends and family, who cannot judge the quality of the software but they can judge the effort that went into how it all looks.

havardAasen, 2022-03-14:
This issue should be reopened, my commit that originally closed this didn't catch everything. Removing the path and suffix does nothing to solve the problem, and renaming one of the icons only fixed part of the issue.

The problem lies in that the desktop files and icons is not placed in the correct (same) package. All the icons is placed in the package `linuxcnc-uspace` while several (not all) desktop files is placed in their respective `linuxcnc-doc-*` package. Since none of the documentation packages depend on `linuxcnc-uspace`, the icons won't exist if we only install the documentation. Resulting in missing icons.

This can probably be solved in several different ways, but I would probably move the language specific icons along with the  .desktop files to their respective documentation package.

These are the files that I propose to move into their respective documentation package:
- linuxcnc_en.svg
- linuxcnc_es.svg
- linuxcnc_fr.svg
- linuxcnc_zh_CN.svg
- gcode.html
- gcode_es.html
- gcode_fr.html
- linuxcnc-gcoderef.desktop

In `debian/rules` there is a rather complicated way to move files into the documentation packages. I would believe that adding build profiles is a more elegant solution. It's more work, though probably easier to maintain in the long run.

andypugh, 2022-03-14:
I agree that the icons should be part of the relevant docs package.

smoe, 2022-03-14:
I'll address that ... this week.

petterreinholdtsen, 2022-07-28:
@smoe The week in March passed a while back.  Did you figure out a way to fix this?

smoe, 2022-07-28:
I somehow deeply feel like I had addressed this, but cannot recall and do not have the time for the next two/three weeks to look into the packaging.

havardAasen, 2022-07-29:
I would believe this could have a rather simple solution if #1835 is merged.
Just add back `debian/linuxcnc-doc-*.install` files and specifically mention the files we want to install.

commit c952a815 (2022-03-12): Merge pull request #1660 from havardAasen/fix_desktop_icon

packaging: Fix desktop icons not found (Fixes #1655)
commit d3effdff (2022-03-12): packaging: Fix desktop icons not found (Fixes #1655)

Drop path and extension for icons. The icon for English documentation
had also been renamed 'linuxcnc-en' -> 'linuxcnc_en'.