LinuxCNC/linuxcnc #1658: Required header Python.h missing during configure

allistar, 2022-03-09:
## Here are the steps I follow to reproduce the issue:

 1. My configure command is:
` ./configure --with-realtime=uspace --enable-non-distributable=yes --with-boost-python=boost_python39`

## This is what I expected to happen:
The configure command should finish without error.

## This is what happened instead:
The last few lines of the configure output:

```
checking for msgfmt... /usr/bin/msgfmt
checking python version... OK
checking version of python libraries... python3.9
checking match between tk and Tkinter versions... 8.6
checking location of Python header files... /usr/include/python3.9
checking for Python headers... -I/usr/include/python3.9
checking for Python libraries... -lpthread -ldl  -lutil -lm
checking whether the Boost::Python headers are available... yes
checking for boost::python shared library... -lboost_python39
checking /usr/include/python3.9/Python.h usability... no
checking /usr/include/python3.9/Python.h presence... no
checking for /usr/include/python3.9/Python.h... no
configure: error: Required header Python.h missing.  Install it, or specify --disable-python to skip the parts of LinuxCNC that depend on Python
```

That .h file exists, so there's some kind of check that causing this to fail:
```
ls -la /usr/include/python3.9/Python.h
-rw-r--r-- 1 root root 3533 Feb 23 10:07 /usr/include/python3.9/Python.h
```

## It worked properly before this:
I have been using LinuxCNC (v2.9.0-pre0) with Python 2.7 successfully for a few years. Python 2 is becoming harder to maintain on my distro so I want to build it with python 3.

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
LSB Version:    n/a
Distributor ID: Gentoo
Description:    Gentoo Base System release 2.7
Release:        2.7
Codename:       n/a

 * I am using this kernel version (shown by `uname -a`):
 Linux legolas 4.19.10-rt-rt8 #10 SMP PREEMPT RT Thu Sep 16 09:36:33 NZST 2021 x86_64 Intel(R) Core(TM)2 Quad CPU Q6600 @ 2.40GHz GenuineIntel GNU/Linux

 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [x] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 v2.9.0-pre0-1019-g2e0f88e2a
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 
parallel port

Questions:
Can I enable a more verbose version of the configure tests so I can see what's going on here?

smoe, 2022-03-10:
Hello,
what is "python3 --version" saying? This is just to follow my hunch that you have the -dev files for 3.9 but are workign with 3.10.
Cheers,
Steffen

allistar, 2022-03-10:
That returns "Python 3.9.9"

smoe, 2022-03-10:
There may be another way to trigger this. Is (as it is in Debian) a second (newer) version of Python available? Please kindly check  what `ls /usr/bin/python3*` offers or for the fun of it try:
```
/usr/bin/python3.10 -c "print('3.10 is also available')" || echo "Nope, nice try."
```

allistar, 2022-03-10:
The result of that is:
3.10 is also available

smoe, 2022-03-10:
That is excellent news in the sense that you can then be helped. I think :) The "Python-detection" will find the 3.10 version and search for include files only for that. Please apply this patch:
```
--- linuxcnc.orig/src/m4/ax_python.m4
+++ linuxcnc/src/m4/ax_python.m4
@@ -55,7 +55,7 @@
 AC_DEFUN([AX_PYTHON],
 [AC_MSG_CHECKING(for python build information)
 AC_MSG_RESULT([])
-for python in python3.10 python3.9 python3.8 python3.7 python3.6 python3.5 python3.4 python3.3 python3.2 python3.1 python3.0 python2.7 python2.6 python2.5 python2.4 python2.3 python2.2 python2.1 python; do
+for python in python3.9 python3.8 python3.7 python3.6 python3.5 python3.4 python3.3 python3.2 python3.1 python3.0 python2.7 python2.6 python2.5 python2.4 python2.3 python2.2 python2.1 python; do
 AC_CHECK_PROGS(PYTHON_BIN, [$python])
 ax_python_bin=$PYTHON_BIN
 if test x$ax_python_bin != x; then
```
read: Please remove the search for python3.10
and enjoy your then most-likely then perfectly-fine build.

Fedora is already more advanced and needs that, so we cannot remove it.

allistar, 2022-03-11:
Thanks for the very helpful suggestion.
What I've done instead is default the system python to 3.10, rebuild python (so it build the 3.10 versions of the various libraries) and changed the configure line to:

`./configure --with-realtime=uspace --enable-non-distributable=yes --with-boost-python=boost_python310`

And that works. There was one minor issue with this error:

`config.status: error: cannot find input file: `../share/desktop-directories/cnc.directory.in'`

I fixed that by renaming ../share/desktop-directories/cnc.directory to ../share/desktop-directories/cnc.directory.in

When running "make" it gets quite far and then gives this error:
```
yapps: no python-exec wrapped executable found in /usr/lib/python-exec.
make: *** [hal/utils/Submakefile:102: objects/hal/utils/halcompile.py] Error 127
```
I recall struggling with yapps a few years ago on python2.7. "which yapps" returns this:

/usr/bin/yapps

I can see there's no "yapps" file in /usr/lib/python-exec/python3.10 but there is a "yappi" file. So I've install "pip" and then installed yapps through that. Now yapps will run, though I get this error during "make":

```
Syntax checking python script halcompile
Copying python script halcompile
Preprocessing thermistor.comp
Traceback (most recent call last):
  File "/home/allistar/Projects/linuxcnc-build/src/../bin/halcompile", line 23, in <module>
    from yapps import runtime
  File "/home/allistar/.local/lib/python3.10/site-packages/yapps/runtime.py", line 438
    except SyntaxError, e:
           ^^^^^^^^^^^^^^
SyntaxError: multiple exception types must be parenthesized
make: *** [hal/user_comps/Submakefile:102: objects/hal/user_comps/thermistor.c] Error 1
```

It seems that yapps is deprecated or perhaps doesn't work properly on python3.10, this looks like a Python2 vs Python3 difference. I've edited that line in that file by adding ( ) around the exception type definitions. The make command finished. When running I get an error: no module named "OpenGL". After installing the python opengl module it works!

I'm very pleased to not have to much around with python 2.7 on this server anymore. Thanks again for your help.

It would be great if there was a gentoo ebuild for LinuxCNC, but I suppose there aren't many people that use that distro for CNC.

smoe, 2022-03-11:
I guess LinuxCNC has you now :) What you might want to do is to open a new issue with Gentoo in its title and ask for contacts. Also an email to the mailing list or the Discord, on IRC or the discord channel may be good. Many thanks for completing your journey!

silopolis, 2022-03-11:
Le ven. 11 mars 2022 à 03:06, Allistar Melville ***@***.***>
a écrit :

> It would be great if there was a gentoo ebuild for LinuxCNC, but I suppose
> there aren't many people that use that distro for CNC.
>
There may not be many today, but Gentoo could be a great addition to the
list of supported OSs for people looking for advanced optimization 👌

Nothing that can't be done with Debian toolchain (which would my choice),
but it's Gentoo's original seed.

TY
Jérémie

smoe, 2022-03-24:
This issue can be closed? Or maybe renamed to "we want a LinuxCNC-gentoo package"?

allistar, 2022-03-24:
I'm happy to close this off. I'll raise a new issue for wanting a Gentoo package at some point.
It does seem like a bug that it will use the highest installed python regardless of is that's the currently active one.

closed 2022-03-24 by allistar

smoe, 2022-03-24:
Sigh. I personally think so, too.