LinuxCNC/linuxcnc #1679: Provide an ebuild for Gentoo

allistar, 2022-03-24:
## Here are the steps I follow to reproduce the issue:

There is currently no easy install process for the Gentoo Linux distribution. It can be built on Gentoo manually by following a set of instructions. It would be useful for Gentoo users if there was an ebuild that automated this so we can "emerge linuxcnc".

Another issue showing steps on how to install on Gentoo is [here](https://github.com/LinuxCNC/linuxcnc/issues/1658).

## This is what I expected to happen:
To be able to run the command:
`emerge linuxcnc
`on Gentoo Linux and have LinuxCNC build and install.

## This is what happened instead:
There is no ebuild.

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
LSB Version: n/a
Distributor ID: Gentoo
Description: Gentoo Base System release 2.7
Release: 2.7
Codename: n/a
 * I am using this kernel version (shown by `uname -a`):

 Linux legolas 4.19.10-rt-rt8 10 SMP PREEMPT RT Thu Sep 16 09:36:33 NZST 2021 x86_64 Intel(R) Core(TM)2 Quad CPU Q6600 @ 2.40GHz GenuineIntel GNU/Linux

 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [x] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 v2.9.0-pre0-1019-g2e0f88e2a
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 
 Parallel port

allistar, 2022-03-24:
An ebuild would live on the main Gentoo repository servers and be mirrored through their network, though I expect that some changes will be necessary to the LinuxCNC code base too. I think the steps for this task are:

- map configure parameters to use flags where possible
- work out dependencies
- work through bugs like:
   - there's no yapps ebuild but LinuxCNC requires it
   - building requires ../share/desktop-directories/cnc.directory be renamed to ../share/desktop-directories/cnc.directory.in
   - the build process assumes the highest installed python is the currently active one.
 
Most of that is straight forward apart from the yapps issue. I had to install it using "pip" and I don't think a ebuild should depend on pip and then call pip to install a package. Also, I then needed to modify the installed yapps package. Is one approach to ship yapps with the gentoo ebuild? I expect that'll be python version dependant so not straight forward. The best approach is to also provide a yapps ebuild for Gentoo. Sigh.

jack9603301, 2022-05-10:
I agree. In this way, users can choose which distribution to use Linux CNC. This will enable Gentoo users to build linux CNC on Gentoo. I will continue to track this issue and hope that the community can consider implementing it.

silopolis, 2022-05-10:
Hi,

AFAIC, an ebuild would be surely welcome :)
Portability is good(c)

Plus, with its optimization bias, Gentoo feels in line with machine control
system building...

The only issue I see there is on the community side... Resources and
support may be even scarser on a new platform and newcomers may need to
have a supplement of motivation to seed this new group of users.

TY

Gahia123, 2022-06-18:
I'll love to experiment with LinuxCNC using gentoo, would feel more comfortable developing some ideas in the future.
I've no experience making ebuilds yet, nevertheless with patience and the help of others surely we can succeed; please tell me if there is someone who want to divide the work, wich strategy can we use and make it possible.

allistar, 2022-06-19:
I've been using LinuxCNC on Gentoo successfully for a few years. The biggest issue I see with an ebuild is that there's no ebuild for yapps and the LinuxCNC build process requires it. I think we'd need to make a yapps ebuild first.

Gahia123, 2022-06-20:
Which version of yapps do we require? 1 or 2?

allistar, 2022-06-20:
When I run "pip show yapps" it returns this:

```
Name: yapps
Version: 14.4.0
Summary: Yet Another Python Parser System
Home-page: https://github.com/mk-fg/yapps
```
I had to made one modification of yapps to make it work in Python 3 after installing it.

allistar, 2022-06-29:
I installed yapps on Gentoo using pip. This didn't work on Python 3 but it was a single change to a single file to get it to work.
It seems strange that LinuxCNC runs on Python 3 and requires yapps, but yapps doesn't officially work on Python 3.

I think a better approach is to make a yapps ebuild and add a python3 use flag, and if that flag is set them apply a patch that makes it Python3 compatible.

Gahia123, 2022-06-29:
I'll study how to make yapps ebuild, meaby it'll take some time.

andypugh, 2022-07-21:
https://forum.linuxcnc.org/9-installing-linuxcnc/46439-linuxcnc-2-9-master-on-gentoo#247908

openlunchbox, 2022-07-23:
2.9/Master is now working on Gentoo. Ebuild will be available shortly.

openlunchbox, 2022-07-23:
https://sourceforge.net/p/emc/mailman/message/37683768/  if you want to follow on the dev ML

NTULINUX, 2022-07-28:
Hello,

I have an ebuild that works, I need to fix docs, implement USE flags, make the CNC menu and do some basic kernel config sanity checks. I might add RTAI ebuilds and modify the LinuxCNC ebuild accordingly to account for both PREEMPT_RT and RTAI but that is down the road, if ever.

It's not perfect:

https://github.com/NTULINUX/ntu_overlay/blob/main/sci-electronics/linuxcnc/linuxcnc-9999.ebuild

silopolis, 2022-07-28:
Le jeu. 28 juil. 2022 à 08:22, NTU ***@***.***> a écrit :

> Hello,
>
> I have an ebuild that works,
>

Great!

I need to fix docs, implement USE flags, make the CNC menu and do some
> basic kernel config sanity checks. I might add RTAI ebuilds and modify the
> LinuxCNC ebuild accordingly to account for both PREEMPT_RT and RTAI but
> that is down the road, if ever.
>

More goodness to come 👍🙂

It's not perfect
>

Even my son isn't 😉

https://github.com/NTULINUX/ntu_overlay/blob/main/sci-electronics/linuxcnc/linuxcnc-9999.ebuild
>

Could this be put in a gentoo directory in linuxcnc repository? Or is it
better in a separate one?

Thanks and congrats again, keep it coming

TY
J

NTULINUX, 2022-07-29:
The ebuild actually didn't work before because TCLLIBPATH wasn't set, this commit handles that error:

https://github.com/NTULINUX/ntu_overlay/commit/a396ce3b87d5dcb9c67d56a53cfac17a0941a85e

Now it's just basic checks, some QA fixes (doc and USE flag stuff) and better graphical menu support but it's good enough to start with. Cheers!

NTULINUX, 2024-07-04:
I haven't tested them in awhile but I actually have 2 ebuilds for LinuxCNC on Gentoo:

https://github.com/NTULINUX/ntu_overlay/tree/main/sci-electronics/linuxcnc