LinuxCNC/linuxcnc #1692: Debian unstable  linuxcnc-uspace is no longer installable (Python version error)

rodw-au, 2022-04-01:
I went to install linuxcnc onto a VPS and was unable to do so due to what I thought was a silly dependency preventing installation with Python 3.10

Depends: python3 (< 3.10) but 3.10.4-1 is to be installed

Python version in Debian are shown here:
https://packages.debian.org/search?suite=default&section=all&arch=any&searchon=names&keywords=python3

If the package was moved to testing branch in terms with what I thought should have been an automatic process, this error would be resolved.

So at this point in time, the effort that went tinto getting linuxcnc into debian is useless as its unusable.
I might add I have installed on unstable several times before so the Python version must have been advanced.

rodw-au, 2022-04-01:
With the changes smoe made as mentioned on the email list, 
I can confirm that linuxcnc debs and RIP compile on  the 5.16 kernel.

I was unable to install linuxcnc-uspace from the debian repository which reported the same error.
Is that becasue the changes have not ended up in debian yet?

smoe, 2022-04-01:
For Debian unstable, the default Python version shifted to 3.10 just a few days ago, while the initial upload was built with 3.9. I admit to be somewhat confused over this, too, but an easy fix would be to indeed just reupload the Debian package.

That new upload will then be auto-built across platforms, also for the RPi. I could of course just take the current snapshot, but would like to get a go-ahead from @SebKuzminsky , @andypugh or @jepler for that.

rodw-au, 2022-04-02:
I for one would love to see a current snapshot used. That would bring  in Dewey's work on HOMEMOD and a lot of other useful changes. I'm just waiting for an ethercat drive to arrive so I can start working on a version of homing for CIA402 internal homing routines in  conjunction with Dan Dempsey in Canada.

I will also testing on the Pi. I did install Debian 11.2 on a Pi  and built Linuxcnc but got stuck with the Ethercat master. I have since found a repo for IgH's etherlabmaster which  supports  most Debian Distros (including unstable) and the Pi.

smoe, 2022-04-04:
I updated the Debian package, but on the build demons (https://buildd.debian.org/status/fetch.php?pkg=linuxcnc&arch=all&ver=2.9.0%7Epre0%2Bgit20220402.2500863908-2&stamp=1649009845&raw=0) ran into problems since it builds the arch-dependent and arch-independent packages separately. The error is reproducible with
```
dpkg-buildpackage -A
```
and caused by src/Makefile's install target assuming that everything was built, when it is possible that only the documentation was built.

smoe, 2022-04-06:
You got a recent snapshot now built by the Debian autobuilders, see https://github.com/LinuxCNC/linuxcnc/pull/1700 . I am not sure about how soon these newly built packages will hit the mirrors, but give it a try, please.

snelweg, 2022-06-17:
I can confirm that LinuxCNC works on Debian unstable, using the Debian repositories.

smoe, 2022-06-17:
There is Python 3.11 now arriving in Debian that likely triggers this kind of an error again.

smoe, 2022-06-17:
The current problem with the tiff library in Debian testing and unstable is independent of this problem (compare https://github.com/LinuxCNC/linuxcnc/pull/1700)

petterreinholdtsen, 2022-07-27:
As the purpose and property of Debian/unstable is to be the location where dependency issues are detected and solved, while Debian/testing have a system to keep packages with release critical bugs out of Debian/testing, this issue seem to be either related to unreasonable expectations (packages in unstable will not always be installable during transitions) or fixed (the package in testing and unstable is installable now).  In any case, I suspect it can be closed.

rodw-au, 2023-01-17:
Closing this.

closed 2023-01-17 by rodw-au