LinuxCNC/linuxcnc #1701: halcompile will not compile custom homecomp.comp with code from homing.c

rodw-au, 2022-04-05:
I am writing a custom homing module for homing cia402 ethercat drives to use its internal homing features. I get this error when compiling exact copies of write_homing_out_pins() and read_homing_in_pins
These are critical functions as a homing module must be able to read and write pins.
Errors are:
`rod@penguin:~$ halcompile --compile ciahome.comp
Compiling realtime ciahome.c
ciahome.comp: In function ‘write_homing_out_pins’:
ciahome.comp:287:33: error: ‘H’ undeclared (first use in this function)
ciahome.comp:287:33: note: each undeclared identifier is reported only once for each function it appears in
ciahome.comp: In function ‘read_homing_in_pins’:
ciahome.comp:304:9: error: ‘H’ undeclared (first use in this function)
make: *** [/home/rod/linuxcnc-dev/src/Makefile.modinc:115: ciahome.o] Error 1`

The error relates to lines taking the following form:
`*(addr->homing)       = H[jno].homing;       // OUT
H[jno].index_enable = *(addr->index_enable); // IO
`
I added all of the include files in homing.c and this did not resolve the error.
`#include "motion.h"
#include "homing.h"
#include "hal.h"
#include "rtapi.h"
#include "rtapi_math.h"
`
I don't know where H is declared or if its some kind of compiler directive
I can only assume the build environment is different in halcompile
Can a workaround for this issue be added to the homecomp environment

dngarrett, 2022-04-05:
1) "error: 'H' undeclared ..." means you are using a
variable named named 'H' without declaring it.

2) One cannot just copy a function from one program to
another if the function refers to variables whose
definitions are not copied or otherwise defined.

The mentioned H[] array var is explicitly defined in the
default homing.c as a static variable with file scope:

https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/motion/homing.c#L130

2) File static vars (like H[] (and *many* others) retain
state information between invocations of the component main
program (rtapi_app_main()) called by motmod at the servo
period interval.

3) The file src/hal/components/homecomp.comp illustrates how
to create a component that supplys all api functions and can
be built/installed with halcompile.  The example can be used
with LinuxCNC but is not fully functional ($ man homecomp).
Note that the example uses only a few static vars sufficient
to support minimal api requirements and a set of hal pins
compatible with sim configs.  A custom homing module can
define additional file static variables as required to implement
more functionality in the same way as done in the default
homing.c

https://github.com/LinuxCNC/linuxcnc/blob/master/src/hal/components/homecomp.comp

4) NOTE: the homing api for the master branch was revised
recently (05apr22):

https://github.com/LinuxCNC/linuxcnc/commit/d4dddb2f93

closed 2022-04-05 by dngarrett

rodw-au, 2022-04-05:
Thanks Dewey. I got it to work after adding the H[] array and home_state_t
Do the deletions and additions affect the skeletons in homecomp.comp?
I really appreciate your help.