LinuxCNC/linuxcnc #1705: HOMEMOD does not install user defined home module

rodw-au, 2022-04-09:
Following the instructions in the homecomp man page http://linuxcnc.org/docs/devel/html/man/man9/homecomp.9.html
I have started to write a fairly complex home module for cia402 ethercat drives.

I ran up a fresh axis-mm sim and set the ini file to

```
[EMC]
HOMEMOD=ciahome
```
My pins were not visible.

Closer inspection revelaed that Linuxcnc was still running HOMEMOD.
No errors were reported in the terminal

I used the command line option
`linuxcnc -m ciahome ./axis_mm.ini`
and it loaded the correct module which in turn reported some errors in my pin structure. Linuxcnc must silently revert to the internal HOMEMOD module.

EXAMPLE command line error:
`HAL: ERROR: pin_new(joint.0.cia-stat-voltage-enabled) called with already-initialized memory
`
It seems there is no error checking being applied when loaded from the ini file

I shall keep working on this to resolve the errors and see how the ini file install goes.

It would seem prudent for Linuxcnc to report an error to the user when encountering errors like I made.

Still its exciting to be able to work on this!

rodw-au, 2022-04-09:
So the error was duplicate pin names from all of the cut and paste. Fixing them and I made some progress.
The ini file method still loads the default homemod
The command line returned an error

```
linuxcnc TPMOD=tpmod HOMEMOD=ciahome EMCMOT=motmod
Note: Using POSIX realtime
@@@ciahome.comp:extra_setup: home_parms=no_home_parms
Found file(lib): /usr/share/linuxcnc/hallib/core_sim.hal
HOMING: make pins completed
Found file(lib): /usr/share/linuxcnc/hallib/sim_spindle_encoder.hal
Found file(lib): /usr/share/linuxcnc/hallib/axis_manualtoolchange.hal
Found file(lib): /usr/share/linuxcnc/hallib/simulated_home.hal
Found file(REL): ./postgui.hal
rtapi_app: caught signal 11 - dumping core
USRMOT: ERROR: command timeout
emcMotionInit: emcTrajInit failed
```
I am out of time but I suspect the most likely cause could be publishing a pin with the wrong underlying data type (eg bit as a float.

I have attached the work in progress. So many pins!
[ciahome.comp.txt](https://github.com/LinuxCNC/linuxcnc/files/8456258/ciahome.comp.txt)

dngarrett, 2022-04-09:
> It would seem prudent for Linuxcnc to report an error
> to the user when encountering errors like I made.

You have to follow instructions:

1)As noted in the homecomp  man page, the ini setting
is  [EMCMOT]HOMECOMP=
NOT [EMC]HOMECOMP=

2) To test interface issues like ini settings and
commandline options, one can use the provided,working
 example (homecomp.comp)

3) Note: the homing api was changed at commit:
d4dddb2f93 (your example is not up-to-date with
master branch)

4) As noted when creating github new issues:

"# The issue tracker is NOT a support forum

The LinuxCNC issue tracker is to report bugs in the
software.  If you have a question about how to use the
software, use one of the other methods detailed on our
community support page: http://linuxcnc.org/community"

closed 2022-04-09 by dngarrett

rodw-au, 2022-04-09:
Thanks, I had updated to the latest master but did not change the base homecomp for my code.
I've only reported what I genuinely believed were issues with the new homing code.
Its an important enhancement but its not likely to be a lot of users exercising it.