LinuxCNC/linuxcnc #1714: Spindle Speed Synch

satiowadahc, 2022-04-14:
```python
import linuxcnc
c = linuxcnc.command()
c.spindle(linuxcnc.SPINDLE_FORWARD, 1000, 0)
c.spindle(linuxcnc.SPINDLE_OFF, 0)

c.mode(linuxcnc.MDI_MODE)
c.mdi("M3")
```

What I expect to happen: turn spindle on at 1000.
What Actually happens Spindle turns on at previous S command, or 0 if not entered.

Debug Messages:

c.spindle(linuxcnc.SPINDLE_FORWARD, 1000, 0):
c.spindle(linuxcnc.SPINDLE_OFF, 0)

```bash
Issuing EMC_SPINDLE_ON --        ( +1304,+64,   +28,1000.000000,0.000000,0.000000,    +1,)
Issuing EMC_SPINDLE_OFF --       ( +1305,+32,   +29,)
```

c.mdi("M3")
```bash
Issuing EMC_TASK_PLAN_EXECUTE --         (  +509,+280,   +19,M3,)
emcTaskPlanLevel() returned 0
NML_INTERP_LIST(0x55f42436a120)::append(nml_msg_ptr{size=64,type=EMC_SPINDLE_ON}) : list_size=1, line_number=0
emcTaskPlanExecute(M3) returned 0
emcTaskPlanLevel() returned 0
NML_INTERP_LIST(0x55f42436a120)::get(): {size=64, type=EMC_SPINDLE_ON}, list_size=0
emcTaskPlanLevel() returned 0
Issuing EMC_SPINDLE_ON --        ( +1304,+64,    +0,0.000000,0.000000,0.000000,    +1,)
mdi_execute_hook: MDI command 'M3' done (remaining: 0)
```

Also for what its worth, qtvcp mdi commands will inject add a new line to the end of each command.
```bash
Issuing EMC_SPINDLE_SPEED --     ( +1316,+56,    +0,0.000000,0.000000,0.000000,)
mdi_execute_hook: MDI command 'S1000
' done (remaining: 0)
```

hansu, 2022-04-14:
Might be related to this https://github.com/LinuxCNC/linuxcnc/issues/1181?

satiowadahc, 2022-04-14:
Interesting.
- command.spidle_on -> spindle turns on at set RPM
- mdi m3 -> spindle turns off.

And Can Confirm #1181 
```python3
CW:~$ python3
Python 3.8.10 (default, Mar 15 2022, 12:22:08) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import linuxcnc
>>> c = linuxcnc.command()
>>> c.mode(linuxcnc.MODE_MDI)
>>> c.spindle(linuxcnc.SPINDLE_FORWARD, 1500, 0)
>>> c.mdi("S1000")  # Spindle turns off
```
output:
```bash
Issuing EMC_SPINDLE_ON --        ( +1304,+64,   +17,1500.000000,0.000000,0.000000,    +1,)
Issuing EMC_TASK_PLAN_EXECUTE --         (  +509,+280,   +18,S1000,)
emcTaskPlanLevel() returned 0
NML_INTERP_LIST(0x56315e1e9120)::append(nml_msg_ptr{size=56,type=EMC_SPINDLE_SPEED}) : list_size=1, line_number=0
emcTaskPlanExecute(S1000) returned 0
emcTaskPlanLevel() returned 0
NML_INTERP_LIST(0x56315e1e9120)::get(): {size=56, type=EMC_SPINDLE_SPEED}, list_size=0
emcTaskPlanLevel() returned 0
Issuing EMC_SPINDLE_SPEED --     ( +1316,+56,    +0,0.000000,0.000000,0.000000,)
mdi_execute_hook: MDI command 'S1000' done (remaining: 0)
```

satiowadahc, 2022-04-14:
in task and motion the same set of commands is called in each case... Is there RS274 debugging I'm not seeing?

satiowadahc, 2022-04-14:
Interesting... https://github.com/LinuxCNC/linuxcnc/blob/7d8e52e767753acf0af1f33acc37ae53b3633aaf/src/emc/task/emctaskmain.cc#L36

Maybe Should emc_nml check the status of the spindle before modifying?

satiowadahc, 2022-04-16:
The More I look at this, the more I feel like emcSendCommand in emcmodule should actually be interp_list.append().

andypugh, 2022-04-18:
Are you able to go back to 2.7 (prior to multispindle support) and see if the behaviour was as-expected there?

zultron, 2022-04-18:
IIRC @robEllenberg and I have worked on this code, but I'd have to go back and look to remember all this.  I asked what he remembers off the top of his head:

> I wonder if that ever behaved in that way. My guess from the description is that the direct spindle
> command is not setting the interp "S" word, and M3 is (correctly) reusing the previous interpreter state.
>
> In effect what they want is for `spindle_on(1000)` to act like `mdi("S1000")`
>
> I think the spindle commands in the python API should be treated like manual overrides vs MDI shortcuts

satiowadahc, 2022-04-22:
https://discord.com/channels/570415950495481859/570415951020032013/966894229437706270

User reporting gmoccapy has random spindle speeds. Either 0.25x previous or infinity.

andypugh, 2022-04-27:
I think that the discord report sounds like a different thing? 

If this is what zultron is hinting at, it isn't an easy thing to address as canonical commands do not necessarily feed back into the interpreter internal state.

LJspice, 2022-04-28:
@andypugh that was my comment on discord - I will try and reproduce the issue this weekend and give proper details

satiowadahc, 2024-01-08:
Ran into this one again, pulling my hair out, and found a comment back to this issue. Maybe time to try and work on this again.

satiowadahc, 2024-01-08:
I was hoping this patched it but nope. https://github.com/LinuxCNC/linuxcnc/pull/2718

andypugh, 2024-01-08:
The starting post here is doing a fairly weird thing, in that it is combining spindle commands through the Python interface with MDI commands. 

I don't see a good reason for this not to work, but I am curious if the behaviour is more as-expected if it is _all_ MDI ("M3 S1000") or all Python?