LinuxCNC/linuxcnc #1717: new asciidoc syntax problem

SebKuzminsky, 2022-04-20:
An asciidoc syntax problem was introduced in @smoe 's commit fb3e8fbc31f871eb94356e7a758a05dd8f75e444:
[image]

The commit before is wrong, but in a different way:
[image]

I'm not sure what the correct asciidoc syntax should be here, but it's neither of these two ;-)

I'm also not sure if any other new bugs were introduced among all the fixes in fb3e8fbc31f871eb94356e7a758a05dd8f75e444.

smoe, 2022-04-20:
Ouch. Thank you. I am also not sure and prior to the release I/everyone should read through the docs for obvious mayhem. I am confident that only whitespace was touched. And the "+"s, I think they can just go. Will check.

silopolis, 2022-04-20:
Le mer. 20 avr. 2022 à 17:13, Sebastian Kuzminsky ***@***.***>
a écrit :

> An asciidoc syntax problem was introduced in @smoe
> <https://github.com/smoe> 's commit fb3e8fb
> <https://github.com/LinuxCNC/linuxcnc/commit/fb3e8fbc31f871eb94356e7a758a05dd8f75e444>
> :
>
> The commit before is wrong, but in a different way:
>
> I'm not sure what the correct asciidoc syntax should be here, but it's
> neither of these two ;-)
>
Have these and others fixed in
https://github.com/silopolis/linuxcnc/tree/docs/fr_sync_en_fixes

silopolis, 2022-04-20:
Le mer. 20 avr. 2022 à 17:34, Steffen Möller ***@***.***> a
écrit :

> Ouch. Thank you. I am also not sure and prior to the release I/everyone
> should read through the docs for obvious mayhem. I am confident that only
> whitespace was touched. And the "+"s, I think they can just go. Will check.
>
That was indeed an indentation issue where linked blocks (with '+')
shouldn't be indented in source and nevertheless inherit formatting (and so
indentation) in output.

Linking is done with a single '+' alone at the beginning of a line between
two consecutive blocks and allow, as an example, to extend the content of a
list entry to several paragraphs, code blocks, etc.

smoe, 2022-04-25:
I am assigned - great :) I do not really know what this means, though. My understanding is that this problem will go away once we merge @silopolis ' changes to the docs-devel branch. And that is nigh, I think :) I'll then check and drop a comment here and/or craft a follow-up patch.

c-morley, 2022-05-14:
I notice that the line breaks were removed in the master version of docs. I noticed in the qtvcp docs.
You can see the difference between the 2.8 and master HTML website docs.
Was this on purpose?
I very strongly dislike long sentences jammed together and put a fair amount of effort into breaking lines up.
I used ' +' i could use some other technique if required.
Thanks

smoe, 2022-05-14:
Firstly, I may need to apologize, I admit to have removed "+"s while syncing between languages. Technically, once the documentation is synced (it is now) we can introduce them again.

My removal of "+"s likely broke something. I just checked, and yes, some newlines need to be added back in. Other removals are just fine (for my reading style). There are quite a number of changes that I could think of to attempt to improve readability without shortening lines - would you allow @silopolis (who likely is the one drafting a style guide) and myself in coordination with you to brainstorm over your qtvcp documentation? It is a very nice (and long) document. I like it.

c-morley, 2022-05-15:
Thanks for your response and your hard work on the docs.
I certainly am open to new ideas but long endless paragraphs (IMHO) make it harder to find the pertinent info.
I'll admit it looks nicer on the page though.
I'm interested in your ideas.

smoe, 2022-05-15:
Thank you for your constructive and kind reply.

My idea is to treat the documentation like a dialog. The first sentence of a paragraph introduces a fact or concept - then you have an (imaginary) look into the eyes of the individual you are (imaginary) talking to, so you can assess if that individual is interested to hear the remainder of that paragraph.

The reader will continue reading that paragraph or jump to the next paragraph and read that next first sentence and so on. This allows to adjust the reading experience to the experience/interest the reader has. Between those paragraphs then also is a bit of extra white space.

I tried to format my reply along that basic concept. Other bits address consistency in spelling (QtVCP/Qtvcp/...), separation of code and narration.

silopolis, 2022-05-15:
aouch, that looks a little bulky now indeed 😬
Let me fix this for you...

Le dim. 15 mai 2022 à 18:31, c-morley ***@***.***> a écrit :

> Thanks for your response and your hard work on the docs.
>
And thank you for such quality docs on QtVCP ! Wish we had the same level
through out...

> I certainly am open to new ideas but long endless paragraphs (IMHO) make
> it harder to find the pertinent info.
>
Absolutely !

My PoV is that paragraphs separation should both follow semantics of the
content as well as, indeed, help identify/spot the infos.

silopolis, 2022-05-15:
Le dim. 15 mai 2022 à 18:43, Steffen Möller ***@***.***> a
écrit :

> Thank you for your constructive and kind reply.
>
> My idea is to treat the documentation like a dialog. The first sentence of
> a paragraph introduces a fact or concept - then you have an (imaginary) [...]
I like it and it's nicely put 👌

petterreinholdtsen, 2022-07-28:
This seem to be fixed in https://github.com/LinuxCNC/linuxcnc/blob/master/docs/src/config/ini-config.adoc .  Do you agree?  If so, this issue can be closed as resolved.  I lack the access to do so.

smoe, 2022-07-28:
I am happy with the current version. Have politely undone my assignment, should not have. Sorry.

SebKuzminsky, 2022-09-12:
This looks fixed in master now, thanks everyone!

closed 2022-09-12 by SebKuzminsky

silopolis, 2022-09-12:
Le lun. 12 sept. 2022 à 19:09, Sebastian Kuzminsky ***@***.***>
a écrit :

> This looks fixed in master now, thanks everyone!
>

This should now be all fine and hopefully improved in #1742