LinuxCNC/linuxcnc #1728: Lintian Errors

jethornton, 2022-05-05:
Opening the latest linuxcnc-uspace_2.9.0~pre0.6424.gf826bc699_amd64.deb and there are a bunch of lintian errors.

E: linuxcnc-uspace: malformed-override Unknown tag elevated-privileges in line 8
E: linuxcnc-uspace: malformed-override Unknown tag elevated-privileges in line 9
W: linuxcnc-uspace: executable-not-elf-or-script usr/lib/tcltk/linuxcnc/bin/cbutton.tcl
W: linuxcnc-uspace: groff-message usr/share/man/man9/toggle.9.gz 17: can't open '../man/images/toggle.ps': No such file or directory
W: linuxcnc-uspace: groff-message usr/share/man/man9/toggle2nist.9.gz 26: can't open '../man/images/toggle2nist.ps': No such file or directory
W: linuxcnc-uspace: mismatched-override groff-message usr/share/man/man9/toggle.9.gz (line 1) 17: can't open '../man/images/toggle.ps': No such file or directory
W: linuxcnc-uspace: mismatched-override groff-message usr/share/man/man9/toggle2nist.9.gz (line 1) 26: can't open '../man/images/toggle2nist.ps': No such file or directory
W: linuxcnc-uspace: no-manual-page usr/bin/melfagui
W: linuxcnc-uspace: no-manual-page usr/bin/millturngui
W: linuxcnc-uspace: no-manual-page usr/bin/tool_mmap_read
W: linuxcnc-uspace: no-manual-page usr/bin/tool_watch
W: linuxcnc-uspace: setuid-binary usr/bin/linuxcnc_module_helper 4755 root/root
W: linuxcnc-uspace: setuid-binary usr/bin/rtapi_app 4755 root/root
W: linuxcnc-uspace: spelling-error-in-changelog "Allow to" "Allow one to"
W: linuxcnc-uspace: spelling-error-in-changelog "allows to" "allows one to"
W: linuxcnc-uspace: spelling-error-in-changelog X X (duplicate word) X
W: linuxcnc-uspace: spelling-error-in-changelog accelleration acceleration
W: linuxcnc-uspace: spelling-error-in-changelog accessable accessible
W: linuxcnc-uspace: spelling-error-in-changelog buitin builtin
W: linuxcnc-uspace: spelling-error-in-changelog bultin builtin
W: linuxcnc-uspace: spelling-error-in-changelog calulation calculation
W: linuxcnc-uspace: spelling-error-in-changelog commited committed
W: linuxcnc-uspace: spelling-error-in-changelog compatibilty compatibility
W: linuxcnc-uspace: spelling-error-in-changelog compatiblity compatibility
W: linuxcnc-uspace: spelling-error-in-changelog consistancy consistency
W: linuxcnc-uspace: spelling-error-in-changelog descritption description
W: linuxcnc-uspace: spelling-error-in-changelog distrubution distribution
W: linuxcnc-uspace: spelling-error-in-changelog documenation documentation
W: linuxcnc-uspace: spelling-error-in-changelog documention documentation
W: linuxcnc-uspace: spelling-error-in-changelog dont don't
W: linuxcnc-uspace: spelling-error-in-changelog eror error
W: linuxcnc-uspace: spelling-error-in-changelog exemple example
W: linuxcnc-uspace: spelling-error-in-changelog exising existing
W: linuxcnc-uspace: spelling-error-in-changelog follwing following
W: linuxcnc-uspace: spelling-error-in-changelog formated formatted
W: linuxcnc-uspace: spelling-error-in-changelog guage gauge
W: linuxcnc-uspace: spelling-error-in-changelog inadvertantly inadvertently
W: linuxcnc-uspace: spelling-error-in-changelog informtion information
W: linuxcnc-uspace: spelling-error-in-changelog interpeter interpreter
W: linuxcnc-uspace: spelling-error-in-changelog intial initial
W: linuxcnc-uspace: spelling-error-in-changelog intialization initialization
W: linuxcnc-uspace: spelling-error-in-changelog lables labels
W: linuxcnc-uspace: spelling-error-in-changelog libray library
W: linuxcnc-uspace: spelling-error-in-changelog menues menus
W: linuxcnc-uspace: spelling-error-in-changelog missmatch mismatch
W: linuxcnc-uspace: spelling-error-in-changelog occassionally occasionally
W: linuxcnc-uspace: spelling-error-in-changelog occurences occurrences
W: linuxcnc-uspace: spelling-error-in-changelog openning opening
W: linuxcnc-uspace: spelling-error-in-changelog overriden overridden
W: linuxcnc-uspace: spelling-error-in-changelog paramater parameter
W: linuxcnc-uspace: spelling-error-in-changelog possibilty possibility
W: linuxcnc-uspace: spelling-error-in-changelog relatd related
W: linuxcnc-uspace: spelling-error-in-changelog remebered remembered
W: linuxcnc-uspace: spelling-error-in-changelog requiere require
W: linuxcnc-uspace: spelling-error-in-changelog runnning running
W: linuxcnc-uspace: spelling-error-in-changelog substract subtract
W: linuxcnc-uspace: spelling-error-in-changelog surounding surrounding
W: linuxcnc-uspace: spelling-error-in-changelog temparary temporary
W: linuxcnc-uspace: spelling-error-in-changelog transistion transition
W: linuxcnc-uspace: spelling-error-in-changelog unuseable unusable
W: linuxcnc-uspace: spelling-error-in-changelog upto up to
N: 3 hints overridden (3 warnings); 13 unused overrides

Lintian finished with exit status 0

hansu, 2022-05-06:
> W: linuxcnc-uspace: executable-not-elf-or-script usr/lib/tcltk/linuxcnc/bin/cbutton.tcl

For this i wonder why it is detected as an executable. It doesn't have the executable file attribute. Is this only because it is located in `bin/`?

>W: linuxcnc-uspace: groff-message usr/share/man/man9/toggle.9.gz 17: can't open '../man/images/toggle.ps': No such file or directory

For this and following three: The ps images are only needed for the HTMLs and PDFs during the build process, so not needed here. To copy it only to suppress the warnings doesn't really make sense, right?

smoe, 2022-05-09:
I admit to be a bit surprised about those lintian errors.  Anyway, many thanks for the pointer, much appreciated. We should get those fixed with the next upload, at least the spelling errors :)

petterreinholdtsen, 2022-07-28:
If debian/update-dch-from-git is used to generate debian/changelog, fixing the typos might be hard, as it generate it from the git history...

andypugh, 2025-11-24:
I am baffled by the current batch of lintian errors:
http://buildbot2.highlab.com/buildbot/#/builders/2/builds/2394/steps/7/logs/warnings__183_
As when I uncompress the files (as they exist on my machine) I can't see any of them.