LinuxCNC/linuxcnc #1737: tutorial error? loadrt debounce -> ERROR: no channels configured

smoe, 2022-05-09:
I am not sure if this is a problem of the 2.9 Debian packaging or of the tutorial. On http://linuxcnc.org/docs/devel/html/hal/tutorial.html it motivates to run "loadrt debounce" but this seems not to work:
```
halcmd: loadrt debounce 
DEBOUNCE: ERROR: no channels configured
debounce: rtapi_app_main: Operation not permitted (-1)
<stdin>:2: waitpid failed /usr/bin/rtapi_app debounce
<stdin>:2: /usr/bin/rtapi_app exited without becoming ready
<stdin>:2: insmod for debounce failed, returned -1
```
other modules work fine like
```
halcmd: loadrt siggen
Note: Using POSIX non-realtime
```
File permissions seem fine, the suid flag is set:
```
$ ls -l /usr/bin/rtapi_app
-rwsr-xr-x 1 root root 125688  6. Apr 16:23 /usr/bin/rtapi_app
```
I have no idea about how to test this.

hansu, 2022-05-09:
According to the [docs](http://linuxcnc.org/docs/devel/html/man/man9/debounce.9.html) it needs a cfg value:

`loadrt debounce cfg=size[,size,...]`

smoe, 2022-05-09:
Works! Thanks! Someone should fix the tutorial (and maybe also give a pointer to that man page :o) ) .

hansu, 2022-06-10:
Fixed in 2a98b4962fd872436f8d57411bf2dcd77200444d

closed 2022-06-10 by hansu