LinuxCNC/linuxcnc #1738: Rotary axis accel exceeded in G93 and combined rotary / linear moves

andypugh, 2022-05-11:
Reported on the forum, and confirmed in a sim config.

To reproduce: Open sim-axis-axis9axis and run the attached .ngc file while running the attached halscope config

The max accel for the rotary axis in the sim config is set to 1200deg/s2 however the sim shows an accel command of 1800 at the point where the cursor is) 

[image]
[Archive.zip](https://github.com/LinuxCNC/linuxcnc/files/8672588/Archive.zip)

mozmck, 2023-03-13:
Looks like this might be something addressed in pull #581?

andypugh, 2023-05-08:
Also still present in master, so apparently not fixed by https://github.com/LinuxCNC/linuxcnc/commit/10c3678ab069f856532e5512a93bfaffcf930491

andypugh, 2023-05-09:
Also not fixed by https://github.com/LinuxCNC/linuxcnc/commit/b244b375ddee0fc684d9ce8341d81f92ad366b42 (disappointingly)