LinuxCNC/linuxcnc #1747: Default sserial port 1 setup is incorrect

pcw-mesa, 2022-05-22:
Sserial port setup is incorrect for systems that have port 1 if the mode
string for port 1 is not specified but port 0's mode string is specified

If port 0's mode string is specified but port 1s is not, all sserial channels
of port 1 are enabled, whether they have connected hardware or not.

This was tested in master

andypugh, 2023-05-22:
I have tried to test this with the 5i25_7i74_4b.bit file (which has 2 ports, 4xSSERIAL and 4xSSERIALB) but seem to have a problem

````
andypugh@rm-one:~$ halrun
halcmd: loadrt hostmot2
Note: Using POSIX realtime
hm2: loading Mesa HostMot2 driver version 0.15
halcmd: loadrt hm2_pci config="sserial_port_0=00000000"
hm2_pci: loading Mesa AnyIO HostMot2 driver version 0.7
hm2_pci: discovered 5i25 at 0000:03:00.0
hm2/hm2_5i25.0: Low Level init 0.15
hm2/hm2_5i25.0: Smart Serial Firmware Version 43
Board hm2_5i25.0.7i69.0.0 Hardware Mode 0 = mesa_pinout
Board hm2_5i25.0.7i69.0.0 Hardware Mode 1 = module_rack_pinout
Board hm2_5i25.0.7i69.0.0 Software Mode 0 = bidir
Board hm2_5i25.0.7i69.0.0 Software Mode 1 = input_only
Board hm2_5i25.0.7i69.0.0 Software Mode 2 = output_only
Board hm2_5i25.0.7i69.0.0 Software Mode 3 = 24in_24out
Board hm2_5i25.0.7i69.0.0 Software Mode 4 = bidir_encoder
hm2/hm2_5i25.0: found duplicate Module Descriptor for Smart Serial Interface (inconsistent firmware), not loading driver
hm2/hm2_5i25.0: failed to parse Module Descriptor 4
hm2_5i25.0: board fails HM2 registration
RTAPI_PCI: Unmapped 65536 bytes at 0x7f8379d5e000
Driver probe function failed!
hm2_pci: error registering PCI driver
hm2_pci: rtapi_app_main: Operation not permitted (-1)
<stdin>:2: waitpid failed /home/andypugh/linuxcnc-dev/bin/rtapi_app hm2_pci
<stdin>:2: /home/andypugh/linuxcnc-dev/bin/rtapi_app exited without becoming ready
<stdin>:2: insmod for hm2_pci failed, returned -1
````

I think that there are two questions here: Is this a good test of this bug? And have I found a new bug?

pcw-mesa, 2023-05-22:
That may be another bug, I don't think I tested mixed (sserial+sserialb) 
configurations

Probably better to test with  a stock sserial+sserial config like this

http://freeby.mesanet.com/5i25_7i74x1_4_4d.bit

First 7I74 channels are port 0, second are port 1

andypugh, 2023-05-22:
I seem to have managed to find a different version of the problem, by specifying a port1 string but not port 0. 
(I turned off the duplicate module check) 
````
halrun
halcmd: loadrt hostmot2
Note: Using POSIX realtime
hm2: loading Mesa HostMot2 driver version 0.15
halcmd: loadrt hm2_pci config="sserial_port_1=xxxxxxxx"
hm2_pci: loading Mesa AnyIO HostMot2 driver version 0.7
hm2_pci: discovered 5i25 at 0000:03:00.0
hm2/hm2_5i25.0: Low Level init 0.15
hm2/hm2_5i25.0: 34 I/O Pins used:
hm2/hm2_5i25.0:     IO Pin 000 (P3-01): Smart Serial Interface #0, pin rx0 (Input)
hm2/hm2_5i25.0:     IO Pin 001 (P3-14): Smart Serial Interface #0, pin rx1 (Input)
hm2/hm2_5i25.0:     IO Pin 002 (P3-02): Smart Serial Interface #0, pin rx2 (Input)
hm2/hm2_5i25.0:     IO Pin 003 (P3-15): Smart Serial Interface #0, pin rx3 (Input)
hm2/hm2_5i25.0:     IO Pin 004 (P3-03): Smart Serial Interface #0, pin tx0 (Output)
hm2/hm2_5i25.0:     IO Pin 005 (P3-16): Smart Serial Interface #0, pin tx1 (Output)
hm2/hm2_5i25.0:     IO Pin 006 (P3-04): Smart Serial Interface #0, pin tx2 (Output)
hm2/hm2_5i25.0:     IO Pin 007 (P3-17): Smart Serial Interface #0, pin tx3 (Output)
hm2/hm2_5i25.0:     IO Pin 008 (P3-05): Smart Serial Interface #0, pin rx0 (Input)
hm2/hm2_5i25.0:     IO Pin 009 (P3-06): Smart Serial Interface #0, pin rx1 (Input)
hm2/hm2_5i25.0:     IO Pin 010 (P3-07): Smart Serial Interface #0, pin rx2 (Input)
hm2/hm2_5i25.0:     IO Pin 011 (P3-08): Smart Serial Interface #0, pin rx3 (Input)
hm2/hm2_5i25.0:     IO Pin 012 (P3-09): Smart Serial Interface #0, pin tx0 (Output)
hm2/hm2_5i25.0:     IO Pin 013 (P3-10): Smart Serial Interface #0, pin tx1 (Output)
hm2/hm2_5i25.0:     IO Pin 014 (P3-11): Smart Serial Interface #0, pin tx2 (Output)
hm2/hm2_5i25.0:     IO Pin 015 (P3-12): Smart Serial Interface #0, pin tx3 (Output)
hm2/hm2_5i25.0:     IO Pin 016 (P3-13): Smart Serial Interface #0, pin txen3 (Output)
hm2/hm2_5i25.0:     IO Pin 017 (P2-01): IOPort
hm2/hm2_5i25.0:     IO Pin 018 (P2-14): IOPort
hm2/hm2_5i25.0:     IO Pin 019 (P2-02): IOPort
hm2/hm2_5i25.0:     IO Pin 020 (P2-15): IOPort
hm2/hm2_5i25.0:     IO Pin 021 (P2-03): IOPort
hm2/hm2_5i25.0:     IO Pin 022 (P2-16): IOPort
hm2/hm2_5i25.0:     IO Pin 023 (P2-04): IOPort
hm2/hm2_5i25.0:     IO Pin 024 (P2-17): IOPort
hm2/hm2_5i25.0:     IO Pin 025 (P2-05): IOPort
hm2/hm2_5i25.0:     IO Pin 026 (P2-06): IOPort
hm2/hm2_5i25.0:     IO Pin 027 (P2-07): IOPort
hm2/hm2_5i25.0:     IO Pin 028 (P2-08): IOPort
hm2/hm2_5i25.0:     IO Pin 029 (P2-09): IOPort
hm2/hm2_5i25.0:     IO Pin 030 (P2-10): IOPort
hm2/hm2_5i25.0:     IO Pin 031 (P2-11): IOPort
hm2/hm2_5i25.0:     IO Pin 032 (P2-12): IOPort
hm2/hm2_5i25.0:     IO Pin 033 (P2-13): IOPort
hm2/hm2_5i25.0: registered
hm2_5i25.0: initialized AnyIO board at 0000:03:00.0
````

There is a 7i96 connected, but it doesn't seem to have been detected this time round.

andypugh, 2023-05-22:
With the new bitfile:
````
hm2/hm2_5i25.0: inconsistent Module Descriptor for Smart Serial Interface, not loading driver
hm2/hm2_5i25.0:     Version = 0, expected 0
hm2/hm2_5i25.0:     NumRegisters = 6, expected 6
hm2/hm2_5i25.0:     InstanceStride = 0x00000010, expected 0x00000040
hm2/hm2_5i25.0:     MultipleRegisters = 0x0000003C, expected 0x0000003C
````

pcw-mesa, 2023-05-22:
Sorry, if you re-fetch the bitfile it should be fixed.

andypugh, 2023-05-23:
This goes deeper than I first thought. 
There is code that actively bypasses the effect of num_sserials in the INI, instead setting the number to the highest port that has a sserial_port_N string in the config. 
But the docs state that the default is 00000000. So something needs to change there to be consistent. 

But, in addition to this, the system does not actually seem to be detecting any devices on port 1, only on port 0. 
(I think I have fixed the not-setting-to-ioport problem, but now it is not detecting hardware) 

Are you sure that the bitfile is right? the driver seems to be doing the right things to addresses that match the regmap. 

````
andypugh@rm-one:~/linuxcnc-dev/src$ halrun -I s.hal
Note: Using POSIX realtime
hm2: loading Mesa HostMot2 driver version 0.15
hm2_pci: loading Mesa AnyIO HostMot2 driver version 0.7
hm2_pci: discovered 5i25 at 0000:03:00.0
hm2/hm2_5i25.0: Low Level init 0.15
hm2/hm2_5i25.0: sserial_num_instances = 2
smalloc_up: shmem available 1046080
hm2/hm2_5i25.0: port 0 sec unit = 0, sec pin = 1 mode=1
hm2/hm2_5i25.0: port 0 sec unit = 0, sec pin = 2 mode=2
hm2/hm2_5i25.0: port 0 sec unit = 0, sec pin = 3 mode=3
hm2/hm2_5i25.0: port 0 sec unit = 0, sec pin = 4 mode=4
hm2/hm2_5i25.0: port 0 sec unit = 0, sec pin = 1 mode=1
hm2/hm2_5i25.0: port 0 sec unit = 0, sec pin = 2 mode=2
hm2/hm2_5i25.0: port 0 sec unit = 0, sec pin = 3 mode=3
hm2/hm2_5i25.0: port 0 sec unit = 0, sec pin = 4 mode=4
hm2/hm2_5i25.0: port 0 sec unit = 1, sec pin = 1 mode=1
hm2/hm2_5i25.0: port 0 sec unit = 1, sec pin = 2 mode=2
hm2/hm2_5i25.0: port 0 sec unit = 1, sec pin = 3 mode=3
hm2/hm2_5i25.0: port 0 sec unit = 1, sec pin = 4 mode=4
hm2/hm2_5i25.0: port 0 sec unit = 1, sec pin = 1 mode=1
hm2/hm2_5i25.0: port 0 sec unit = 1, sec pin = 2 mode=2
hm2/hm2_5i25.0: port 0 sec unit = 1, sec pin = 3 mode=3
hm2/hm2_5i25.0: port 0 sec unit = 1, sec pin = 4 mode=4
hm2/hm2_5i25.0: port 0 sec unit = 1, sec pin = 4 mode=4
hm2/hm2_5i25.0: Smart Serial Firmware Version 43
hm2/hm2_5i25.0: Start-mode = f01
hm2/hm2_5i25.0: stopstart writing 1000000 to 5d00
hm2/hm2_5i25.0: Start-mode = f03
hm2/hm2_5i25.0: stopstart writing 2000000 to 5d04
hm2/hm2_5i25.0: Start-mode = f07
hm2/hm2_5i25.0: stopstart writing 3000000 to 5d08
hm2/hm2_5i25.0: Start-mode = f0f
hm2/hm2_5i25.0: stopstart writing 4000000 to 5d0c
hm2/hm2_5i25.0: stopstart writing f0f to 5b00
hm2/hm2_5i25.0: Inst 0 Chan 0 Addr 5e00 User0 = 130000fa
hm2/hm2_5i25.0: Inst 0 Chan 0 Addr 5f00 User1 = 33374937
hm2/hm2_5i25.0: Inst 0 Chan 0 Addr 6000 User2 = aaa6a8ea
hm2/hm2_5i25.0: Inst 0 Chan 1 Addr 5e04 User0 = 0
hm2/hm2_5i25.0: Inst 0 Chan 1 Addr 5f04 User1 = 0
hm2/hm2_5i25.0: Inst 0 Chan 1 Addr 6004 User2 = 0
hm2/hm2_5i25.0: Masking pin 1
hm2/hm2_5i25.0: Masking pin 5
hm2/hm2_5i25.0: Inst 0 Chan 2 Addr 5e08 User0 = 0
hm2/hm2_5i25.0: Inst 0 Chan 2 Addr 5f08 User1 = 0
hm2/hm2_5i25.0: Inst 0 Chan 2 Addr 6008 User2 = 0
hm2/hm2_5i25.0: Masking pin 2
hm2/hm2_5i25.0: Masking pin 6
hm2/hm2_5i25.0: Inst 0 Chan 3 Addr 5e0c User0 = 0
hm2/hm2_5i25.0: Inst 0 Chan 3 Addr 5f0c User1 = 0
hm2/hm2_5i25.0: Inst 0 Chan 3 Addr 600c User2 = 0
hm2/hm2_5i25.0: Masking pin 3
hm2/hm2_5i25.0: Masking pin 7
...
hm2/hm2_5i25.0: Instance 0, channel 0 / 0
hm2/hm2_5i25.0: reg_cs_addr = 5d00
hm2/hm2_5i25.0: rw_addr[0] = 5e00
hm2/hm2_5i25.0: rw_addr[1] = 5f00
hm2/hm2_5i25.0: rw_addr[2] = 6000
hm2/hm2_5i25.0: rw_addr[3] = 6100
[... 70 lines cut ...]
hm2/hm2_5i25.0:     IO Pin 010 (P3-07): IOPort
hm2/hm2_5i25.0:     IO Pin 011 (P3-08): IOPort
hm2/hm2_5i25.0:     IO Pin 012 (P3-09): IOPort
hm2/hm2_5i25.0:     IO Pin 013 (P3-10): IOPort
hm2/hm2_5i25.0:     IO Pin 014 (P3-11): IOPort
hm2/hm2_5i25.0:     IO Pin 015 (P3-12): IOPort
hm2/hm2_5i25.0:     IO Pin 016 (P3-13): IOPort
hm2/hm2_5i25.0:     IO Pin 017 (P2-01): IOPort
hm2/hm2_5i25.0:     IO Pin 018 (P2-14): IOPort
hm2/hm2_5i25.0:     IO Pin 019 (P2-02): IOPort
hm2/hm2_5i25.0:     IO Pin 020 (P2-15): IOPort
hm2/hm2_5i25.0:     IO Pin 021 (P2-03): IOPort
hm2/hm2_5i25.0:     IO Pin 022 (P2-16): IOPort
hm2/hm2_5i25.0:     IO Pin 023 (P2-04): IOPort
hm2/hm2_5i25.0:     IO Pin 024 (P2-17): IOPort
hm2/hm2_5i25.0:     IO Pin 025 (P2-05): IOPort
hm2/hm2_5i25.0:     IO Pin 026 (P2-06): IOPort
hm2/hm2_5i25.0:     IO Pin 027 (P2-07): IOPort
hm2/hm2_5i25.0:     IO Pin 028 (P2-08): IOPort
hm2/hm2_5i25.0:     IO Pin 029 (P2-09): IOPort
hm2/hm2_5i25.0:     IO Pin 030 (P2-10): IOPort
hm2/hm2_5i25.0:     IO Pin 031 (P2-11): IOPort
hm2/hm2_5i25.0:     IO Pin 032 (P2-12): IOPort
hm2/hm2_5i25.0:     IO Pin 033 (P2-13): IOPort
````

pcw-mesa, 2023-05-23:
Did you have a sserial_port_1=something on the commandline? 
sserial port_1 has never worked without this

pcw-mesa, 2023-05-24:
I don't think anything is wrong with that configuration, but you might try an older
known-to-work configuration like 5i25_7i77_7i74 at least to test port 1
(7I74 on the second 5I25 connector has 8 channels all port 1)

andypugh, 2023-05-24:
I don't have the second connector on the 5i25 broken out, so the 5i25_7i77_7i74 config isn't trivial to use. 
Can you get anything to connect to sserial port 1 with the bitfile above if you add a sserial_port_1 entry?

pcw-mesa, 2023-05-24:
I dont have a permanent LinuxCNC setup for PCI cards so I will have to set one up
Do you have an Ethernet FPGA card that can connect to the 7I74?

pcw-mesa, 2023-05-24:
http://freeby.mesanet.com/7i96_7i74d2.bit
7I74 on expansion with 2 port 1 channels (0,1)
7I96 TB2  sserial is port 0 channel 0

andypugh, 2023-05-30:
I think that this is fixed by cbdb967b0d7983f209a3c2f362e1778a30a35733
As a bonus the behaviour now matches the docs, in that the ports default to 0000000 if no sserial_port string is used, ie port1 will be polled even if there is no sserial_port_1 configurator.

closed 2023-05-30 by andypugh