LinuxCNC/linuxcnc #1749: Compile Error

jack9603301, 2022-05-29:
## Here are the steps I follow to reproduce the issue:

Compile using Gentoo ebuild as follows:

```
# Copyright 1999-2015 Gentoo Foundation
# Distributed under the terms of the GNU General Public License v2
# $Header: $

EAPI="7"

PYTHON_COMPAT=( python2_7 )

inherit autotools eutils flag-o-matic multilib python-single-r1

DESCRIPTION="LinuxCNC "
HOMEPAGE="http://linuxcnc.org/"
#SRC_URI="mirror://sourceforge/gentoocnc/distfiles/${P}.tar.gz"
#SRC_URI="https://github.com/LinuxCNC/linuxcnc/archive/2.7.zip"
SRC_URI="https://github.com/LinuxCNC/linuxcnc/archive/refs/tags/v2.8.2.tar.gz"

S="${WORKDIR}"/linuxcnc-2.8.2/src

LICENSE="LGPL-3"
SLOT="linuxcnc-2.8"
KEYWORDS="~amd64 ~x86"
IUSE="+python +uspace +X +gtk -rt gstreamer modbus simulator usb"
# TODO: add shmdrv use flag

# --disable-python uses python anyways and fails so python is a required use flag
REQUIRED_USE="
    python
    python? ( ${PYTHON_REQUIRED_USE} )
	rt? ( !simulator )
	"

DEPEND="dev-lang/tcl
	dev-lang/tk
	dev-tcltk/tkimg
	dev-tcltk/tclx
	dev-libs/boost[python]
	modbus? ( <dev-libs/libmodbus-3.1 )
	dev-tcltk/bwidget
	gtk? ( x11-libs/gtk+ )
    gstreamer? (
        dev-python/gst-python:0.10
        media-libs/gst-plugins-base:0.10
    )
	|| (
        net-analyzer/netcat
        net-analyzer/netcat6
    )
	x11-libs/libXinerama
	usb? ( virtual/libusb )
	dev-lang/python:3.10[tk]
	$(python_gen_cond_dep 'dev-python/lxml[${PYTHON_USEDEP}]')
	$(python_gen_cond_dep 'dev-python/numpy[${PYTHON_USEDEP}]')
	$(python_gen_cond_dep 'dev-python/configobj[${PYTHON_USEDEP}]')
	$(python_gen_cond_dep 'dev-python/pillow[tk,${PYTHON_USEDEP}]')
	media-libs/mesa
    x11-apps/mesa-progs
	X? (
        x11-libs/libXaw
        $(python_gen_cond_dep 'dev-python/python-xlib[${PYTHON_USEDEP}]')
    )
	rt? ( sys-kernel/rt-sources )
	virtual/opengl
	virtual/glu
	${PYTHON_DEPS}
	sys-devel/gettext
[... 151 lines cut ...]
    inlined from ‘int Interp::convert_threading_cycle(block_pointer, setup_pointer, double, double, double)’ at emc/rs274ngc/interp_convert.cc:4720:5:
/usr/include/bits/string_fortified.h:95:34: warning: ‘char* __builtin_strncpy(char*, const char*, long unsigned int)’ output truncated copying 80 bytes from a string of length 89 [-Wstringop-truncation]
   95 |   return __builtin___strncpy_chk (__dest, __src, __len,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
   96 |                                   __glibc_objsize (__dest));
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~

```

## It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): gentoo linux
 * I am using this kernel version (shown by `uname -a`): Linux jack-B250-HD3 5.17.1-zen1 #5 ZEN SMP PREEMPT Mon May 16 23:19:10 CST 2022 x86_64 Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz GenuineIntel GNU/Linux
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): 2.8.2 tag
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):

jack9603301, 2022-05-29:
It seems that only master can be built under py3

rene-dev, 2022-05-29:
yes, only master supports py3, and py3 only.

jack9603301, 2022-05-29:
However, I found the following problems. I don't know if I need to install any additional modules:

```
┌─(/var/db/repos/gentoo_overlay_local)─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────(jack@jack-B250-HD3:pts/4)─┐
└─(19:24:02 on main)──> linuxcnc                                                                                                                                                                                             ──(日,5月29)─┘
LINUXCNC - 2.9.0~pre0
Machine configuration directory is '/home/jack/linuxcnc/configs/sim.axis'
Machine configuration file is 'axis.ini'
can't find package Linuxcnc
    while executing
"package require Linuxcnc "
    (file "/usr/share/linuxcnc/hallib/check_config.tcl" line 187)
check_config validation failed
LinuxCNC terminated with an error.  You can find more information in the log:
    /home/jack/linuxcnc_debug.txt
and
    /home/jack/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
┌─(/var/db/repos/gentoo_overlay_local)──────────────────────────────────────
```

rene-dev, 2022-05-29:
can't find package Linuxcnc
means python doesnt find the linuxcnc module. either a path problem, or a python version mismatch.

jack9603301, 2022-05-29:
It seems that the problem is caused by the fact that the TCL package path of linuxcnc is not set to the environment variable