LinuxCNC/linuxcnc #1752: Wrong translation license

comradekingu, 2022-06-04:
https://hosted.weblate.org/projects/linuxcnc/linuxcnc-docs/#information says **GPLv2-only**, whereas
https://github.com/LinuxCNC/linuxcnc/search?q=any+later says **GPLv2-or-later**.

It can be fixed in
https://hosted.weblate.org/settings/linuxcnc/linuxcnc/
https://hosted.weblate.org/settings/linuxcnc/linuxcnc-docs/
https://hosted.weblate.org/settings/linuxcnc/gmocappy/
https://hosted.weblate.org/settings/linuxcnc/linuxcnc-glossary/

hansu, 2022-06-06:
I am not sure if that is correct - https://github.com/LinuxCNC/linuxcnc/blob/master/COPYING says GPL v2

silopolis, 2022-06-06:
Actually, that's none of these but GFDL 1.1

https://github.com/LinuxCNC/linuxcnc/blob/master/docs/src/common/gpld-copyright.adoc

hansu, 2022-06-06:
Good point :+1: 
But only for the docs, right?

silopolis, 2022-06-06:
Le lun. 6 juin 2022 à 21:56, Hans ***@***.***> a écrit :

> Good point 👍
> But only for the docs, right?
>
Yes indeed

petterreinholdtsen, 2022-07-28:
I've updated the license info in Weblate to specify GPLv2+ for linuxcnc and gmoccapy and GFDL 1.1+ for linuxcnc-doc and linuxcnc-glossary.  This issue can be closed as solved.  I lack the access required to do so.

comradekingu, 2022-07-28:
Top stuff.

closed 2022-07-28 by comradekingu