LinuxCNC/linuxcnc #1762: Test all sample configs

andypugh, 2022-06-14:
From a discussion between a sub-set of active developers this evening, we decided that it was important to check that all the sample configs a) Start b) Run and c) Have no serious graphical glitches. 
The reason to check this is that the Master branch is now fully converted to Python 3 and GTK3 and the GUIs are the main area where the updates are likely to have caused issues.

Below is a list of all the sample configs. Please add your name in the "to be tested by" column if you are offering to check a config, and put the result ("pass" or "fail") in the "Result" column. Add any comments (especially in the case of failure) under "Comments".

|Config|To Be Tested By|Result|Comments|
|-------|----------------|------|-----------|
|./attic/dallur-thc/dallur-advanced.ini||| |
|./attic/demo_mazak/demo_mazak.ini||| |
|./by_machine/qtplasmac/qtplasmac_s_imperial.ini||| |
|./by_machine/qtplasmac/qtplasmac_s_metric.ini||| |
|./by_machine/qtplasmac/qtplasmac_l_metric.ini||| |
|./by_machine/qtplasmac/qtplasmac_p_imperial.ini||| |
|./by_machine/qtplasmac/qtplasmac_l_imperial.ini||| |
|./by_machine/qtplasmac/qtplasmac_p_metric.ini||| |
|./by_machine/boss/pid_test.ini||| |
|./by_machine/boss/boss.ini||| |
|./by_machine/scorbot-er-3/scorbot-er-3.ini||| |
|./by_machine/tormach/pcnc-770.ini||| |
|./by_machine/tormach/pcnc-1100-seriesII.ini||| |
|./by_machine/sherline/Sherline4Axis/Sherline4Axis_inch.ini||| |
|./by_machine/sherline/Sherline4Axis/Sherline4Axis_mm.ini||| |
|./by_machine/sherline/SherlineLathe/SherlineLathe_inch.ini||| |
|./by_machine/sherline/SherlineLathe/SherlineLathe_mm.ini||| |
|./by_machine/sherline/Sherline3Axis/Sherline3Axis_inch.ini||| |
|./by_machine/sherline/Sherline3Axis/Sherline3Axis_mm.ini||| |
|./by_machine/cooltool/Uni-fraes-v3.ini||| |
|./by_machine/cooltool/Uni-dreh-g.ini||| |
|./by_machine/cooltool/Uni-dreh.ini||| |
|./by_machine/cooltool/Uni-cnc-set.ini||| |
|./by_machine/cooltool/Basic540_4axes.ini||| |
|./by_machine/cooltool/Uni-mill-cnc.ini||| |
|./by_machine/cooltool/Uni-fraes-h3.ini||| |
|./by_machine/cooltool/Uni-fraes-4.ini||| |
|./by_machine/cooltool/Basic540_3axes.ini||| |
|./by_machine/smithy/1034gecko.ini||| |
|./by_machine/smithy/622.ini||| |
|./by_machine/smithy/1240combined.ini||| |
|./by_machine/smithy/516gecko.ini||| |
|./by_machine/smithy/622leadshine_4axis.ini||| |
|./by_machine/smithy/1240rutex_4axis.ini||| |
|./by_machine/smithy/622keyence.ini||| |
|./by_machine/smithy/622leadshine.ini||| |
|./by_machine/smithy/1240combined_4axis.ini||| |
|./by_machine/smithy/622_4axis.ini||| |
|./by_machine/smithy/622gecko.ini||| |
|./by_machine/smithy/1240combined_mm.ini||| |
|./by_machine/smithy/1240rutex.ini||| |
|./by_machine/smithy/924.ini||| |
|./by_machine/smithy/1240_4axis.ini||| |
|./by_machine/smithy/1240gecko.ini||| |
|./by_machine/smithy/1315.ini||| |
|./by_machine/smithy/1240.ini||| |
|./by_interface/parport/classicladder/demo_step_cl/demo_step_cl.ini||| |
|./by_interface/parport/classicladder/cl-estop/cl-estop.ini||| |
|./by_interface/parport/plasma-thc-sim/plasma_mm.ini||| |
|./by_interface/parport/gantry/gantry_joints.ini||| |
|./by_interface/parport/gantry/gantry.ini||| |
|./by_interface/parport/etch-servo/etch.ini||| |
|./by_interface/parport/stepper-xyza/inch.ini||| |
|./by_interface/parport/stepper-gantry/stepper_mm.ini||| |
|./by_interface/parport/gecko/Gecko_540B4/Gecko_540B4.ini||| |
|./by_interface/parport/gecko/Gecko_540B3/Gecko_540B3.ini||| |
|./by_interface/parport/nist-lathe/inch.ini||| |
|./by_interface/parport/stepper/sim_inch.ini||| |
|./by_interface/parport/stepper/stepper_inch.ini||| |
|./by_interface/parport/stepper/stepper_mm.ini||| |
|./by_interface/vigilant/vti.ini||| |
|./by_interface/pico/USC_encod/univstep.ini||| |
|./by_interface/pico/ppmc_vel/ppmc.ini||| |
[... 212 lines cut ...]
|./sim/qtvcp_screens/qtdragon/qtdragon_xyz.ini||| |
|./sim/qtvcp_screens/qtdragon/qtdragon_auto_tool_probe.ini||| |
|./sim/qtvcp_screens/qtdragon/qtdragon_xyyz.ini||| |
|./sim/gscreen/silverdragon/silverdragon.ini||| |
|./sim/gscreen/gscreen_custom/9axis.ini||| |
|./sim/gscreen/gscreen_custom/industrial.ini||| |
|./sim/gscreen/gscreen_custom/tester_metric.ini||| |
|./sim/gscreen/gscreen_custom/gscreen_gaxis.ini||| |
|./sim/gscreen/gscreen_custom/gaxis_no_plot.ini||| |
|./sim/gscreen/gscreen_custom/tester.ini||| |
|./sim/gscreen/industrial_lathe_wear/industrial_lathe_wear.ini||| |
|./sim/gscreen/ngcgui/pyngcgui_gscreen.ini||| |
|./sim/gscreen/ngcgui/pyngcgui_gscreen_popupkeyboard.ini||| |
|./sim/gscreen/gscreen.ini||| |
|./sim/gscreen/spartan/spartan.ini||| |
|./sim/gscreen/gscreen_mm.ini||| |
|./sim/gscreen/gscreen_lathe.ini||| |
|./sim/pyvcp_demo/pyvcp_bottom_panel.ini||| |
|./sim/pyvcp_demo/pyvcp_demo1.ini||| |
|./sim/touchy/touchy.ini|andypugh|| |
|./sim/touchy/ngcgui/pyngcgui_touchy.ini|andypugh|| |
|./sim/touchy/ngcgui/pyngcgui_touchy_lathe.ini|andypugh|| |
|./sim/touchy/ngcgui/pyngcgui_touchy_moveoff.ini|andypugh|| |
|./sim/touchy/gladevcp/touchy.ini|andypugh|| |
|./sim/touchy/gladevcp/gladevcp_touchy.ini|andypugh|| |

phillc54, 2022-06-14:
I started to go through the by_machine configs and quite a few require hardware.

I guess we could assume that any config using Axis as the GUI would rate a pass.

cakeslob, 2022-06-20:
Regarding the "by machine", and possibly "by interface" category,  but forgive me if my comments are unacceptable. 

For many of them, I dont know if they are relevant anymore. 

the cooltool - they appear to be running a beaglebone/machinekit controller now, that is being maintained by one of the machinekit guys it looks like

smithy - I dont know the situation before, but they dont appear to sell any cnc anymore

sherline - I think their newer stuff is using a masso controller, but they do appear to still sell a linuxcnc setup, and not to be a narc, but they also sell a debian/lcnc distrobution disc (i dont know how kosher this is) , if anyone is in contact with them, perhaps we ask them for a working configuration or to test the sample configs

tormach -  do they even support linuxcnc configs anymore now that they have pathpilot? 

by interface, obviously paraport, pico, and mesa are still relevant, but what about the others?

silopolis, 2022-06-20:
Le lun. 20 juin 2022 à 02:20, cakeslob ***@***.***> a écrit :

> Regarding the "by machine", and possibly "by interface" category
>

If confirmed, I'd propose a DEPRECATION notice for 2.9, meaning end of
support, before removal in a future release.