LinuxCNC/linuxcnc #1767: qtDragon, Selection boxes in front of the options on Settings tab are missing

snelweg, 2022-06-16:
Here are the steps I follow to reproduce the issue:

 1. opened the tab [settings]

This is what I expected to happen:
  Selection boxes in front of the options

This is what happened instead:
No selection boxes

It used to work properly a few updates ago.

[image]

Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

 * I am using this kernel version (shown by `uname -a`):
 Linux cnc-machine 4.19.0-20-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.235-1 (2022-03-17) x86_64 GNU/Linux

 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
   * [X] the apt version, 
   ** deb     http://buildbot.linuxcnc.org/ buster master-rtpreempt
   ** deb-src http://buildbot.linuxcnc.org/ buster master-rtpreempt

 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 LINUXCNC - 2.9.0-pre0-6687-g3bbfd0017

 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 qtDragon
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 
  Mesa 7i96, 7i84

c-morley, 2022-06-17:
Is problem this just in this theme?

snelweg, 2022-06-17:
It happens in qtDragon_hd too

c-morley, 2022-06-17:
Yes but just the argentium style?
I pushed a fix for both qtdragon and dragon_hd

snelweg, 2022-06-18:
I have to check at my machine, will be there tomorrow. Thanks

snelweg, 2022-06-25:
Oops, yes there is an update. The machine that uses the apt method of updating works. The other machine (buildbot) does not.
I have taken a rest from LinuxCNC, after updating my production machine everything stopped working. Maybe next week I will spend some more time on it.

c-morley, 2022-07-06:
I think these are fixed. i will close the issues. please reopen if the problems persists.

closed 2022-07-06 by c-morley