LinuxCNC/linuxcnc #1770: qtDragon, M6 remapping does not work

snelweg, 2022-06-16:
Following the documentation at https://linuxcnc.org/docs/devel/html/gui/qtdragon.html#_auto_tool_measurement the remapping just does not work. 
There are loads of discussions on the forum about this issue. A few versions ago I kinda got it to work, but now with the 2.9 pre nothing goes. I noticed that pathnames and even some routine names are different than mentioned in the documentation.

Here are the steps I follow to reproduce the issue:
MDI: M6 T11

GCode file containing a toolchange, for example M6 T11 

This is what I expected to happen:
- Measure the workpiece with the touch probe
- load job
- start job
- spindle moves to the tool change position and requests the first tool
- tool clamped and confirmed
- spindle drives to the tool sensor and measures the first tool
- first milling begins
- first milling finished
- spindle moves to the tool change position and requests the second tool
- tool clamped and confirmed
- spindle moves to the tool sensor and measures the second tool
- second milling begins
- etc. until the job is finished

This is what happened instead:
Nothing. The machine hangs and gives up after a while.

It all works in gmoccapy.
'
Information about my hardware and software:

    I am using this Linux distribution and version (often, shown by lsb_release -a):
    Distributor ID: Debian
    Description: Debian GNU/Linux 10 (buster)
    Release: 10
    Codename: buster

    I am using this kernel version (shown by uname -a):
    Linux cnc-machine 4.19.0-20-rt-amd64 

https://github.com/LinuxCNC/linuxcnc/pull/1 SMP PREEMPT RT Debian 4.19.235-1 (2022-03-17) x86_64 GNU/Linux

I am running ...

A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
A binary I built myself
A binary version from some other source besides linuxcnc.org
the apt version,
** deb http://buildbot.linuxcnc.org/ buster master-rtpreempt
** deb-src http://buildbot.linuxcnc.org/ buster master-rtpreempt

I am using this LinuxCNC version (shown in package manager or, for git versions, scripts/get-version-from-git):
LINUXCNC - 2.9.0-pre0-6687-g3bbfd0017

I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
qtDragon

I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):
Mesa 7i96, 7i84

c-morley, 2022-06-17:
Can you run this from the terminal and pastbin the output? most likey the remap code is having a problem beuing found.

c-morley, 2022-06-17:
Ok that good info. thanks

c-morley, 2022-06-17:
1) yes the docs are wrong the autoprobe is called  qt_auto_tool_probe.ngc
2) Ok I will add more info to docs
3) is there: ~/linuxcnc/nc_files/examples/remap_subroutine?

snelweg, 2022-06-18:
3. No, that directory does not exist either. The whole remap_subroutine dir is not found on my machine.

c-morley, 2022-06-18:
What is in nc_files? or is that missing too?

snelweg, 2022-06-18:
Here are all the files and directories in the ~/linuxcnc directory

`@DESKTOP-9FIBCI1:~/linuxcnc$ tree
.
├── configs
│   ├── my-mill
│   │   ├── custom.hal
│   │   ├── custom_postgui.hal
│   │   ├── my-mill.hal
│   │   ├── my-mill.ini
│   │   ├── postgui_call_list.hal
│   │   ├── pyvcp_options.hal
│   │   ├── qtvcp_postgui.hal
│   │   └── tool.tbl
│   ├── my-mill.stepconf
│   ├── sim.gmoccapy
│   │   ├── 4_axis.ini
│   │   ├── 6_axis.ini
│   │   ├── 7_axis.ini
│   │   ├── 7_axis_tool.ini
│   │   ├── 8_axis.ini
│   │   ├── 9_axis.ini
│   │   ├── button.glade
│   │   ├── core_sim4.hal
│   │   ├── core_sim_6.hal
│   │   ├── core_sim_7.hal
│   │   ├── core_sim_8.hal
│   │   ├── core_sim_9.hal
│   │   ├── core_sim.hal
│   │   ├── core_sim_XYZAB.hal
│   │   ├── core_sim_XYZAC.hal
│   │   ├── core_sim_XYZCW.hal
│   │   ├── dro.glade
│   │   ├── gmoccapy2.glade
│   │   ├── gmoccapy_4_axis.ini
│   │   ├── gmoccapy.ini
│   │   ├── gmoccapy_left_panel.ini
│   │   ├── gmoccapy_messages.ini
│   │   ├── gmoccapy_pedant.ini
│   │   ├── gmoccapy_postgui.hal
│   │   ├── gmoccapy.pref
│   │   ├── gmoccapy_right_panel.ini
│   │   ├── gmoccapy_sim_hardware_button.ini
│   │   ├── gmoccapy_tool_sensor.ini
│   │   ├── gmoccapy_tool_sensor.para
│   │   ├── gmoccapy_with_user_tabs.ini
│   │   ├── gmoccapy_XYZAB.ini
│   │   ├── gmoccapy_XYZAC.ini
│   │   ├── gmoccapy_XYZCW.ini
│   │   ├── gv.in
│   │   ├── hallib -> /usr/share/linuxcnc/hallib
│   │   ├── h_buttonlist.glade
│   │   ├── h_buttonlist.hal
│   │   ├── jogwheel.glade
│   │   ├── jogwheel.hal
│   │   ├── lathe.tbl
│   │   ├── left.hal
│   │   ├── left_panel.hal
│   │   ├── left_panel.ini
│   │   ├── macros
│   │   │   ├── change.ngc
│   │   │   ├── go_to_position.ngc
│   │   │   ├── halo_world.ngc
│   │   │   ├── i_am_lost.ngc
│   │   │   ├── images
│   │   │   │   ├── goto_x_y_z.png
│   │   │   │   ├── i_am_lost.png
│   │   │   │   └── macro_8.png
│   │   │   ├── increment.ngc
│   │   │   ├── jog_around.ngc
│   │   │   ├── macro_0.ngc
│   │   │   ├── macro_10.ngc
│   │   │   ├── macro_11.ngc
│   │   │   ├── macro_12.ngc
│   │   │   ├── macro_13.ngc
│   │   │   ├── macro_14.ngc
│   │   │   ├── macro_15.ngc
│   │   │   ├── macro_1.ngc
│   │   │   ├── macro_2.ngc
│   │   │   ├── macro_3.ngc
│   │   │   ├── macro_4.ngc
│   │   │   ├── macro_5.ngc
│   │   │   ├── macro_6.ngc
│   │   │   ├── macro_7.ngc
│   │   │   ├── macro_8.ngc
│   │   │   ├── macro_9.ngc
│   │   │   ├── macro_Instructions.txt
│   │   │   └── on_abort.ngc
│   │   ├── messages.ini
│   │   ├── pendant.glade
│   │   ├── pendant.hal
│   │   ├── pin.out
│   │   ├── position.txt
│   │   ├── python
│   │   │   ├── remap.py
│   │   │   ├── stdglue.py
│   │   │   └── toplevel.py
│   │   ├── README
│   │   ├── README_es
│   │   ├── right_panel.ini
│   │   ├── sig.out
│   │   ├── sim_hardware_button.ini
│   │   ├── simulated_home.hal
│   │   ├── simulated_home_lathe.hal
│   │   ├── sim.var
│   │   ├── sim.var.bak
│   │   ├── spindle_sim.hal
│   │   ├── tool_sensor.glade
│   │   ├── tool_sensor.hal
│   │   ├── tool_sensor.ini
│   │   ├── tool_sensor.para
│   │   ├── tool.tbl
│   │   ├── v_buttonlist.glade
│   │   ├── v_buttonlist.hal
│   │   ├── vcp_box.glade
│   │   ├── vcp_box.hal
│   │   ├── with_user_tabs.ini
│   │   ├── XYZAB.ini
│   │   ├── XYZAC.ini
│   │   └── XYZCW.ini
│   └── sim.qtvcp_screens.qtdragon
│       ├── core_sim.hal
│       ├── gantrysim.hal
│       ├── hallib -> /usr/share/linuxcnc/hallib
│       ├── python
│       │   ├── __pycache__
│       │   │   └── remap.cpython-310.pyc
│       │   ├── qt_auto_probe_tool.ngc -> ../../../nc_files/remap-subroutines/qt_auto_probe_tool.ngc
│       │   ├── remap.py
│       │   ├── stdglue.py -> ../../../../nc_files/remap_lib/python-stdglue/stdglue.py
│       │   └── toplevel.py
│       ├── qtdragon_auto_tool_probe.ini
│       ├── qtdragon_hd.ini
│       ├── qtdragon.ini
│       ├── qtdragon.log
│       ├── qtdragon.log.txt
│       ├── qtdragon_postgui.hal
│       ├── qtdragon_postgui_ya.hal
│       ├── qtdragon.pref
│       ├── qtdragon.var
│       ├── qtdragon.var.bak
│       ├── qtdragon_xyyz.ini
│       ├── qtdragon_xyza.ini
│       ├── qtdragon_xyz.ini
│       ├── qtdragon_xyzya.ini
│       ├── qtdragon_xyzy.ini
│       ├── qtvcp
│       │   └── screens
│       │       └── qtdragon
│       │           ├── __pycache__
│       │           │   └── resources.cpython-310.pyc
│       │           └── resources.py
│       ├── README
│       ├── silver_dragon.png
│       └── tool.tbl
├── hal_sigs_graphviz.py
└── nc_files
    ├── examples -> /usr/share/linuxcnc/ncfiles
    ├── gcmc_lib -> /usr/share/linuxcnc/ncfiles/gcmc_lib
    ├── gladevcp_lib -> /usr/share/linuxcnc/ncfiles/gladevcp_lib
    ├── ngcgui_lib -> /usr/share/linuxcnc/ncfiles/ngcgui_lib
    ├── ngcgui.txt
    ├── readme.ngc
    ├── remap_lib -> /usr/share/linuxcnc/ncfiles/remap_lib
    └── remap-subroutines -> /usr/share/linuxcnc/ncfiles/remap-subroutines
`

c-morley, 2022-06-18:
try using:
SUBROUTINE_PATH =/usr/share/linuxcnc/ncfiles/remap-subroutines:/usr/share/linuxcnc/ncfiles/remap_lib

c-morley, 2022-06-19:
Sorry this is what i meant to type
SUBROUTINE_PATH =/usr/share/linuxcnc/ncfiles/remap-subroutines

c-morley, 2022-07-06:
I added some work on paths for owords/remaps to allow absolute paths.
Getting paths right  seems to be more of a pain then it should.
In development the paths are different then when linuxcnc are installed so installed paths don't get tested by me.

I assumed that the examples directory is added to linuxcnc/nc_files when linuxcnc is installed - maybe i was wrong.

rmu75, 2025-12-18:
Is this still relevant?