LinuxCNC/linuxcnc #1788: G73: Add word to set retract height

silopolis, 2022-07-03:
Object says it all: it would be nice to have a word allowing to set the desired chip breaking retract height.

c-morley, 2022-07-04:
address?

silopolis, 2022-07-04:
Le lun. 4 juil. 2022 à 07:50, c-morley ***@***.***> a écrit :

> address?
>

If i'm not mistaken, G M T S F O are G-code words, and X Y... R Q L... are
G-code addresses

>

c-morley, 2022-07-15:
I had to loo it up :) of course you are right.

c-morley, 2022-07-23:
Not quite what you wanted but a start in the right direction I think #1860

silopolis, 2022-07-23:
Le sam. 23 juil. 2022 à 08:25, c-morley ***@***.***> a écrit :

> Not quite what you wanted but a start in the right direction I think #1860
> <https://github.com/LinuxCNC/linuxcnc/pull/1860>
>

A very nice move indeed! 👍

If I could set an expression using tool dia for each that'd be a perfect
default setting feature...

c-morley, 2022-07-23:
I bet you could override these setting with remap code, but I'm not sure.
using an address in the Gcode would be most universal - I'm not sure if there is industrial gcode that shows this. it would be nice to try and follow a standard of some kind. We generally try to stick with fanuc.

silopolis, 2022-07-23:
Le sam. 23 juil. 2022 à 20:05, c-morley ***@***.***> a écrit :

> I bet you could override these setting with remap code, but I'm not sure.
> using an address in the Gcode would be most universal - I'm not sure if
> there is industrial gcode that shows this. it would be nice to try and
> follow a standard of Somme kind. We generally try to stick with fanuc.
>
I believe this is for fanuc

https://gcodetutor.com/fanuc-training-course/g73-g83-drilling-cycle.html

And this is is for sinumerik

http://www.helmancnc.com/siemens-sinumerik-cycle83-deep-hole-peck-drilling-cycle/

c-morley, 2022-07-23:
Simumerik isn't generic gcode so that doesn't help with deciding how to set up the gcode command in linuxcnc.
The fanuc examples are what linuxcnc already does, aside from G83 P for dwell and that the peck distance is hard coded at the moment. But your reference does say that the peck distance is set in parameters roughly the same as INI.
 
In either example I don't see how you set it to peck a bit then retract - it seems either or.
which actually makes sense to me.
-long chips need the drill to pull out to clear
-short chips would tend to stay in the hole when you pull out to clear.

Maybe I misunderstood what you want.

grandixximo, 2026-10-04:
Done in #2528 (merged 2026-01-20): G73 and G83 take an optional D word setting the peck retract distance, with [RS274NGC]PARAMETER_G73_PECK_CLEARANCE / PARAMETER_G83_PECK_CLEARANCE as INI defaults. Closing.

closed 2026-10-04 by grandixximo